lm-sensors/lm-sensors #3: lm-sensors site down ?

promeneur, 2015-11-10:
since several day i have tried to connect without success to www.lm-sensors.org site .

i check it is a general pb with http://www.downforeveryoneorjustme.com/

thanks

wdlkmpx, 2016-04-09:
I really wonder what happened, it's still down..

groeck, 2016-04-09:
On 04/09/2016 03:19 PM, wdlkmpx wrote:
> I really wonder what happened, it's still down..
>
It won't come back either. Try https://hwmon.wiki.kernel.org,
though all that is there for now is the somewhat outdated list
of supported chips.

Guenter

promeneur, 2016-04-09:
there is an archive
https://web.archive.org/web/20150319170446/http://lm-sensors.org/browser/lm-sensors

groeck, 2016-04-10:
Yes, I know. That doesn't make the information appear on the new site automatically.

LeSpocky, 2016-04-10:
I wonder where release tarballs are available? Embedded build systems like buildroot, openwrt, and ptxdist all depend on downloading the source and accessed e.g. http://dl.lm-sensors.org/lm-sensors/releases in the past. Where is the official place to get lm_sensors nowadays? Many other projects rely on it. :-/

promeneur, 2016-04-10:
Jean Delvare is one of the main developers of lm-sensors project.
i found his site
http://jdelvare.nerim.net/
and
https://fr.linkedin.com/in/jean-delvare-3391135

He is the maintainer of hwmon et dmi of linux kernel and also many drivers of i2c
see
http://jdelvare.nerim.net/devel.html

i tried to join him with his mail khali@linux-fr.org but i get an error ".. mail server access fails.."
i found another address jdelvare@suse.de. i just send a question. this time there is no mail server failure. i wait for his answer.

also we can't access to the  mirror ftp://ftp.netroedge.com/pub/lm-sensors/
we can access to
http://jdelvare.nerim.net/devel/lm-sensors/drivers/
http://jdelvare.nerim.net/mirror/i2c-tools/
http://www.nongnu.org/dmidecode/
http://savannah.nongnu.org/projects/dmidecode/
http://ftp.igh.cnrs.fr/pub/nongnu/dmidecode/

LeSpocky, 2016-04-10:
The mirror ftp://ftp.netroedge.com/pub/lm-sensors/ is the one buildroot uses since January this year.

promeneur, 2016-04-12:
hello some news from Jean Delvare
the admin of lm-sensors.org is inaccessible.
they are transfering the contents. 
the new list is linux-hwmon@vger.kernel.org
the new wiki (work in progress) is https://hwmon.wiki.kernel.org/
for the source Jean says  "the github repo of Guenter" thus this site.

MichaelBitard, 2016-06-15:
Still nothing new on this ?

groeck, 2016-07-16:
lm-sensors is and will remain down. the hwmon wiki is far from complete, and will be for a while as I have little time to do the work.

closed 2016-07-16 by groeck

grajeshkannan, 2016-09-28:
Refer to this forum, it got a work around for this issue
https://forum.openwrt.org/viewtopic.php?id=64528