lm-sensors/lm-sensors #8: Reflect new lm-sensors webpage

msehnout, 2016-08-16:
It seems that the lm-sensors web page is moving here:
[hwmon.wiki.kernel.org](https://hwmon.wiki.kernel.org/)

Source code should reflect that change, because it is referencing the previous web page many times:
```
$ git remote -v
origin	https://github.com/groeck/lm-sensors.git (fetch)
origin	https://github.com/groeck/lm-sensors.git (push)
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
$ git grep lm-sensors.org | wc -l
37
```

terrycloth, 2016-11-20:
Moreover, lm-sensors.org seems to be dead. At least, it was when I tried to visit the site tonight.

groeck, 2018-08-17:
See README.

olysonek, 2018-08-21:
> See README.

No longer relevant. Reopening. I'll fix this issue as soon as I can.

commit 6413b808 (2018-09-01): Fix stale links

Fixes #8
closed 2018-09-01 by olysonek