lm-sensors/lm-sensors #10: Question: Nuvoton with ASUS Mobo

borgdylan, 2016-09-19:
I have a Sabertooth X99 mobo and ran sensors-detect. The nuvoton sensor chip was successfully detected but no readings come out from it when running sensors. The BIOS reads all fan speeds just fine. What can I do to read fan speeds while booted into Linux?

groeck, 2016-09-19:
On Mon, Sep 19, 2016 at 11:00:53AM -0700, Dylan Borg wrote:
> I have a Sabertooth X99 mobo and ran sensors-detect. The nuvoton sensor chip was successfully detected but no readings come out from it when running sensors. The BIOS reads all fan speeds just fine. What can I do to read fan speeds while booted into Linux?
> 
"no readings come out from it when running sensors"

What does that mean ?

Do you see anything in dmesg ? Did you boot your system with
acpi_enfore_resources=lax ?

Guenter

borgdylan, 2016-09-19:
I did not use that parameter for fear that it may interfere with the BIOS fan controller. I run some workloads that generate plenty of heat (the Core i7 6900K is an extreme edition chip that makes too much heat) and would like to monitor fan speeds without compromising system stability.

borgdylan, 2016-09-19:
Is is it safe enough for me to enable that kernel command line param without messing with the mobo's ability to throttle the speeds. I only need readonly access to those sensors.

groeck, 2016-09-20:
On Mon, Sep 19, 2016 at 01:35:46PM -0700, Dylan Borg wrote:
> I did not use that parameter for fear that it may interfere with the BIOS fan controller. I run some workloads that generate plenty of heat (the Core i7 6900K is an extreme edition chip that makes too much heat) and would like to monitor fan speeds without compromising system stability.
> 
Well, you'll have to make a decision here. FWIW, the fan controller runs
in the SuperIO chip. But without the option you won't be able to
see any fan speeds (or board temperatures, or voltages) in Linux.

Guenter

closed 2017-03-12 by groeck