lm-sensors/lm-sensors #11: Support for sensors on A88XM-A/USB 3.1 ?

DeadSix27, 2016-12-16:
How can I find out which chip it uses  and wether thats supported?

E.g wether its a ITE or Nuvoton etc.

All I get to work is:

```
$ sensors
k10temp-pci-00c3
Adapter: PCI adapter
temp1:         +0.0 C  (high = +70.0 C)
                       (crit = +70.0 C, hyst = +69.0 C)

radeon-pci-0008
Adapter: PCI adapter
temp1:         -7.0 C  (crit = +120.0 C, hyst = +90.0 C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

fam15h_power-pci-00c4
Adapter: PCI adapter
power1:           N/A  (crit =  65.19 W)
```

Only thing detection outputs is:
```
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)
Probing for `SPD EEPROM'...                                 YYes
    (confidence 8, not a hardware monitoring chip)
```

groeck, 2016-12-16:
On 12/16/2016 05:59 AM, DeadSix27 wrote:
> How can I find out which chip it uses and wether thats supported?
>
> E.g wether its a ITE or Nuvoton etc.
>
> All I get to work is:
>
> |$ sensors k10temp-pci-00c3 Adapter: PCI adapter temp1: +0.0 C (high = [...]

If it doesn't report any known or unknown Super-IO chips, there is nothing else you can do.
The presence of asus-isa suggests that Super-IO features are managed by ACPI (which is common
on ASUS boards), and the fact that only one fan with speed 0 is shown suggests that the
ACPI DSDT information in the driver is incomplete. Since ASUS does not support Linux and does
not publish the source of its ACPI DSDT, all someone could do would be to disassemble it,
try to extract the necessary information, and enhance the driver accordingly. That is
quite unlikely to happen.

Guenter

DeadSix27, 2016-12-16:
That's unfortunate, do you know if Gigabyte has better support in that regard? (Its the 3rd Mainboard RMA I am on... first one being ASRock, second one too, but that one kept freezing my system every night for whatever reasons, both had nuvoton chips correctly detecting, but I never bought gigabyte before, so I wonder about wether they just use Nuvoton too and are likely to be supported.

(In fact, Gigabyte is the only other choice there is.)

groeck, 2016-12-16:
On 12/16/2016 07:40 AM, DeadSix27 wrote:
> That's unfortunate, do you know if Gigabyte has better support in that regard? (Its the 3rd Mainboard RMA I am on... first one being ASRock, second one too, but that one kept freezing my system [...]

All I can say is that I personally prefer Gigabyte. Their technical support has been
very helpful whenever we had problems with Super-IO chip support on one of their boards.

I think ASRock may also use ACPI, which can cause interference with the Linux driver
if you actively poll sensors.

Guenter

DeadSix27, 2016-12-16:
Thanks for the info, I for some reason always have bad feelings of Gigabyte(and MSI) and good of ASUS
I had many MSI boards die in the past and my friends had dying Gigabyte boards, but I have yet to come across friend or own ASUS boards to die.

groeck, 2016-12-17:
On 12/16/2016 07:59 AM, DeadSix27 wrote:
> Thanks for the info, I for some reason always have bad feelings of Gigabyte(and MSI) and good of ASUS
> I had many MSI boards die in the past and my friends had dying Gigabyte boards, but I have yet to come across friend or own ASUS boards to die.
>

Different people, different experience. It may well be that the wuality of ASUS boards
is better; I simply don't know. What I do know is that they don't support Linux.

[ and my MSI board works quite well ;-) ]

Guenter

DeadSix27, 2016-12-17:
Well, what I look for is Good linux support and if Gigabyte is better in that regard I take the risk and try it out.
So far the ASUS board actually runs really well, compared to the ASRock one crashing every damn night..

But ye, the lack of sensor support is a big bummer as that gives me pretty much 0 possibilities to monitor the system status without expensive or weird gimmicks. (As the AMD APU sensor is borked.. unfortunately)
Yet AMD APUs make so nice cheap tiny servers for private use...

DeadSix27, 2016-12-18:
By the way: I ordered the Gigabyte one now, can you give any information on this?:
https://bluehatrecord.wordpress.com/2014/03/13/temperature-monitoring-in-the-gigabyte-ga-f2a88xm-d3h/
And the sensor ITE IT8620E being supported in lm-sensors?

I'd guess it is supported by now, the post is old anyway, (oh and the motherboard is as previously mentioned another version of that one, but its safe to assume it uses the same ITE chip, It uses a ITE one at least for sure, as thats stated on the specs website.

groeck, 2016-12-18:
On 12/18/2016 03:14 AM, DeadSix27 wrote:
> By the way: I ordered the Gigabyte one now, can you give any information on this?:
> https://bluehatrecord.wordpress.com/2014/03/13/temperature-monitoring-in-the-gigabyte-ga-f2a88xm-d3h/
> And the sensor ITE IT8620E being supported in lm-sensors?
>
IT8620E has been supported since early 2015.

Guenter

DeadSix27, 2016-12-20:
@groeck 
Just checked the high res images:
[image]

It's definitely a IT8620E :) Board is on its way, thanks again for your support @groeck

DeadSix27, 2016-12-23:
_(new reply for the email notification ;) )_

Anyway, thanks a lot @groeck 
I just finished setting up the new motherboard, sensor works fine, even if its not that accurate at least I have some values to work with!

Oh and since we're on this, do you happen to know what tempXX represents what?

I mean it says "sensor = Intel PECI" but I do not even have a Intel system.... (see bottom)

```
$sensors
fam15h_power-pci-00c4
Adapter: PCI adapter
power1:           N/A  (crit =  65.19 W)

radeon-pci-0008
Adapter: PCI adapter
temp1:         -7.0°C  (crit = +120.0°C, hyst = +90.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
temp1:         +0.0°C  (high = +70.0°C)
                       (crit = +80.0°C, hyst = +79.0°C)

it8620-isa-0228
Adapter: ISA adapter
in0:          +0.74 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +1.51 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +2.23 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +2.23 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.31 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +2.88 V
fan1:         624 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:         -8.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +13.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp4:        +34.0°C
temp5:        +45.0°C
temp6:        +45.0°C
intrusion0:  ALARM
```

groeck, 2016-12-23:
> /(new reply for the email notification ;) )/
>
> Anyway, thanks a lot @groeck <https://github.com/groeck>
> I just finished setting up the new motherboard, sensor works fine, even if its not that [...]

No, that depends on the board.

> I mean it says "sensor = Intel PECI" but I do not even have a Intel system.... (see bottom)

Yes, that should report AMDTSI. Driver bug - it does not distinguish between the two.

Guenter

DeadSix27, 2016-12-23:
Thanks anyway :) I'll probably figure it out, and a merry Christmas (or whatever you celebrate)

closed 2016-12-23 by DeadSix27