lm-sensors/lm-sensors #12: ga-z170n-wifi

gbdu, 2016-12-22:
Hey, I can't seem to get my lm-sensors to recognize my cpu fan, I tried adding acpi_enforce_resources=lax to my boot line and re-running sensors-detect to no avail. The following is the output from sensors-detect http://pastebin.com/zryy0iEi

Thank you

groeck, 2016-12-22:
On Thu, Dec 22, 2016 at 04:40:46AM -0800, gbdu wrote:
> Hey, I can't seem to get my lm-sensors to recognize my cpu fan, I tried adding acpi_enforce_resources=lax to my boot line and re-running sensors-detect to no avail. The following is the output from sensors-detect http://pastebin.com/zryy0iEi
> 
Did you try to load the it87 driver ?

Guenter

gbdu, 2016-12-27:
That worked, thank you!

closed 2016-12-27 by gbdu