lm-sensors/lm-sensors #14: libsensors seems to write cpu data into an array w/o sanity checks on the index

luebking, 2017-02-15:
linux 4.9 somehow messes up cpu node counting; while this smells like a bug in the kernel, /proc/stat is input and should not be trusted (ie. libsensors should not crash on this)

Might be a bug in ksysguardd, I don't know enough about the implementation details.

https://bbs.archlinux.org/viewtopic.php?pid=1691281#p1691281

groeck, 2017-02-16:
I am pretty much completely at loss with this one. The bug mentions nothing about libsensors, but suddenly there is a statement out of the blue blaming libsensors for an observed crash. Also, I am unable to parse "/proc/stat is input and should not be trusted (ie. libsensors should not crash on this)". 

Please provide more information. What is the association between /proc/stat and libsensors ? What cpu data does libsensors write into which array ? Where does libsensors access /proc/stat ?

luebking, 2017-02-16:
As mentioned, this might be a bug in ksysguardd alone.
I don't know *what* the bug is, and whether ksysguardd really reads /proc/stat by itself (what sounds weird enough) but whoever reads proc stat expects a certain structure which is no longer valid with kernel 4.9.

If libsensors doesn't even touch /proc/stat and ksysguard has its own implementation for this, feel free to close this (just if it had been libsensors, this might have qualified for a critical overflow)

I'll then file a bug against ksysguard (I have neither ksysguardd nor that CPU myself, so I couldn't test - sorry for the noise, in case)

luebking, 2017-02-16:
https://bugs.kde.org/show_bug.cgi?id=376537

closed 2017-02-16 by luebking