lm-sensors/lm-sensors #16: Support for AMD Family 17h Processors

sarnex, 2017-03-10:
Hi, 

The new AMD Ryzen chips are 17h, and are unsupported by lm-sensors. 

Please add support for them, and let me know if you need any information, as I have one.

Thanks,
Sarnex

sjug, 2017-03-12:
Is this tool actively maintained?

groeck, 2017-03-12:
Not paid, not 24/7.

This was always supposed to be a shadow repository. Unfortunately the main repository is gone.

sjug, 2017-03-12:
@groeck How much work is it to add sensor compatibility with new processors and motherboards? Ryzen is a pretty big deal, maybe you can crowdfund something if that would help? What can we do to help short of learning C?

groeck, 2017-03-12:
Money isn't an issue. Time is.
Question though is what you are asking for. Support for the processor with sensors-detect ? That would be a couple of lines of python code, but doesn't really mean much. Support for the processors with the "sensors" command ? That has nothing to do with this repository; it would have to be added with a hardware monitoring driver, probably by extending the existing k10temp and fam15h_power drivers. That might require a few lines of C code if the functionality is the same as with family 15/16 chips, or it might require new drivers. Patches for all those changes are welcome.

xpander69, 2017-03-12:
thanks for explaining.

ggrats, 2017-03-21:
Thanks for being literally no help groeck. Maybe point us somewhere?
Don't post if you "don't have time" and want to complain about life instead of giving these guys 10 seconds of your time to admit you're not sure instead of lead them in loops "oh it's just a few lines of code" so do it?

Ryzen is difficult to read, currently. Not much supports it. there are many directories for cpu info
From what I can see the CPU's very supported for reading thermal yet or AMD hasn't told us how to.

everyone is welcome to run:
sudo cat /sys/class/thermal/cooling_device*/device/thermal_cooling/cur_state
or something similar which will be the same output as acpi which really never helps anyone
sudo cat /sys/class/hwmon/hwmon0 or hwmon1 etc will have /temp1_input in it (possibly) mine shows gpu and motherboard

and in /sys/bus/cpu/devices/cpu0 cpu1 cpu2 etc there's TONS of thermal info to be had in any supported cpu, but on ryzen is empty, there's no info there

so, at least on the 4.4 kernel.. looks like kernel 4.10 added lots of multithreading support for ryzen specifically but I'll try out 4.9 to see if I can find any thermal info. If anyone actually wants to participate feel free to give input. I don't know much but crying about "not enough time and it's only a few lines of code" isn't something I'll sit here and do, since that isn't the case. But if it really is, prove me wrong groeck.

groeck, 2017-03-22:
Datasheet for Family 17h CPUs is not published by AMD. Adding temperature sensor support to the kernel driver will require additional technical information which is not currently available. Specifically, it is not known in which PCI device the temperature sensor resides, and if the registers are identical to earlier chips.

sarnex, 2017-04-21:
https://support.amd.com/TechDocs/54945_PPR_Family_17h_Models_00h-0Fh.pdf

Does that datasheet contain anything useful? Thanks

groeck, 2017-04-21:
@sarnex: Unfortunately not. The required information is traditionally in the "BIOS and Kernel
Developer’s Guide" which has not been published.

BeerSerc, 2017-04-21:
@groeck : https://support.amd.com/TechDocs/54945_PPR_Family_17h_Models_00h-0Fh.pdf
Does that help?
And thanks for lm-sensors in general!
Cheers,
B

sjug, 2017-04-21:
@BeerSerc are you serious?

groeck, 2017-05-07:
@sarnex @BeerSerc : Some additional information: I was told that the document needed is in fact the PPR. Problem is that there are two versions of this document, one that is public and one that is available only under NDA. The version available under NDA supposedly includes the required information.

sjug, 2017-05-07:
Great news, how do we sign the NDA and get the document?

sarnex, 2017-05-08:
I assume if someone does get ahold of it by signing the NDA, they won't be
able to release code based on information in it.

groeck, 2017-05-08:
@sarnex: Yes, that is exactly the problem.

tomreyn, 2017-06-30:
Related:
- [Need BIOS and Kernel Developer’s Guide Family 17h (community.amd.com)](https://community.amd.com/thread/213584)
- [Tech docs (support.amd.com)](http://support.amd.com/en-us/search/tech-docs)
- [Developer Guides, Manuals & ISA Documents (developer.amd.com)](http://developer.amd.com/resources/developer-guides-manuals/)

At the time of writing, the latest _[Processor Programming Reference (PPR) for AMD Family 17h Models 00h-0Fh Processors](http://support.amd.com/TechDocs/54945_PPR_Family_17h_Models_00h-0Fh.pdf)_ (publication #54945) is revision 1.14, released 5/3/2017.

And it seems the information buyers of this CPU require to reliably make use of the product is still missing.

AlexandreBonneau, 2017-07-28:
Anybody here contacted AMD to ask for such info?

johnbridgman, 2017-07-29:
Is lack of information still an issue ? If so I can ask around internally. I'm on the GPU side so will probably have to ask a few dumb questions re: exactly what we are looking for... what I'm a bit fuzzy on is "on die" temperature sensors vs "on motherboard" sensors which I believe are what are being discussed here:

https://linuxconfig.org/monitor-amd-ryzen-temperatures-in-linux-with-latest-kernel-modules

The reason for my confusion is that from what I remember all of the temperature measurement for our CPUs on Linux has been via motherboard sensors (which I thought talked to the CPU) rather than on-die sensors, but that is only what I have picked up from casual browsing while setting up my own systems at home. 

Thanks,
John

groeck, 2017-08-03:
@johnbridgman: Sure, if you can get an exception, that would help. Which information is needed: The PCI ID and location of the REPORTED_TEMP_CTRL_OFFSET register (or whatever it is called on family 17h; the name is from family 15h model 0x60 and 0x70) as well as the location of the index register to read it. The index register on family 15h is 0xb8, the offset is 0xd8200ca4. Also, it would help to know if there is a means to read the temperature offset from the CPU (20 degrees C for 1700X and 1800X), or if it is necessary to calculate the offset from the CPU type.

Re drivers here - yes, those are development versions, and sometimes experimental. Normally I keep drivers in sync with upstream, but the it87 driver has deviated so much that I'll need several weeks to bring the drivers in sync. Unfortunately I don't have that time right now.
Since someone is going to ask: No, I can not just copy the drivers over, even though I am the upstream hwmon maintainer. The rule of "one logical change per patch" also applies to maintainers.

rozhuk-im, 2017-09-04:
echo $((0x`setpci -s 00:0.0 60.l=0x59800 && setpci -s 00:0.0 64.l`/2097152*5/4))

groeck, 2017-09-05:
@rozhuk-im: That helped, thanks. Patch for the k10temp driver submitted upstream. Too late for the v4.14 kernel, but it will be available in v4.15.

closed 2017-09-05 by groeck

cemeyer, 2017-09-05:
@groeck Re: https://lkml.org/lkml/2017/9/4/503 , FYI At least some (1950X and I think 1920X) Threadripper CPU models have a 27°C offset too, if you want to try to correct a few more models automatically.

groeck, 2017-09-05:
On 09/04/2017 06:53 PM, Conrad Meyer wrote:
> @groeck <https://github.com/groeck> Re: https://lkml.org/lkml/2017/9/4/503 , FYI At least some (1950X and I think 1920X) Threadripper CPU models have a 27°C offset too, if you want to try to correct a few more models automatically.
> 

Fun part with the Threadripper CPU is that they have the same CPU model number.
The same microcode runs on both Ryzen and Threadripper. This means I will also
need a means to detect the affected CPU models, probably based on the model string.
Any idea what the CPU model string (as reported by /proc/cpuinfo) is ?

For 1700X, as example, it is "AMD Ryzen 7 1700X Eight-Core Processor".
I suspect it might be "AMD Ryzen Threadripper 1950X ...", but that is just
a guess and not good enough here.

Thanks,
Guenter

cemeyer, 2017-09-05:
My system isn't running Linux, but FreeBSD reports "AMD Ryzen Threadripper 1950X 16-Core Processor" in the early boot messages.  `x86info` also reports: "Processor name string (BIOS programmed): AMD Ryzen Threadripper 1950X 16-Core Processor".

AlexandreBonneau, 2017-09-05:
For info the Ryzen 5 1600 `model name` string as reported by `/proc/cpuinfo` under Linux is `AMD Ryzen 5 1600 Six-Core Processor`.

groeck, 2017-09-05:
Ryzen 5 should not have temperature offsets.

groeck, 2017-09-05:
On Tue, Sep 05, 2017 at 01:53:02AM +0000, Conrad Meyer wrote:
> @groeck Re: https://lkml.org/lkml/2017/9/4/503 , FYI At least some (1950X and I think 1920X) Threadripper CPU models have a 27°C offset too, if you want to try to correct a few more models automatically.
> 

Yes, and I am told from AMD that older models have various offsets
as well. This is going to be fun in a negative sense :-(.

Guenter

cemeyer, 2017-09-05:
Yeah, it's awful.  I don't know what AMD is thinking.  In FreeBSD we just punt on the issue and have the user configure an offset if they want to.

ar1111, 2017-09-11:
@groeck Here's what I see in /proc/cpuinfo for the Threadripper 1950x:

vendor_id	: AuthenticAMD
cpu family	: 23
model		: 1
model name	: AMD Ryzen Threadripper 1950X 16-Core Processor

Goddard, 2017-09-22:
Here is output of cat /proc/cpuinfo

sudo cat /proc/cpuinfo
processor       : 0
vendor_id       : AuthenticAMD
cpu family      : 23
model           : 1
model name      : AMD Ryzen Threadripper 1950X 16-Core Processor
stepping        : 1
microcode       : 0x8001129
cpu MHz         : 2200.000
cache size      : 512 KB
physical id     : 0
siblings        : 32
core id         : 0
cpu cores       : 16
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc extd_apicid amd_dcm aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_l2 mwaitx hw_pstate vmmcall fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves clzero irperf arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic overflow_recov succor smca
bugs            : fxsave_leak sysret_ss_attrs null_seg
bogomips        : 6787.11
TLB size        : 2560 4K pages
clflush size    : 64
cache_alignment : 64
address sizes   : 48 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate eff_freq_ro [13] [14]

processor       : 1
vendor_id       : AuthenticAMD
cpu family      : 23
model           : 1
model name      : AMD Ryzen Threadripper 1950X 16-Core Processor
stepping        : 1
microcode       : 0x8001129
cpu MHz         : 2200.000
cache size      : 512 KB
physical id     : 0
siblings        : 32
core id         : 1
cpu cores       : 16
apicid          : 2
initial apicid  : 2
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc extd_apicid amd_dcm aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_l2 mwaitx hw_pstate vmmcall fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves clzero irperf arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic overflow_recov succor smca
bugs            : fxsave_leak sysret_ss_attrs null_seg
bogomips        : 6787.11
TLB size        : 2560 4K pages
clflush size    : 64
cache_alignment : 64
address sizes   : 48 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate eff_freq_ro [13] [14]

and repeats 32 times of course

ElvishJerricco, 2017-10-14:
Is it currently possible to use `sensors` to read 1950X cpu temps? I've tried rebasing up to [here](https://git.kernel.org/pub/scm/linux/kernel/git/groeck/linux-staging.git/commit/?h=hwmon-next&id=faffc88b5ca6682a323e10de2d65ee9bdf17af48) onto Linux v4.13.5, and am unable to see cpu temps. Do I need to do more than just patch my kernel?

groeck, 2017-10-14:
linux-next has support for it. For convenience I created a repository named 'k10temp' which supports it.

ElvishJerricco, 2017-10-14:
I had a bunch of issues with v4.14rc4 that make me fairly sure `linux-next` won't work either, but I'll give it a shot. As for the `k10temp` repo, I believe that's no better than rebasing the commits I rebased, right?

groeck, 2017-10-14:
Hard for me to say; I don't know what you mean with "rebase". Do you mean you cherry-picked the patches on top of v4.13.5 ? That is pretty much what I did as well, though I only have ryzen systems, not threadripper. Threadripper does have pci device 1022:1463 (two of them, actually), so the driver should work. Questions then would be if dmesg shows anything, if the driver is loaded, and if "lspci -vv -d 1022:1463" shows anything useful.

ElvishJerricco, 2017-10-15:
The driver does seem to work for those two pci devices. But is that representative of the CPU temp?

cemeyer, 2017-10-15:
Yes?  It's the documented temperature sensor for the CPU.

ElvishJerricco, 2017-10-15:
OH geez that's embarrassing. Sorry, I'm really out of my element here. I thought there'd be one per core, like on all my other machines.

e-dard, 2017-10-19:
@groeck not sure if this is useful or not since there are already comments for the 1950X above, but here is `/proc/cpuinfo` for the `1920X`. It's as you might expect:

```
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 23
model		: 1
model name	: AMD Ryzen Threadripper 1920X 12-Core Processor
```

groeck, 2017-10-19:
On Thu, Oct 19, 2017 at 07:57:47AM +0000, Edd Robinson wrote:
> @groeck not sure if this is useful or not since there are already comments for the 1950X above, but here is `/proc/cpuinfo` for the `1920X`. It's as you might expect:
>
> ```
> processor	: 0
> vendor_id	: [...]
Yes, that one is covered. Tempterature offset is assumed to be 27 degrees C.

Guenter

aliensonneptune, 2017-11-03:
Okay, so I'm a noob to the whole Ryzen shebang, after all of the aforementioned updates would sensors return for Ryzen 3 1200 on latest Ubuntu out of box? Or would I have to mess with it?

groeck, 2017-11-03:
The changes are not yet upstream, so the latest Ubuntu won't be sufficient. Upstream kernel 4.15 will support Ryzen (and Threadripper) temperature sensors.

Goddard, 2017-11-03:
Some rumblings 18.04 will have it so I assume through hwe people on older version like 16.04 might get it as well?
https://www.phoronix.com/scan.php?page=news_item&px=Ubuntu-18.04-LTS-Linux-4.15

groeck, 2017-11-04:
I am not involved in Canonical's decisions about Linux kernel releases, sorry.

Goddard, 2017-11-04:
It would be cool if we had a way to private message people on GitHub for instances like this.

whompy, 2017-11-13:
I just noticed a bug in the present code: 
+ { 0x17, "AMD Ryzen 7 1600X", 20000 },
1600X is a Ryzen 5 model as shown in cpuinfo:
AMD Ryzen 5 1600X Six-Core Processor

Not sure if there is a better place to report bugs on this, so I threw it here (Sorry!)

groeck, 2017-11-13:
Thanks for the k10temp report. It might be better to report k10temp issues in the k10temp repository or upstream.

FireFrenzy, 2017-12-16:
For what its worth HWInfo64 can read the core temp, and on my 1950x (Aorus gaming 7) the offset does indeed appear to be 27 degree between Tdie and Tcng or whatever the short hand for "reported temps" is... 

Now i am not good enough to find logfiles or whatever but if you all can explain the steps to me i am more then willing to provide whatever data i can since i love me some OHWM

groeck, 2017-12-16:
Not sure what the (remaining) problem is. sensors-detect from ToT here supports reporting family 17h sensors, and k10temp in the ToT kernel does as well. The problem reported by whompy@ has been fixed as well. If there is still a problem, please open a separate issue and provide details.

Iambecomeroot, 2018-01-22:
So is the final verdict that we have to wait for kernel 4.15 in order to have cpu temps?

groeck, 2018-01-22:
You can always run the driver from here.

Iambecomeroot, 2018-01-22:
@groeck Sorry for the dumb question but what does that mean?

sarnex, 2018-01-22:
You will have to build the driver here manually for the kernel you want to
use, or run 4.15+

Iambecomeroot, 2018-01-22:
@sarnex ok thanks.

h3po, 2018-02-21:
I'm running 4.15.3-2-ARCH on a 1950x + asrock taichi x399 and the 27° offset doesn't seem to be applied (windows and bios report ~60°):

```
k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +31.8 C  (high = +70.0 C)

k10temp-pci-00cb
Adapter: PCI adapter
temp1:        +31.8 C  (high = +70.0 C)
```

```
$ cat /proc/cpuinfo| grep model | tail -n1
model name	: AMD Ryzen Threadripper 1950X 16-Core Processor
$ uname -rv
4.15.3-2-ARCH #1 SMP PREEMPT Thu Feb 15 00:13:49 UTC 2018
```

cemeyer, 2018-02-21:
@h3po The offset is negative in sign.  Likely it is your BIOS and Windows that are missing the offset.  60°C - 27° = ~33°C, which is a plausible temperature for this CPU.

groeck, 2018-02-21:
Depends if the system is under load or not. 33°C would be reasonable for an idle system. Another question is the Windows application used to report the temperature. Hwinfo should take it into account, for example (afaik it shows both temperatures).

h3po, 2018-02-22:
I guess I'll have to verify with a thermometer.

For the system in question, I'd be surprised if it's actually at 33° in idle. It's stuffed into a 2U supermicro server case with a tiny epyc SP3 heatsink.

groeck, 2018-02-22:
@h3po: Since the offset is negative, you are effectively arguing that, on your system, the offset should _not_ be applied, or that the offset should not be a 27° offset but something else. You'll need to provide actual evidence that this is the case.

measly, 2018-03-07:
My Debian Sid (4.15) system is a 1950X @ 3.9 + Asrock X399 Fatality Pro + Enermax 360 AIO. Running sensors (lm_sensors) at near idle gives me:

```
$ sensors
k10temp-pci-00cb
Adapter: PCI adapter
temp1:        +35.0°C  (high = +70.0°C)

nct6779-isa-0290
Adapter: ISA adapter
Vcore:                  +0.44 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.08 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.33 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.33 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.86 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +1.12 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +1.35 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.46 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.28 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in10:                   +0.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.85 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.68 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.92 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +0.83 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                   797 RPM  (min =    0 RPM)
fan3:                  2934 RPM  (min =    0 RPM)
fan4:                   358 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +31.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                 +34.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                 +8.0°C    sensor = thermistor
AUXTIN1:                +36.0°C    sensor = thermistor
AUXTIN2:                +34.0°C    sensor = thermistor
AUXTIN3:                +35.0°C    sensor = thermistor
SMBUSMASTER 0:          +62.0°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM                                                        
beep_enable:           disabled                                                     
                                                                                    
k10temp-pci-00c3                                                                    
Adapter: PCI adapter                                                                
temp1:        +35.0°C  (high = +70.0°C)
```

However, I am uncertain which of these degrees are there relevant to me and what to watch out for. "_k10temp-pci-00cb_" and "_k10temp-pci-00c3_" appear normal given the thread, but "_SMBUSMASTER 0_" has risen to over 98°C when running some intensive programs, despite the Enermax watercooling.

@groeck, any help or link to how to interpret these and other degrees reported is appreciated!

cemeyer, 2018-03-07:
@measly This ticket is not really the right forum for support requests.

That said, given the offset between nct6779-isa-0290:SMBUSMASTER and k10temp-pci-00cb:temp1, the former could easily be the unadjusted raw value (and thus, subtracting 27° yields real temperature).  Or it could be a bogus sensor — those are *very* common.  (My system reports "AUXTIN: -26.5°C" which is of course wrong.)

(If it is just the unadjusted raw sensor, 98° - 27° = 71°C, which is still quite high.  It's possible your CPU cooling solution is insufficient or improperly installed.)

[19 more replies]