lm-sensors/lm-sensors #17: Add list of risk factors to documentation

terrycloth, 2017-03-25:
The man page and interactive messages from the helper program `sensors-detect` mentions there being varying degree of risk --- with possibilities apparently ranging from a kernel panic to hardware damage.

However, I can't find any official documentation about what the risk factors are, or the likelihood of issues, for any of the `sensors-detect` operations. Users shouldn't have to roll the dice and just hope their computer won't melt or whatever. We need more specific information to make an educated decision --- something along the lines of "hardware having feature A, or lacking feature B, causes problems C..."

groeck, 2017-03-25:
Such information is not available because the exact risk factors are not known, much less the probability for problems to be seen. After all, there is no automated feedback mechanism which would tell us about any problems observed, or no problems observed. We can only _hope_ that people tell us. Since they won't tell us if there are no problems, there is no means to establish any sense of probability. The best we could possibly achieve might be something like a list with "On this computer, with this configuration, the following problem was observed". That would not exactly provide risk factors, but it would be a start. If someone is willing to spend the time going through the mailing list and the Internet and collect such information, I'll be more than happy to help making it available.

terrycloth, 2017-03-27:
Since the documentation website is gradually moving to the kernel.org subdomain, does that mean you could ask the Linux Foundation for resources to help test?

groeck, 2017-03-28:
No, that is not how it works. This would be akin of expecting Amazon to test code running on AWS, or a hosting service to test the hosted code [or web page(s)].

terrycloth, 2017-03-29:
Okay, I guess I'll start the research process then. I'll post reported issues here as I have the time to find them, and then maybe we can compile a table of the data or something...

First one is from [a thread on the Fedora "devel" mailing list](https://lists.fedoraproject.org/archives/list/devel@lists.fedoraproject.org/thread/VW3URNSYGRNBNNYP5YNWMFOJFSSZD66V/).

One user says lm_sensors caused a kernel panic, "particularly when probing the ISA bus on a 64-bit ARM system which does not have an ISA bus".

Another user says `sensors-detect` seems to be safe on x86-based Intel, *older* x86-based AMD, and on PCI/PCI-e cards, but risky for everything else. They give two recent examples with specific numbers from lm_sensors, falsely indicating a temperature of 0°C, or even *negative* temperatures, for two recent AMD Radeon devices. See the thread, linked above, for details (it's a short thread).

groeck, 2017-03-29:
The panic caused by sensors-detect is not a risk factor, it is a bug (obviously ;-), and a severe one for that matter.

"lm_sensors falsely indicated a temperature of 0°C, or even negative temperatures, for two recent AMD Radeon devices"

This has nothing to do with lm-sensors, or sensors-detect, or with the sensors command. It is a bug in the kernel driver(s) reading the data from the respective chips, or possibly a problem with the chips in question.

terrycloth, 2017-04-07:
The [Arch Linux wiki page about lm_sensors](https://wiki.archlinux.org/index.php/Lm_sensors#Sensors_not_working_since_Linux_2.6.31) references and comments on [an Arch Linux forum thread](https://bbs.archlinux.org/viewtopic.php?id=193048), talking about `sensors-detect` screwing up the gamma settings of laptop screens. The wiki says this is "caused by lm-sensors messing with the Vcom values of the screen while probing for sensors."

This issue was apparently reversible, but not until after a lot of trial and error in the long forum thread.

terrycloth, 2017-04-26:
I'm actually having a hard time finding anything else when I limit search results to the last 3 years, but maybe "hardware damage" is too vague. Should I be looking for more specific issues, like CPUs overheating or something?

groeck, 2018-08-17:
See README.

olysonek, 2018-08-21:
> See README.

No longer relevant. Reopening. I'll review this issue as soon as I can.