lm-sensors/lm-sensors #18: Json Output

shadethedaemon, 2017-03-26:
I was wondering if it would be possible to add an output mode for json for easier post processing.

groeck, 2017-03-26:
Sorry - an output more to what ? The sensors command ? To sensors-detect ?

shadethedaemon, 2017-03-27:
Sorry, I should have added more detail.
Currently sensors -u outputs, this output is easily read and useful. The only issue is it requires custom filters to read and process properly.
[image]

I propose an output for sensors, maybe -j, with a json output that would look like the following, though it it was without new line characters it would be fine.
{
   "acpitz-virtual-0":{
      "Adapter":"Virtual device",
      "temp1":{
         "temp1_input":48.000,
         "temp1_crit":128.000
      },
      "temp2":{
         "temp2_input":0.000,
         "temp2_crit":128.000
      },
      "temp3":{
         "temp3_input":42.000,
         "temp3_crit":128.000
      },
      "temp4":{
         "temp4_input":48.000,
         "temp4_crit":128.000
      },
      "temp5":{
         "temp5_input":30.000,
         "temp5_crit":128.000
      },
      "temp6":{
         "temp6_input":114.000,
         "temp6_crit":128.000
      }
   },
   "coretemp-isa-0000":{
      "Adapter":"ISA adapter",
      "Physical id 0":{
         "temp1_input":47.000,
         "temp1_max":84.000,
         "temp1_crit":100.000,
         "temp1_crit_alarm":0.000
      },
      "Core 0":{
         "temp1_input":46.000,
         "temp1_max":84.000,
         "temp1_crit":100.000,
         "temp1_crit_alarm":0.000
      },
      "Core 1":{
         "temp1_input":46.000,
         "temp1_max":84.000,
         "temp1_crit":100.000,
         "temp1_crit_alarm":0.000
      }
   }
}

groeck, 2017-03-28:
Sounds reasonable. Care to submit a patch ?

shadethedaemon, 2017-03-28:
I've created a pull request #19 for the functionality. This is one of the first C programs I have messed with, so please let me know if I did not do it in the most efficient way.

groeck, 2017-03-29:
Why two patches where the first is broken instead of one patch ?

shadethedaemon, 2017-03-29:
I'm used to other git projects where you make a commit when you stop work, i had considered the pull request the patch. I've created a new pull request #20 with the two commits merged.

shadethedaemon, 2017-04-03:
Sorry it took so long but pull request #21 has been added.

shadethedaemon, 2017-04-14:
Json added, closing issue.

closed 2017-04-14 by shadethedaemon

Alives, 2021-07-31:
There's a bug in this implementation.  When -f is specified, all output is converted to fahrenheit regardless of whether it's a temperature or not...

```
      "fan1":{
         "fan1_input": 1106.600,
         "fan1_min": 32.000,
         "fan1_alarm": 32.000,
         "fan1_beep": 32.000,
         "fan1_pulses": 35.600
      },
```
vs

```
      "fan1":{
         "fan1_input": 613.000,
         "fan1_min": 0.000,
         "fan1_alarm": 0.000,
         "fan1_beep": 0.000,
         "fan1_pulses": 2.000
      },
```

Filed https://github.com/lm-sensors/lm-sensors/issues/357