lm-sensors/lm-sensors #24: Fintek F71868A

Jigoku, 2017-06-19:
I have an MSI Gaming 970 motherboard, and have built lm-sensors from the master branch (where changelogs mention support for Fintek F71868A. 

Snippet from **sensors-detect**
```
Driver `f71882fg':
* ISA bus, address 0x485
Chip `Fintek F71868A Super IO Sensors' (confidence: 9) 
```

However; loading the module seems to have no effect, and **sensors** shows nothing new;
```
[root][0]>>[darkstar|~]
# uname -a
Linux darkstar 4.9.30 #1 SMP Fri May 26 14:54:17 CDT 2017 x86_64 AMD FX(tm)-8320 Eight-Core Processor AuthenticAMD GNU/Linux

[root][0]>>[darkstar|~]
# /etc/init.d/lm_sensors status
k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +28.6°C  (high = +70.0°C)
                       (crit = +80.0°C, hyst = +77.0°C)

fam15h_power-pci-00c4
Adapter: PCI adapter
power1:       63.81 W  (crit = 125.02 W)

[root][0]>>[darkstar|~]
# /etc/init.d/lm_sensors reload
Stopping lm_sensors:                                       [  OK  ]
Starting lm_sensors: loading module f71882fg               [  OK  ]

[root][0]>>[darkstar|~]
# lsmod  | grep f71882fg

[root][0]>>[darkstar|~]
# /etc/init.d/lm_sensors status
k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +39.6°C  (high = +70.0°C)
                       (crit = +80.0°C, hyst = +77.0°C)

fam15h_power-pci-00c4
Adapter: PCI adapter
power1:      114.68 W  (crit = 125.02 W)

[root][0]>>[darkstar|~]
#

```

Has anyone had any success or found a workaround? Is it possible i should be looking for and enabling any specific settings in my kernels config related to this module? I am running Slackware -current if this is any use.

Jigoku, 2017-06-19:
One thing i did notice is that the F71868A is not directly mentioned in the hwmon kernel config settings, [image]

groeck, 2017-06-19:
> I have an MSI Gaming 970 motherboard, and have built lm-sensors from the master branch (where changelogs mention support for Fintek F71868A.
>
> Snippet from **sensors-detect**
> ```
> Driver `f71882fg': [...]

Do you see anything in the kernel log ?

Guenter

groeck, 2017-06-19:
> I have an MSI Gaming 970 motherboard, and have built lm-sensors from the master branch (where changelogs mention support for Fintek F71868A. 

Please note that building and using lm-sensors does not affect
driver support in the kernel.

Guenter

Jigoku, 2017-06-19:
This appears in dmesg
```
[  258.510880] f71882fg: Found f71868a chip at 0x480, revision 48
[  258.510910] platform f71882fg.1152: failed to claim resource 0: [io  0x0480-0x0487]
[  258.510910] f71882fg: Device addition failed

```
I've also just upgraded my kernel to the latest mainline (4.12.0-rc5) just in case, but no changes from my previous kernel.

Jigoku, 2017-06-19:
Well i did a bit of digging, and found this on the old mailing list for lm-sensors;
https://marc.info/?l=lm-sensors&m=135178594809078&w=2

It mentions a workaround is to add **pnpacpi=off** to the kernel boot line. I adjusted my lilo.conf to do so, rebooted, and **sensors** now shows this:

```
[ricky][0]>>[darkstar|~]
$ sensors
f71868a-isa-0480
Adapter: ISA adapter
in0:          +1.67 V  
in1:          +0.98 V  
in2:          +1.49 V  
in3:          +0.94 V  
in4:          +1.10 V  
in5:          +1.25 V  
in6:          +1.23 V  
in7:          +1.67 V  
in8:          +1.65 V  
in9:          +1.67 V  
fan1:         831 RPM
fan2:           0 RPM  ALARM
fan3:           0 RPM  ALARM
temp1:        +34.0°C  (high = +85.0°C, hyst = +81.0°C)
                       (crit = +80.0°C, hyst = +76.0°C)  sensor = transistor
temp2:        +54.0°C  (high = +85.0°C, hyst = +81.0°C)
                       (crit = +98.0°C, hyst = +94.0°C)  sensor = thermistor
temp3:        +41.0°C  (high = +70.0°C, hyst = +68.0°C)
                       (crit = +85.0°C, hyst = +83.0°C)  sensor = transistor

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +32.1°C  (high = +70.0°C)
                       (crit = +80.0°C, hyst = +77.0°C)

fam15h_power-pci-00c4
Adapter: PCI adapter
power1:       32.29 W  (crit = 125.02 W)
```
So it would seem something to do with ACPI was taking the device before the f71868a/f71882fg module could load itself. I guess this is a solution? Could probably close this now?

groeck, 2017-06-19:
On Mon, Jun 19, 2017 at 08:41:59AM -0700, Ricky T wrote:
> Well i did a bit of digging, and found this on the old mailing list for lm-sensors;
> https://marc.info/?l=lm-sensors&m=135178594809078&w=2
>
> It mentions a workaround is to add **pnpacpi=off** to the [...]
Correct.

Guenter

closed 2017-06-19 by Jigoku