lm-sensors/lm-sensors #25: Found unknown chip with ID 0x2901

bhack, 2017-06-21:
Do you know what kind of SMSC chip is this?

groeck, 2017-06-21:
No idea, sorry.

closed 2017-06-21 by groeck

bhack, 2017-10-08:
Any solution for sensor-detect:
`Trying family 'SMSC'...                                     Yes
Found unknown chip with ID 0x2901`

groeck, 2017-10-08:
Even if sensors-detect would be able to detect the chip, there would still be no driver for it. So a "solution" for sensors-detect alone would not be helpful; all it does is to try to detect which chips are in the system (and then report that no driver is available for it).
Anyway, on top of that, the information provided is more than vague ... it isn't even possible to find information about the chip from the motherboard or system documentation because nothing was provided. Just the chip ID. Nothing else. I'll be happy to add the few lines of code to report the chip to sensors-detect, but someone would have to tell me what chip we are talking about in the first place.

bhack, 2017-10-08:
I've detected this on a Thinkpad T470s.

bhack, 2017-10-08:
A sort of datasheet [is available](http://psref.lenovo.com/syspool/Sys/PDF/ThinkPad/ThinkPad%20T470s/ThinkPad_T470s_Platform_Specifications.pdf)

c0d3z3r0, 2018-07-22:
Found on Lenovo x260

bhack, 2018-07-22:
Is it this http://www.ti.com/general/docs/lit/getliterature.tsp?baseLiteratureNumber=sles034&fileType=pdf?

c0d3z3r0, 2018-07-22:
@bhack no! The chip is SMSC

c0d3z3r0, 2018-07-22:
Yes, maybe...
Hm.. it provides Floppy, 2x Serial, 1x Parallel and IR I/O... the Lenovo x260 does not have any of these o.O

> Am 23.07.2018 um 00:29 schrieb bhack <notifications@github.com>:
> 
> SMSC

bhack, 2018-07-22:
Have you run https://coreboot.org/Superiotool on Lenovo x260?

bhack, 2018-07-23:
@groeck Do you think that superiotool output and that dataseht is enought?

bhack, 2018-07-23:
Yes I meant to report in sensors-detect.
> superiotool -d -e
No dump available for this Super I/O
No extra registers known for this chip.

groeck, 2018-07-23:
Try with isadump. sensors-detect tells you the IO base address (0x2e or 0x4e). Based on that, something like
`sudo isadump -k 0x55 0x2e 0x2f
`
should work (in this example, the address is 0x2e).

bhack, 2018-07-23:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 05 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 29 01 00 00 04 00 00 00 18 00 00 00 00 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

groeck, 2018-07-23:
This is not a SMSC FDC37N3869/FDC37N869. That chip has its device ID registers at 0x0d/0x0e (see datasheet, page 98). I suspect it is one of SMSC/Microchip's MEC170x embedded controllers. 
Unfortunately, I have no idea if it is possible to identify those chips; the datasheet does not provide the chip ID.

bhack, 2018-07-23:
So why superiotool is detecting it has `SMSC FDC37N3869/FDC37N869`?

bhack, 2018-07-23:
For MEC170X i see:
> The Device ID is visible as an 8-bit number at Plug and Play Configuration Index 20h.  
The HW Revision Number is visible as an 8-bit  number at Plug and Play  Configuration Index 21h

http://ww1.microchip.com/downloads/en/DeviceDoc/80000704D.pdf

groeck, 2018-07-23:
That would be 0x2d / 0x2e for MCE170x. Too bad, still no match. It might be some other chip from the MCE1xxx series, but at least for my part I did not find it. As for "why superiotool is detecting it has SMSC FDC37N3869/FDC37N869", you would have to ask the superiotool authors. This is not supported by the register dump and the chip datasheet.

bhack, 2018-07-23:
Mhh.. /cc @uwehermann any hint about this supertool detection?

bhack, 2018-07-23:
/cc also @pgeorgi Cause I don't know who is active actually on the tool.