lm-sensors/lm-sensors #27: Can't detect Motherboard sensor (ASRock Z97)

ilje, 2017-07-15:
Hi lm-sensors can't detect my motherboard sensors
/dev/port: No such file or directory
Machine:   Device: desktop Mobo: ASRock model: Z97 Pro3 serial: xxxxxxxxxx
           UEFI [Legacy]: American Megatrends v: P2.20 date: 05/27/2016

groeck, 2017-07-15:
Please provide output from "sudo sensors-detect". Also please provide your kernel version and distribution.

ilje, 2017-07-15:
Ok 
my log :

Operating System: Fedora 26 (Workstation Edition)
CPE OS Name: cpe:/o:fedoraproject:fedora:26
Kernel: Linux 4.11.9-300.fc26.x86_64
Architecture: x86-64

sensors detect

```
# sensors-detect revision $Revision$
# Board: ASRock Z97 Pro3
# Kernel: 4.11.9-300.fc26.x86_64 x86_64
# Processor: Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz (6/60/3)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): yes
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): yes
/dev/port: No such file or directory

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): yes
/dev/port: No such file or directory

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): yes
/dev/port: No such file or directory

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): yes
Found unknown SMBus adapter 8086:8ca2 at 0000:00:1f.3.
Sorry, no supported PCI bus adapters found.
Module i2c-dev loaded successfully.

Next adapter: SMBus I801 adapter at f000 (i2c-0)
Do you want to scan it? (YES/no/selectively): yes
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)

Next adapter: NVIDIA i2c adapter 0 at 1:00.0 (i2c-1)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 2 at 1:00.0 (i2c-2)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 6 at 1:00.0 (i2c-3)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 7 at 1:00.0 (i2c-4)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 8 at 1:00.0 (i2c-5)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 11 at 1:00.0 (i2c-6)
Do you want to scan it? (yes/NO/selectively): yes
Adapter cannot be probed, skipping.

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no): yes
Unloading i2c-dev... OK
```
```
sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +34.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +33.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:        +34.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:        +31.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:        +30.0°C  (high = +80.0°C, crit = +100.0°C)
```

groeck, 2017-07-15:
It appears that your kernel configuration may have CONFIG_DEVPORT disabled, or the /dev/port device node was explicitly removed. This is a matter of your system configuration, not of lm-sensors.

closed 2017-07-15 by groeck