lm-sensors/lm-sensors #29: New Release Checklist

shadethedaemon, 2017-07-28:
Can we put together  a todo list on what we need to do for a new release? The last release was 3.4.0 (2015-06-25).

gergesh, 2018-01-12:
I'd also suggest a new release, it's been a lot of time since the last one and support has been added for many newer systems.

groeck, 2018-01-14:
A new release would at least require an update of the outdated information, ie mostly all references to the old web page and to SVN. At this point I don't have time to do that. Contributions are more than welcome.

leighbb, 2018-03-15:
Hi Guenter,

I'm happy to have a go at copying the old lm-sensors.org wiki to somewhere new and then start updating it and the references to the old website in the source code.  Do you want to use the github wiki on this repository or hwmon.wiki.kernel.org?

Regards,

Leigh.

groeck, 2018-03-15:
We should use the kernel.org wiki. Go ahead!

groeck, 2018-03-15:
No idea what to do, sorry. It seems that I can create pages, but I don't know how to enable that for others. What happens when you try ? If I follow the lm-sensors link I can create the page by clicking on the pen on the right (which shows a pen with a plus-sign).

groeck, 2018-03-15:
As a workaround you can tag me here when you need new pages created and I can create them for you. I did that with the lm-sensors page a minute ago.

leighbb, 2018-03-15:
Okay, thanks @groeck 

Please add the following pages to get me started.  I've not indicated in the page names that they are part of lm_sensors.  Is that okay?

* https://hwmon.wiki.kernel.org/projectinformation
* https://hwmon.wiki.kernel.org/devices
* https://hwmon.wiki.kernel.org/installhowto
* https://hwmon.wiki.kernel.org/kernel2.6
* https://hwmon.wiki.kernel.org/documentation
* https://hwmon.wiki.kernel.org/feedbackandsupport
* https://hwmon.wiki.kernel.org/authorsandcontributors
* https://hwmon.wiki.kernel.org/configurations

Thanks!

leighbb, 2018-03-15:
By the way I definitely don't see a pen with plus sign :-(

groeck, 2018-03-16:
Most added. I changed the link to "Devices" because that page already exists  under a different name.

groeck, 2018-08-17:
No new release planned from this repository - see README.

olysonek, 2018-08-21:
> No new release planned from this repository - see README.

No longer relevant. Reopening. I'll see what needs to be done before a new release can be made and then make one.

olysonek, 2018-09-01:
I just fixed stale links and removed references to SVN. A new release is hopefully coming soon.

chrysophylax, 2018-10-30:
Any updates on the thought train here? Is this repo still the candidate for sourcing releases?

olysonek, 2018-10-30:
Yes, it is. I simply can't find the time to do a release right now. I need to finish the fix for https://github.com/lm-sensors/lm-sensors/issues/133 and review the changes since the last release. Then I'll make a release. No promises when that will be. The master branch should be safe to use however if you just want a newer version.

commit e8afbda1 (2018-11-23): 3.5.0 release

soname was bumped due to commit dcf23676cc264927 which introduced an ABI
change.

Fixes #29
closed 2018-11-23 by olysonek