lm-sensors/lm-sensors #30: lm-sensors does not detect integrated temperature sensor on Raspberry Pi

jonsnow1357, 2017-08-10:
I know i am running a slightly older version on my raspberry pi, but I believe this is still valid.

There is a integrated temperature sensor on th Raspberry Pi, that is not detected by lm-sensors.

[~]$ sudo sensors-detect 
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# Kernel: 4.9.35+ armv6l
# Processor: ARMv6-compatible processor rev 7 (v6l) (//)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
modprobe: FATAL: Module cpuid not found in directory /lib/modules/4.9.35+
Failed to load module cpuid.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Sorry, no supported PCI bus adapters found.
Module i2c-dev loaded successfully.

Sorry, no sensors were detected.
Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
http://www.lm-sensors.org/wiki/Devices for driver status.

You can access the sensor directly through /sys:

[~]$ cat /sys/class/thermal/thermal_zone0/type 
bcm2835_thermal
[~]$ cat /sys/class/thermal/thermal_zone0/temp 
45464
[~]$ awk '{printf("\nrPI Temperature = %.1f °C\n\n",$1/1e3)}' /sys/class/thermal/thermal_zone0/temp

rPI Temperature = 44.9 °C

chris001, 2017-08-25:
This test script detects CPU and GPU temps on raspberry pi.
This should be added to `lm-sensors` package so that `sensors-detect` finds these two RPi temp sensors.
```
#!/bin/bash
# Script: my-pi-temp.sh
# Purpose: Display the ARM CPU and GPU  temperature of Raspberry Pi 2/3 
# Author: Vivek Gite <www.cyberciti.biz> under GPL v2.x+
# -------------------------------------------------------
cpu=$(</sys/class/thermal/thermal_zone0/temp)
echo "$(date) @ $(hostname)"
echo "-------------------------------------------"
echo "GPU => $(/opt/vc/bin/vcgencmd measure_temp)"
echo "CPU => $((cpu/1000))'C"
```

groeck, 2017-08-27:
> This test script detects CPU and GPU temps on raspberry pi.
> This should be added to `lm-sensors` package so that `sensors-detect` finds these two RPi temp sensors.

Did you enable THERMAL_HWMON in your kernel configuration ?

Guenter

Jufo84, 2017-08-27:
> Did you enable THERMAL_HWMON in your kernel configuration ?
How to check on raspbian (Debian) ?

groeck, 2017-08-27:
Normally the configuration should be available in /boot and/or in /proc/config.gz. If enabled, the "sensors" command would show the temperature. For this to work, CONFIG_THERMAL and CONFIG_HWMON would both have to be configured as module or both would have to be built into the kernel.

Jufo84, 2017-08-28:
In the / boot directory I only see config.txt. Sensors-detect command does not detect sensors:
`# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# Kernel: 4.9.41+ armv6l
# Processor: ARMv6-compatible processor rev 7 (v6l) (//)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
modprobe: FATAL: Module cpuid not found in directory /lib/modules/4.9.41+
Failed to load module cpuid.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): YES
Sorry, no supported PCI bus adapters found.
Module i2c-dev loaded successfully.

Sorry, no sensors were detected.
Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
http://www.lm-sensors.org/wiki/Devices for driver status.`

groeck, 2017-08-28:
it seems to be important for you that sensors-detect finds the sensors, but not or not as much that they are actually supported by the hardware monitoring subsystem in the kernel. Is that correct ? If so, sorry, I won't be able to help you due to lack of time. You or someone else will have to provide a patch to add that support, and I'll be happy to accept it.
As for the kernel configuration, you mentioned that you have /boot/config.txt available. Did you check it for CONFIG_THERMAL and.or CONFIG_HWMON ?

chris001, 2017-08-28:
@groeck Thanks for your help.
It appears `CONFIG_THERMAL` exists in both Raspbian (Debian) kernel AND RPi Foundation kernel, since kernel 3.12 over two years ago.
See here for the `config` files used to build each kernel:
https://gist.github.com/pierre-haessig/3a925dfc6d0f7c540355
@Jufo84 
Here's detailed instructions for editing `/boot/config.txt` to change the raspberry pi kernel settings.  Maybe you need to go in there and see if the `THERMAL` and `HWMON` related settings are on or off.
http://elinux.org/RPiconfig

Jufo84, 2017-08-28:
These parmeters are not in http://elinux.org/RPiconfig - it is recommended to use: "This is the Raspberry Pi's temperature monitor, look at: /sys/class/thermal/thermal_zone0/temp"

chris001, 2017-08-28:
@Jufo84 
These might help to try:
https://wiki.archlinux.org/index.php/Raspberry_Pi#I2C
This thread:
http://forum.tinycorelinux.net/index.php/topic,19910.msg123618.html#msg123618

jonsnow1357, 2017-08-29:
Hi all,

If I understand correctly, the claim is that if CONFIG_THERMAL and/or CONFIG_HWMON are enabled for raspbian, then:
1) **sensors-detect** will not detect any temperature sensors
but
2) **sensors** will show the CPU temperature (GPU also, but I'm not realy interested in that)

Is my understanding correct?

Thanks

groeck, 2017-08-29:
sensors-detect and sensors are completely independent of each other; there is no correlation. CONFIG_THERMAL and CONFIG_HWMON only affects the output of the sensors command, but not sensors-detect.

chris001, 2017-08-29:
So the wikipedia page for lm-sensors says the new parent project of lm-sensors is hwmon:
https://hwmon.wiki.kernel.org/

The chip on the Raspberry Pi 3 is an ARM cpu made by Broadcom as a SoC (System On Chip) BCM2837.

The BCM2837 chip isn't listed on the HWMON list of supported chips ?!  How can this be possible.  The ARM CPU which the RPi runs on and is an evolution of, is the most popular and accessible CPU on Earth right now with something like 8 billion in use - phones, tablets, RPi, robots, automotive cars, airliner planes, drones, industrial factories, smart watches, they're everywhere.

groeck, 2017-08-29:
@chris001: The BCM2837 chip's thermal sensor is supported in the Linux kernel as thermal device, and the thermal-hwmon bridge is used to make it visible as hwmon device. SoC thermal sensors are often supported by a thermal subsystem driver since they are used to not only monitor but also control the CPU temperature. It does not make sense to have two drivers for the same sensor, and this works perfectly well.

Having said that, please keep in mind that this is not a commercial project. I am not getting paid to maintain the hardware monitoring subsystem, and I am not getting paid to provide support for lm-sensors either. If you have an expectation of such support, I would suggest to consider switching to a commercial operating system such as Windows. Alternatively, you might possibly consider getting actively involved instead of complaining about the real or perceived lack of features in Linux.

chris001, 2017-08-29:
* I found a patch to add support for Raspberry Pi sensors, to another very similar perl package, how they did it is they essentially just added that thermal zone search path to the list of search paths in the perl script.
Here it is, "Patch to add support for Raspberry Pi CPU Temp ":
https://bugs.launchpad.net/byobu/+bug/1587139
* Detailed step by step how one man worked hard and finally got `lm-sensors` working on an ARM based samsung Chromebook in Ubuntu
http://virtuallyhyper.com/2013/10/monitor-thermal-sensors-lm-sensors/

jonsnow1357, 2017-08-30:
ok, after having done some reading on hwmon, is it fair to say then that the problem is that the BCM does not implement the hwmon API.
It seems that **sensors** relies on hwmon to get sensor data.

I have the following on my rpi:
```
[pi@rpi01 02:49:54]
[~]$ sudo modprobe -v hwmon
[pi@rpi01 02:50:03]
[~]$ lsmod
Module                  Size  Used by
hwmon                  10533  0
i2c_bcm2708             5740  0
tun                    22606  2
spidev                  7034  0
cfg80211              525742  0
rfkill                 21373  2 cfg80211
snd_bcm2835            23131  0
snd_pcm                97825  1 snd_bcm2835
snd_timer              22706  1 snd_pcm
snd                    68784  3 snd_timer,snd_bcm2835,snd_pcm
i2c_bcm2835             6433  0
spi_bcm2835             7424  0
bcm2835_gpiomem         3791  0
uio_pdrv_genirq         3718  0
fixed                   3029  0
uio                    10166  1 uio_pdrv_genirq
i2c_dev                 6642  0
ip_tables              12512  0
x_tables               20921  1 ip_tables
ipv6                  384468  20
[pi@rpi01 02:50:08]
[~]$ ls /sys/class/hwmon/
[pi@rpi01 02:50:19]
[~]$ 
```

which I think means that there is CONFIG_HWMON support in the kernel, but there's nothing available in /sys/class/hwmon

@groeck 👍 could you please give some info about this _thermal-hwmon bridge_ you mentioned. Thanks

groeck, 2017-08-30:
The thermal-hwmon bridge only works if THERMAL and HWMON are both configured to be built as module, or if both are configured to be built into the kernel. Based on the above, it appears that THERMAL is built into the kernel, and HWMON is built as module. With this configuration, the bridge does not work since code built into the kernel can not call functions declared in a module. On top of that, it is necessary to enable THERMAL_HWMON.

chris001, 2017-08-30:
@groeck Consequently, would the best solution to this issue be, to request to RPi Foundation and/or to Raspbian, to default configure THERMAL and HWMON to both be built into the kernel, and enable THERMAL_HWMON by default, and release that built kernel as soon as possible as a hotfix, so that the hwmon-thermal bridge will start to work again..?

groeck, 2017-08-30:
@chris001: Yes, if the RPi Foundation is interested in reporting temperatures through the hwmon subsystem. Otherwise I'd suggest to make the necessary changes in your local configuration.

rkarlsba, 2017-12-18:
It'd be rather nice to have the system temperature reported in a standardised way so that 'sensors' show it and things like sar -m TEMP, munin, whatever also does so. Doesn't seem to be the case today. Interestingly, with an orange pi with armbian, this works flawlessly.

groeck, 2018-02-13:
The problem is not an lm-sensors problem but a system configuration problem, and a problem in the interaction between the thermal and hwmon subsystems in the kernel. Nothing to do here; closing.

closed 2018-02-13 by groeck