lm-sensors/lm-sensors #34: ga-970a-d3p

pr0dukter, 2017-09-12:
cant figure out how to contact you as a user so ima copypasta this sensors.conf i made taking averages from the net as suggested.
please forgive the issue, but i know how hard these are to come by so i figure maybe it will be appreciated especially the gigabytes are motherf*ckers.
feel free to clean it up, i had to teach myself to do this so theres different syntax for same result.  still need to fix temp3 but who knows on that, it will take a better man than i.
ill check back in couple days to close this.

this is with modprobing it87 force_id=0x8728

drop in sensors.d
file: ga-970a-d3p.conf
=========start=======

chip "it8728-isa-0228"
    label  in0  "VCore"    # CPU Core Voltage
    label  in1  "VDDR"     # DDR2 1.8V
    label  in2  "+3.3V"    # VCC3
    label  in3  "+5V"      # VCC
    label  in4  "+12V"
    label  in5  "-12V"     # Not in the BIOS, maybe not really -12V
    label  in6  "-5V"      # Not in the BIOS, maybe not really -5V
    label  in7  "3.3VSB"     # VCCH
    label  in8  "VBat"

    label  fan1  "CPU Fan"
    label  fan2  "PWR Fan"
    label  fan3  "SYS Fan"

    label temp1  "NB Temp"
    label temp2  "CPU Temp"

    compute  in2  @ * (178/120), @ / (178/120)
    compute  in3  @ * (294/120), @ / (294/120)
    compute  in4  @ * (728/120), @ / (728/120)
    compute  in5 -@ * ((520/60) +1), -@ / ((520/60) +1)
    compute  in6 -@ * ((162/120) +1), -@ / ((162/120) + 1)

    #compute in6 (1+294/120)*@ - 4.096*294/120, (@ + 4.096*294/120)/(1+294/120)

   set in0_min  0.7
   set in0_max  1.5
   set in1_min  1.5   * 0.95
   set in1_max  1.5   * 1.05

   set in2_min 3.3 * 0.95
   set in2_max 3.3 * 1.05
   set in3_min 5.0 * 0.95
   set in3_max 5.0 * 1.05
   set in4_min 12 * 0.95
   set in4_max 12 * 1.05
   set in5_min (-1) * (12 * 0.95)
   set in5_max (-1) * (12 * 1.05)
   set in6_min (-5) * 0.95
   set in6_max (-5) * 1.05
   set in7_min  3     * 0.95
   set in7_max  3     * 1.05

   =========end=======

pr0dukter, 2017-09-13:
soon as this commented seen ill close it

groeck, 2017-09-13:
What driver are you using ? The driver published here should not require force-id.
Side note: Please refrain from bashing vendors here.

pr0dukter, 2017-09-14:
it87, wasnt trying to bash a vender with untrue slander, gigabyte fan control is notoriously underdocumented and supported for linux. thats just a fact.

closed 2017-09-14 by pr0dukter

groeck, 2017-09-14:
@pr0dukter: You are correct, Gigabyte does not support Linux, and that is well known. Just like other motherboard vendors do not support Linux either, which is also well known. That does not give you rhe right to curse at any of them here. You can make statements of fact, but please take the bashing elsewhere.
Besides, even though Gigabyte does not support Linux, they are more responsive to my inquiries than other motherboard vendors who often don't even reply when I ask for support. People like me (Linux kernel maintainers) depend on any support we can get. Bashing them does not help me, and at the end it does not help you either. Look at it from their side (ie the side of the engineers who are paid to provide support, not the managers making the policy): Why would they ever want to help you in the future ?

vagrale13, 2017-11-16:
Here the _"ga-970a-ud3p.conf"_ for **Gigabyte GA-970A-UD3P** motherboard 
if someone have the same motherboard and have problems with temperatures results.
Thanks @pr0dukter your file helped me very much.

```
chip "it8728-isa-0228"

# GA-970A-UD3P (rev. 1.0)

    label  in0  "VCore"    # CPU Core Voltage
    label  in1  "VDDR"     # DDR3 1.5V
    label  in2  "+3.3V"    # VCC3
    label  in3  "+5V"      # VCC
    label  in4  "+12V"
    label  in5  "-12V"     # Not in the BIOS, maybe not really -12V
    label  in6  "-5V"      # Not in the BIOS, maybe not really -5V
    label  in7  "3.3VSB"   # VCCH also seen STANDBY -from russian site
    label  in8  "VBat"
    label  fan1  "PWR Fan"
    label  fan2  "1st SYS Fan"
    label  fan3  "2nd SYS Fan"
    label  fan5  "CPU Fan"
    label temp1  "NB Temp"
    label temp2  "CPU Temp"

    compute  in2  @ * (192/120), @ / (192/120)
    compute  in3  @ * (300/120), @ / (300/120)
    compute  in4  @ * (737/120), @ / (737/120)
    compute  in5 -@ * ((520/60) +1), -@ / ((520/60) +1)
    compute  in6 -@ * ((162/120) +1), -@ / ((162/120) + 1)
#compute in6 (1+294/120)*@ - 4.096*294/120, (@ + 4.096*294/120)/(1+294/120)

    set in0_min 0.7
    set in0_max 1.5
    set in1_min 1.5 * 0.95
    set in1_max 1.5 * 1.05
    set in2_min 3.3 * 0.95
    set in2_max 3.3 * 1.05
    set in3_min 5.0 * 0.95
    set in3_max 5.0 * 1.05
    set in4_min  12 * 0.95
    set in4_max  12 * 1.05
    set in5_min (-1) * (12 * 0.95)
    set in5_max (-1) * (12 * 1.05)
    set in6_min (-5) * 0.95
    set in6_max (-5) * 1.05
    set in7_min  3  * 0.95
    set in7_max  3  * 1.05

```

akontsevich, 2020-01-05:
Where to take config for Gigabyte 970A-DS3P? Thanks!

0rbadvent, 2020-01-05:
I dont have this board anymore but /etc/sensors.conf

Sent from ProtonMail mobile

-------- Original Message --------

akontsevich, 2020-01-05:
> I dont have this board anymore but /etc/sensors.conf 

I see, so need to add it8620-isa-0228 chip to `/etc/sensors.conf`? And if it is not there?

0rbadvent, 2020-01-05:
No add the values from my post. The sensor math?

Sent from ProtonMail mobile

-------- Original Message --------

akontsevich, 2020-01-05:
> No add the values from my post. The sensor math? 

Values from Your post? Which one? What if they do not match to my MB: seems I have other set of sensors? I have this output:
```

> sensors
k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +11.6°C  (high = +70.0°C)
                       (crit = +80.0°C, hyst = +77.0°C)

nouveau-pci-0100
Adapter: PCI adapter
GPU core:    900.00 mV (min =  +0.88 V, max =  +1.08 V)
fan1:        1530 RPM
temp1:        +35.0°C  (high = +95.0°C, hyst =  +3.0°C)
                       (crit = +105.0°C, hyst =  +5.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)

it8620-isa-0228
Adapter: ISA adapter
in0:         876.00 mV (min =  +0.00 V, max =  +3.06 V)
in1:           1.46 V  (min =  +0.00 V, max =  +3.06 V)
in2:           1.99 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in5:           1.16 V  (min =  +0.00 V, max =  +3.06 V)
in6:           2.23 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.12 V
fan1:        1149 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +32.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +31.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermal diode
temp3:        +11.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp4:        +45.0°C
temp5:        +45.0°C
temp6:        +45.0°C
intrusion0:  ALARM

fam15h_power-pci-00c4
Adapter: PCI adapter
power1:       48.36 W  (crit = 125.19 W)
```