lm-sensors/lm-sensors #35: FR: option to report fan speeds as percent of max

ITwrx, 2017-09-15:
right now fan speeds are reported as RPMs. it would be nice to have the option to display fan speeds as percent of their max RPMs instead. i don't know if the chip will report the max rpm to lm-sensors or not. If not, this FR may be out of scope.
thanks

groeck, 2017-09-15:
On Fri, Sep 15, 2017 at 07:10:04PM +0000, ITwrx wrote:
> right now fan speeds are reported as RPMs. it would be nice to have the option to display fan speeds as percent of their max RPMs instead. i don't know if the chip will report the max rpm to lm-sensors or not. If not, this FR may be out of scope.

If a fan controller supports it, there will be a respective property,
and userspace can do the calculation. On top of that, I am not aware
of a chip reporting its speed as percentage of a given maximum.
Given that, yes, this feature request is out of scope.

ITwrx, 2017-09-15:
"If a fan controller supports it, there will be a respective property, and userspace **can** do the calculation."

so are you saying that if the controller supports it lm-sensors **will** do the calculation?
thanks

groeck, 2017-09-16:
On 09/15/2017 03:53 PM, ITwrx wrote:
> "If a fan controller supports it, there will be a respective property, and userspace *can* do the calculation."
> 
> so are you saying that if the controller supports it lm-sensors *will* do the calculation?
> thanks
> 

No. Maybe you need to be more specific with your requests. I _think_ what you are saying is that
you would like the "sensors" program to report the fan speed as percentage of the maximum. I had
assumed that you want the kernel to report it.

For the "sensors" command, I don't really see the benefit, and it does not fit well into
the mechanism of information reported by the command. I'll be open minded to review patches,
though.

Guenter

ITwrx, 2017-09-16:
yes, i'm talking about "sensors". i think most people would say that they would use percent of max more often than RPMs, but i'm sure almost all would like to have the option available. I'm just a lowly web/php dev, just now looking into c/c++, so i probably won't be submitting a patch any time soon. :) I'll leave it to you to close this FR if you choose.
have a nice day.

groeck, 2018-01-17:
I am open to reviewing a patch or patch series to add a respective parameter to the sensors command. Otherwise this won't be addressed.

closed 2018-01-17 by groeck