lm-sensors/lm-sensors #36: Asus Z97-P, no fan speed, no voltage

promeneur, 2017-09-28:
opensuse 42.2, kernel 4.487-18-29
lisensor4 3.4.0

result of sensors
**************************************************************************************************
acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +105.0°C)
temp2:        +29.8°C  (crit = +105.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +33.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:         +31.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:         +33.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:         +33.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:         +29.0°C  (high = +80.0°C, crit = +100.0°C)
*********************************************************************************************************************
result of sensors-detect:
*******************************************************************************************************************
linux-bmy3:~ # sensors-detect
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: ASUS All Series
# Board: ASUSTeK COMPUTER INC. Z97-P
# Kernel: 4.4.87-18.29-default x86_64
# Processor: Intel(R) Core(TM) i5-4590 CPU @ 3.30GHz (6/60/3)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8606
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:8ca2 at 0000:00:1f.3.
Sorry, no supported PCI bus adapters found.
Module i2c-dev loaded successfully.

Next adapter: i915 gmbus ssc (i2c-0)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus vga (i2c-1)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus panel (i2c-2)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpc (i2c-3)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpb (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpd (i2c-5)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: SMBus I801 adapter at f040 (i2c-6)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x51
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no): no
To load everything that is needed, add this to one of the system
initialization scripts (e.g. /etc/rc.d/rc.local):

#----cut here----
# Chip drivers
modprobe coretemp
/usr/bin/sensors -s
#----cut here----

You really should try these commands right now to make sure everything
is working properly. Monitoring programs won't work until the needed
modules are loaded.

Unloading i2c-dev... OK
Unloading cpuid... OK

linux-bmy3:~ # 
*******************************************************************************************************

we see this
...
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8606
    (logical device 4 has address 0x290, could be sensors)
...

don't hesitate to ask for me more info

groeck, 2017-09-28:
See issue #14, it87 driver. I don't have a datasheet, and without it is all but impossible to add support for the chip.

closed 2017-09-28 by groeck

promeneur, 2017-09-28:
hello

i don't see anything in #14 about my problem or it87

groeck, 2017-09-28:
https://github.com/groeck/it87/issues/14 subject  "Add support for IT8606E and IT8626E", states: 
"IT8606E is known to be used on Asus Z97-P. Unknown where IT8626E is used. Information and register dumps would be appreciated."
I am not sure how could describe the connection better. Please let m know what additional information you would like to see.