lm-sensors/lm-sensors #37: [v3.3.5]: Values set by custom config file in /etc/sensors.d/ are ignored by "sensors -s"

jleveque, 2017-10-26:
I are using a custom config file (`/etc/sensors.d/sensors.conf`), however, when I call `sensors -s`, the values from my custom config file do not get set. Instead, the default values from `/etc/sensors3.conf` are used.

However, if I explicitly tell sensors to load the custom file (`sensors -s -c /etc/sensors.d/sensors.conf`), the custom values are loaded properly.

Examining the basic `sensors -s` call using strace appears to show the config files getting loaded in the proper order, so it doesn't appear as though the custom values are getting overwritten by the defaults. Instead, it seems they are getting ignored:

```
open("/etc/sensors3.conf", O_RDONLY)    = 3
ioctl(3, SNDCTL_TMR_TIMEBASE or SNDRV_TIMER_IOCTL_NEXT_DEVICE or TCGETS, 0x7ffd70c2ef80) = -1 ENOTTY (Inappropriate ioctl for device)
fstat(3, {st_mode=S_IFREG|0644, st_size=10344, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fd940d57000
read(3, "# libsensors configuration file\n"..., 8192) = 8192
read(3, " label in4 \"+12V\"\n    label in5 "..., 8192) = 2152
read(3, "", 4096)                       = 0
read(3, "", 8192)                       = 0
ioctl(3, SNDCTL_TMR_TIMEBASE or SNDRV_TIMER_IOCTL_NEXT_DEVICE or TCGETS, 0x7ffd70c2d910) = -1 ENOTTY (Inappropriate ioctl for device)
close(3)                                = 0
munmap(0x7fd940d57000, 4096)            = 0
openat(AT_FDCWD, "/etc/sensors.d", O_RDONLY|O_NONBLOCK|O_DIRECTORY|O_CLOEXEC) = 3
getdents(3, /* 4 entries */, 32768)     = 112
getdents(3, /* 0 entries */, 32768)     = 0
close(3)                                = 0
stat("/etc/sensors.d/sensors.conf", {st_mode=S_IFREG|0644, st_size=1449, ...}) = 0
open("/etc/sensors.d/sensors.conf", O_RDONLY) = 3
ioctl(3, SNDCTL_TMR_TIMEBASE or SNDRV_TIMER_IOCTL_NEXT_DEVICE or TCGETS, 0x7ffd70c2de70) = -1 ENOTTY (Inappropriate ioctl for device)
fstat(3, {st_mode=S_IFREG|0644, st_size=1449, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fd940d57000
read(3, "# libsensors configuration file\n"..., 8192) = 1449
read(3, "", 4096)                       = 0
read(3, "", 8192)                       = 0
ioctl(3, SNDCTL_TMR_TIMEBASE or SNDRV_TIMER_IOCTL_NEXT_DEVICE or TCGETS, 0x7ffd70c2c800) = -1 ENOTTY (Inappropriate ioctl for device)
close(3)                                = 0
munmap(0x7fd940d57000, 4096)            = 0
```
I didn't see any closed issues that appear to address this, so I thought I'd bring it to your attention. Please let me know if I can be of further assistance.

travisszucs, 2017-12-11:
It sounds like the SMBus names in sensors.conf doesn't match the actual bus names used by the sensors. This will cause the set calls to be skipped in lm-sensors since it can't find a matched bus name.

Please double check that the specified bus names in sensors.conf matches the bus names given by running 'sensors'.

For example if sensors gives:
```
max6697-i2c-2-1a
Adapter: **SCD 0000:02:00.0 SMBus master 0 bus 0**
Board sensor:              +38.0 C  (high = +95.0 C)
Switch chip left sensor:   +41.2 C  (high = +95.0 C, crit = +105.0 C)
Switch chip right sensor:  +40.0 C  (high = +95.0 C, crit = +105.0 C)
Front panel temp sensor:   +34.0 C  (high = +65.0 C, crit = +75.0 C)
```

Then the configuration should have
```
bus "i2c-2" "**SCD 0000:02:00.0 SMBus master 0 bus 0**"
```

groeck, 2018-02-13:
Unfortunately the submitter did not provide the configuration files, so it is all but impossible to determine what went wrong. I can not reproduce the problem with sensors v3.4.0; it works as expected. Closing.

closed 2018-02-13 by groeck