lm-sensors/lm-sensors #38: NCT6793D Fan info not being displayed

ttocskcaj, 2017-11-07:
I've been playing around for a while to get this chip working. Compiling the latest versions of the ntc6775 and lm-sensors worked to get the chip detected correctly in `sensors-detect`, but `sensors` isn't showing any values from the chip.

I'm getting the message
```
[ 1027.458977] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[ 1681.422655] nct6775: Found NCT6793D or compatible chip at 0x2e:0x290
[ 1681.422659] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000290-0x0000000000000299 (\_GPE.HWM) (20170531/utaddress-247)
```

Does `acpi_enforce_resources=lax` need to be set to display these values? Or should it be working without?

groeck, 2017-11-07:
On Tue, Nov 07, 2017 at 08:35:18PM +0000, Jack Scott wrote:
> I've been playing around for a while to get this chip working. Compiling the latest versions of the ntc6775 and lm-sensors worked to get the chip detected correctly in `sensors-detect`, but `sensors` isn't showing any values from the chip.
> 

I still wonder why people believe that the output of "sensors-detect"
has anything to do with Linux kernel support.

> I'm getting the message
> ```
> [ 1027.458977] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
> [ 1681.422655] nct6775: Found NCT6793D or compatible [...]
Well, here you have the problem.

> Does `acpi_enforce_resources=lax` need to be set to display these values? Or should it be working without?
> 
Yes, it does have to be set. At your own risk, since the BIOS may access
the chip in parallel.

Guenter

ttocskcaj, 2017-11-07:
Cool, thanks for clarifying.

Like most people, I'm hesitant to set `acpi_enforce_resources=lax`
Since this seems to be a fairly common issue, do you know if anyone is working on ACPI drivers for the fan controls?

groeck, 2017-11-07:
On Tue, Nov 07, 2017 at 09:12:03PM +0000, Jack Scott wrote:
> Cool, thanks for clarifying.
> 
> Like most people, I'm hesitant to set `acpi_enforce_resources=lax`
> Since this seems to be a fairly common issue, do you know if anyone is working on ACPI drivers for the fan controls? 
> 
No one ever will. There is no standard for such a driver, meaning a new driver
would almost certainly be needed for each new board. Board vendors don't release
the ACPI methods they use, meaning it would be necessary to analyze the DSDT.
And the DSDT changes from board to board, even for boards from the same vendor.
Worse, useful information is often not exported in the first place.
Especially in recent boards, the chip isn't even accessed by the BIOS but
by some EC (enbedded controller) on the board, and the method to talk to that
EC is not published.

In short, it is hopeless.

Guenter