lm-sensors/lm-sensors #40: Fan problem after running sensors-detect

aurimus, 2017-11-29:
I ran `sensors-detect` and ever since the fan screams like crazy (100% all the time, even if the temperatures are very low). The only thing that works is to pull the charging out, then it assumes low power mode and the fan completely stops! (starts if something heavy is run, but again only 100% or 0)

I am not 100% sure that it's the `sensors-detect` problem, but I answered yes to all the questions (even the ones that had NO as default).

Any way to fix this?

    Laptop: Samsung Series 9 NP900x3c
    OS: Ubuntu 17.04

groeck, 2017-11-29:
> I ran |sensors-detect| and ever since the fan screams like crazy (100% all the time, even if the temperatures are very low). The only thing that works is to pull the charging out, then it assumes [...]

There is a reason for some of the questions to default to NO,
especially for some of the i2c interfaces. Reason is that there are i2c devices,
especially those connected to graphics cards, which interpret i2c read requests
as commands, with weird results. This is why sensors-detect warns about it.

> Any way to fix this?

Even after a cold reboot ? That would be bad. And I have no idea how to fix it, sorry.

Guenter

aurimus, 2017-11-29:
I have just tested with the other (identical) fan in this slot and it works as expected. Would it be possible that `sensors-detect` broke the fan somehow? (some kind of a feedback mechanism that should be in the fan is no more, and that's why it goes 0 or 100% only)

Thanks

groeck, 2017-11-29:
Never heard about anything like that. I could imagine that the fancontrol script might have such impact, but sensors-detect is unlikely. Having said that, sensors-detect could have "directed" some embedded controller to issue commands to the fan which made it blow up. But even that is very unlikely.

closed 2017-11-29 by aurimus