lm-sensors/lm-sensors #41: AMDGPU DM i2c hw - Slow scan & Found unknown SMBus adapter

denji, 2017-12-12:
```
$ sensors --version
sensors version 3.4.0 with libsensors version 3.4.0
$  uname -r
4.15.0-041500rc3-generic
$ sudo sensors-detect
...
Next adapter: AMDGPU DM i2c hw bus 0 (i2c-0)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: dmdc (i2c-1)
Do you want to scan it? (yes/NO/selectively): yes
.....
Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): yes
Found unknown SMBus adapter 8086:a123 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.

Next adapter: AMDGPU DM i2c hw bus 3 (i2c-6)
Do you want to scan it? (yes/NO/selectively): yes
Client found at address 0x4a
Probing for `National Semiconductor LM75'...                No
Probing for `National Semiconductor LM75A'...               No
Probing for `Dallas Semiconductor DS75'...                  No
Probing for `National Semiconductor LM77'...                No
Probing for `Analog Devices ADT7410/ADT7420'...             No
Probing for `Analog Devices ADT7411'...                     No
Probing for `Maxim MAX6642'...                              No
Probing for `Texas Instruments TMP435'...                   No
Probing for `National Semiconductor LM73'...                No
Probing for `National Semiconductor LM92'...                No
Probing for `National Semiconductor LM76'...                No
Probing for `Maxim MAX6633/MAX6634/MAX6635'...              No
Probing for `NXP/Philips SA56004'...                        No
Client found at address 0x4b
Probing for `National Semiconductor LM75'...                No
Probing for `National Semiconductor LM75A'...               No
Probing for `Dallas Semiconductor DS75'...                  No
Probing for `National Semiconductor LM77'...                No
Probing for `Analog Devices ADT7410/ADT7420'...             No
Probing for `Analog Devices ADT7411'...                     No
Probing for `Maxim MAX6642'...                              No
Probing for `Texas Instruments TMP435'...                   No
Probing for `National Semiconductor LM92'...                No
Probing for `National Semiconductor LM76'...                No
Probing for `Maxim MAX6633/MAX6634/MAX6635'...              No
Probing for `NXP/Philips SA56004'...                        No
Probing for `Analog Devices ADT7481'...                     No

Next adapter: AMDGPU DM i2c hw bus 4 (i2c-7)
Do you want to scan it? (yes/NO/selectively): yes
...
```

groeck, 2017-12-12:
GPU i2C controllers are notoriously slow and dangerous to scan. I would not recommend it.
Try again for the rest with the version of the script here.

groeck, 2017-12-15:
AFACIS there is nothing we can do about the slowness problem; this is a driver and hardware problem, not a lm-sensors problem. The unknown adapter problem should be resolved.

closed 2017-12-15 by groeck