lm-sensors/lm-sensors #43: CPU Ryzen 1700 temperature monitoring lm_sensors & Conky in Linux Mint Cinnamon 18.3 x64

bruswlad, 2017-12-21:
My computer: motherboard MSI B350 TOMAHAWK, processor AMD Ryzen 7 1700 Eight-Core, 16Gb RAM, graphics card NVIDIA GeForce GT 730 with 2Gb memory and solid state drive 250Gb. The operating system Linux Mint Cinnamon 18.3 x64 rus is installed.
Full details at: https://linux-hardware.org/index.php?probe=a46305d3d4
It is necessary to configure monitoring of the CPU temperature in the Conky program. This program I chose because of its beauty and convenience of customization.
Screenshot at the end of the publication.
On older computers, for example, with the Atom D525 processor, this task is solved by installing the lm-sensors utility. The sensors command displays a list of sensors, including temperature sensors:

AtomSSD ~ # sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:       +60.0°C  (crit = +256.0°C)
temp2:       +53.0°C  (crit = +110.0°C)
temp3:       +48.0°C  (crit = +105.0°C)
temp4:       +27.3°C  (crit = +110.0°C)
temp5:       +55.0°C  (crit = +110.0°C)
coretemp-isa-0000
Adapter: ISA adapter
Core 0:      +54.0°C  (crit = +100.0°C)
Core 1:      +49.0°C  (crit = +100.0°C)

The temp1-5 sensors do not interest us. The sensor of the processor is defined as Core 0 (for dual-core processors, Core 1 and so on should be defined as well). It remains only to correct the configuration file Conky (in my case it is "conky lmc-18-eng"), located in the home directory /home/wlad/.conky.
In this file we specify the command "execi", the measurement period "10" in seconds and the "sensors" command, from which we select "grep", the name of the string "Core 0" and the selection of symbols from it "cut -c14-20":

$Температура CPU: ${execi 10 sensors | grep 'Core 0' | cut -c14-20}

The display shows: "CPU temperature: + 54.0 ° C" (naturally, without quotes).
By the way, in the picture is a screenshot of the new computer. The temperature is displayed by the string Conky:

$ Processor temperature: + 54.0 ° C

Because the sensors command does not see the sensors:

RyzenSSD ~ # sensors
No sensors found!
Make sure you have loaded all the kernel drivers you need.
Try sensors-detect to find out which these are.

That is the problem. Search the Internet did not give anything.
In Windows 10 professional x64 rus, the processor temperature output is without problems, for example, in the AIDA64 Business program.
I ask you to solve this problem, please write the result, including by e-mail: wladbrus@gmail.com

groeck, 2017-12-21:
amd family 17h temperature sensors are supported with Linux 4.15+. ToT lm-sensors detects those sensors.

closed 2017-12-21 by groeck

bruswlad, 2017-12-21:
What do you mean by "Linux 4.15+"
Complete information about my computer and operating system:
https://linux-hardware.org/index.php?probe=a46305d3d4

groeck, 2017-12-21:
I mean exactly what I said. The Linux kernel will support AMD Ryzen temperature sensors starting with Linux kernel version 4.15 and later. You are running a 4.14 kernel which does not and will not support it. Also, sensors-detect ToT from here does detect the sensors.

bruswlad, 2017-12-21:
Many thanks!
Got it