lm-sensors/lm-sensors #44: Support for Z370 motherboards (i.e. Gigabyte Technology Co., Ltd. Z370 AORUS Ultra Gaming)

Emanem, 2017-12-27:
Hi,

Not sure if this chipset is supported yet, would like to understand and potentially debug/expand sensors script - where would you suggest to start?

Thanks!

groeck, 2017-12-27:
You wouldn't, really. sensors-detect normally does not have specific support for chipsets. You might run sensors-detect and check if anything interesting is reported.

Emanem, 2017-12-28:
I'm using a version of sensors-detect rather old (2015 - included with Ubuntu 16.04). Does this still apply?
I've run it, and it can see the right mobo name, but only get the *coretemp* CPU monitor...

groeck, 2017-12-28:
What else does it report ?

Emanem, 2017-12-29:
Copied and pasted the output from terminal - let me know if you need more info!
Thanks again!

```
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Gigabyte Technology Co., Ltd. Z370 AORUS Ultra Gaming [Default string]
# Board: Gigabyte Technology Co., Ltd. Z370 AORUS Ultra Gaming-CF
# Kernel: 4.10.0-42-generic x86_64
# Processor: Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz (6/158/10)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8686
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:a2a3 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.

Next adapter: NVIDIA i2c adapter 1 at 1:00.0 (i2c-0)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 2 at 1:00.0 (i2c-1)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 4 at 1:00.0 (i2c-2)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 6 at 1:00.0 (i2c-3)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 7 at 1:00.0 (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 8 at 1:00.0 (i2c-5)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 9 at 1:00.0 (i2c-6)
Do you want to scan it? (yes/NO/selectively): 

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

To load everything that is needed, add this to /etc/modules:
#----cut here----
# Chip drivers
coretemp
#----cut here----
If you have some drivers built into your kernel, the list above will
contain too many modules. Skip the appropriate ones!

Do you want to add these lines automatically to /etc/modules? (yes/NO)

Unloading cpuid... OK

```

groeck, 2017-12-29:
The it87 driver from here supports the Super-IO chip. The version of sensors-detect available from here should also give you better results. I don't immediately see anything else that would need to be done.

Emanem, 2017-12-30:
That would be great - do you know if there's a Ubuntu repo with latest version of *lm_sensors* which I could use to install it?

Or is it just about the perl script *sensors-detect*?

groeck, 2017-12-30:
- You'll have to download and install lm-sensors and it87 from here. For lm-sensors, just download it and run prog/sensors/sensors-detect. Not that it makes a difference, really. See README in the it87 driver source.
- If you read the _entire_ source of it87.c, you may notice that the listed motherboards are only motherboards which need special treatment.
- Installing a driver in /etc/modules won't help. You'll have to download the source and run "make; make install".
- The latest version of it87.c isn't upstream yet.

EdiDD, 2017-12-30:
Will it land in near future in debian's (testing) repository or it is necessary to compile from source ?

groeck, 2017-12-30:
"Future" is relative. If I ever find the time to do it, and if I find someone to review the code, it may be available upstream at some point. I would not bet on it, though.

EdiDD, 2017-12-30:
Ok I assume Nuvoton nct67930 will work, probably with nct6775 driver

Emanem, 2017-12-30:
Hi @groeck as per my edits I've done a bit of digging:

* indeed realized list motherboards are only needed for specialized treatment.
* as you're saying, old *lm_sensors* is not really the problem here, but the old *it87* driver is.
* of course just enabling the driver in `/etc/modules` won't help because I have an older version, we would be left with `make` and `make install` the sources ourselves

It is quite sad that various distros/maintainers (Ubuntu/Debian/...) won't include an updated version of this driver.
At the end of the day Linux is becoming more and more used and better hardware support should be a _default_ priority for distros/maintainers.

Thanks again for your help and responses - I may raise a bug in Ubuntu to ensure this gets included in next `hwe` releases...

Let me know if you want me to close this ticket - I think you've done more than enough...

groeck, 2017-12-30:
That isn't a matter of Ubuntu/debian maintainers. It is a matter of the driver author (me) to send the changes upstream. I am not getting paid for this work. I don't have the time to clean up the code, split it into one-per-logical-change patches, and send it upstream. Finding someone to review the patches would be even more challenging. Many people want to see the code upstream, almost no one is willing to invest actual time into it. Really, I must say that I am always amazed to see the demands people make of open source code, without willingness to actually pay for it.
Sorry, you really caught me on the wrong foot (as many people do nowadays). At some point I may just pull everything from github.

groeck, 2017-12-30:
nct67930 may work if it is NCT6793D. I have not heard of that chip, though, so I don't really know.

EdiDD, 2017-12-30:
I also count on it ...

Emanem, 2017-12-31:
@groeck I've asked Ubuntu community to feature your changes https://answers.launchpad.net/ubuntu/+question/662421

If they get interested in including your changes, then perhaps someone will help you out reviewing this?

I looked at the diffs and it seems your driver:

* adds new models (such as `it8686`)
* new features inside `struct it87_devices`, `struct it87_sio_data` and `struct it87_data`
* using some magic numbers for permissions (inside `SENSOR_DEVICE_ATTR_2`)
* re-factored code (*locking*, ...)

I get it would take time to review and upstream - hopefully someone else competent can help you with this?

groeck, 2017-12-31:
It just collected differences over time. Ideally I should have submitted patches upstream in parallel. If I had the time, and if there were actual reviewers. Upstream submissions from others are not a problem - after all, I _am_ the upstream maintainer.

Emanem, 2017-12-31:
> Upstream submissions from others are not a problem - after all, I *am* the upstream maintainer.

That's what I'm counting on mate! You're the *lead* here! :-)

closed 2018-01-14 by groeck