lm-sensors/lm-sensors #45: HP Microserver gen10 not detected

fejesjoco, 2017-12-27:
I tried the latest version of sensors-detect from master but that couldn't find the sensors either. I answered yes to everything.

I can see this in the BIOS: https://drive.google.com/open?id=1MrCO7wYga9y8WkiTCnVAVlgPxYBUMJ8A

I'm running Debian testing with a recent kernel and UEFI.

Can you help how to find the sensors? Should I look for something physically on the motherboard maybe?

groeck, 2017-12-27:
How about providing the output of sensors-detect ?

fejesjoco, 2017-12-27:
This is the output with --auto: https://pastebin.com/AALQc5BH

fejesjoco, 2017-12-27:
With the latest git revision: https://pastebin.com/PFfrWpTF

groeck, 2017-12-27:
So we know there is a JC42 temperature sensor. I don't find anything else documented, so it is questionable if we are going to be able to do better. Maybe the chip on bus 3, address 0x4c, can be identified. Can you send a register dump (sudo i2cdump 3 0x4c) ?

fejesjoco, 2017-12-27:
Yeah, absolutely no documentation. I'll happily probe every address of every bus to figure out where it is. Or do you think it's possible that this thing is only visible to the BIOS?

https://pastebin.com/NzcUbGW6

groeck, 2017-12-27:
Doesn't ring a bell. Can you also provide output of "sudo i2cdump 3 0x4c w" ?

groeck, 2017-12-27:
Re BIOS: I am sure the chip is (or the chips are) visible to the OS. That won't help though if we don't know what the chips are and how to access them.

fejesjoco, 2017-12-27:
sudo i2cdump 3 0x4c w -> XXXX all the way.

Do you think I could find the chip by looking closely at the motherboard?

groeck, 2017-12-27:
Possibly, if you can open the chassis, unless it is covered by a heat sink.

groeck, 2017-12-27:
Can you also provide output of lspci -nn ?

groeck, 2017-12-27:
k10temp might work to report the CPU temperature. Please give it a try.

fejesjoco, 2017-12-27:
Just upgraded to kernel 4.14. k10temp now appeared in sensors-detect under family 15h. Wonder why it didn't show up before.

lspci: https://pastebin.com/ipXBgXq3

groeck, 2017-12-27:
I added detection of your CPU to sensors-detect an hour or so ago.

fejesjoco, 2017-12-27:
Ah, awesome. This is where we are: https://pastebin.com/gwMRNEA1

fejesjoco, 2017-12-27:
What exactly should I look for in i2cdump? I have some interesting dumps like: https://pastebin.com/ahzpcJ6D

fejesjoco, 2017-12-27:
I took a look at the motherboard. I couldn't really pull it apart, so I couldn't see everything, the heatsink is also quite huge. The only interesting chip I could identify is an NCT7491.

fejesjoco, 2017-12-27:
According to its datasheet (http://www.onsemi.com/pub/Collateral/NCT7491-D.PDF), it's somewhat (?) similar to ADT7490. Register 0x3e should return 0x1a. I modified the code so that this chip is probed at all possible I2C addresses and I printed $mid in adt7490_detect(). The set of values I got: 0xff, -1, 0x00, 0x16.

groeck, 2017-12-27:
nct7491 would not be at address 0x4c; it should be at 0x2c, 0x2d, or 0x2e. Can you run i2cdetect on i2c bus 2 and 3 (sudo i2cdetect 2; sudo i22cdetect3) to tell us what is there ?

groeck, 2017-12-27:
Note that nct7491 registers are paged, so you probably can not just read a register value. You would also have to check the page number.

fejesjoco, 2017-12-28:
i2cdetect: https://pastebin.com/YkhgJ0sn

fejesjoco, 2017-12-28:
The BIOS screeshot shows that thermal shutdown is enabled. Is it possible that this setting is keeping the device busy all the time?

fejesjoco, 2017-12-28:
According to the datasheet, 0xff is the page select register, with value 0 or 1. No match.

groeck, 2017-12-28:
Not good; this means the chip is not visible on any of the supported i2c busses. It either sits behind an i2c multiplexer, or behind an unsupported i2c bus controller, or behind an EC (Embedded Controller). Either case, I don't think there is anything else we can do without additional information from the board manufacturer. 
Re thermal shutdown: The NCT4791 supports that as configuration option if I recall correctly. I don't understand what you mean with "setting is keeping the device busy all the time", sorry.

groeck, 2017-12-28:
Correction to what I just said - it should be somewhat safe to run i2cdump on adapters 0 and 1. Try to run it on addresses 0x2c..0x2e on those adapters (i2cdump 0 0x2c; i2cdump 0 0x2e; ...).
Important: Please don't run i2cdetect or i2cdump on any of the AMDGPU or DP I2C adapters. On some systems that can permanently mess up your graphics.

groeck, 2017-12-28:
i2cdumps at addresses 0x50..0x53 are usually EEPROMs. The one at 0x50 is the eeprom on a DIMM, for example.

fejesjoco, 2017-12-28:
I tried 0x2c-0x2d-0x2e on adapters 0-1-2-3-4-5 and all result in XX. The rest are GPU/DP.

From the source code, it looks like 0x4c is also a common address, which I do have on device 3, does this look familiar? https://pastebin.com/PkSyCfTB

fejesjoco, 2017-12-28:
dmidecode talks a lot about sensors, there must some clue in here: https://pastebin.com/ARQKuNDy

groeck, 2017-12-28:
The data at 0x4c doesn't look familiar. The chip only supports i2c addresses 0x2c..0x2e, so it won't be visible elsewhere. dmidecode doesn't tell anything except that sensors are available. You could try to decode and interpret the DSDT, but that can take a lot of time and the thing is quite cryptic and hard to understand.

fejesjoco, 2017-12-28:
I have the decompiled ACPI tables, not sure what to do with them. One of them is related to sensors and alerts but it doesn't contain too much info. Turns out we work at the same place so we can chat if you like.

fejesjoco, 2017-12-29:
[acpi.zip](https://github.com/groeck/lm-sensors/files/1592853/acpi.zip)

groeck, 2017-12-29:
Hmm, interesting - according to the DSDT, there should be four AMD0010 devices, pointing to Synopsys I2C bus master controllers, but only two are registered. Do you seen any related output in dmesg ? Also, can you post /proc/iomem and /proc/interrupts ? I suspect though that the other two may be controlled by an EC - OperationRegion (ECMC, ...) looks suspicious.

fejesjoco, 2017-12-29:
iomem and interrupts: https://pastebin.com/Nhpssqm0

groeck, 2017-12-29:
Two i2c controllers are not there (at fedc4000 and fedc5000), but that can be because they are disabled, not present in this HW, or controlled by an EC. No idea what else we can do.

fejesjoco, 2017-12-29:
I asked here: https://community.hpe.com/t5/System-Administration/Sensors-can-t-be-detected-under-Linux-on-Microserver-gen10/td-p/6992424

fejesjoco, 2017-12-30:
I booted into a Fedora Live with legacy booting. It saw the same number of I2C buses, among them the 4 PIIX4's, but interestingly all of them presented the same devices and addresses, which is clearly buggy.

groeck, 2017-12-30:
See upstream patch 0fe16195f891 ("i2c: piix4: Fix SMBus port selection for AMD Family 17h chips").

groeck, 2018-02-13:
At this point we have no known means for sensors-detect to detect the missing devices. We have to assume that this is a system problem. Most likely the missing sensors are not direectly accessible from Linux in the first place. Closing.

closed 2018-02-13 by groeck

acooks, 2018-02-22:
0x4c is possibly the SB-TSI interface for reading CPU temperature. On Family 16h Model 30h, there main smbus device can access 0x4c on both ports, and the temperature value is in register 0x01.

I don't see the nct7491 in the attached acpi dumps (but it needs to be configured somehow), so it seems plausible that the nct7491 isn't connected to the host smbus controller. I'm also seeing strange behaviour with this chip (on a different board) when its smbus master pins share a bus with the host, so it could be that the HP board doesn't connect the sensors to the same bus as the host's smbus controller.

FYI: I found this thread while looking for information on the nct7491. I'm working on a Linux driver for it.

fejesjoco, 2018-02-22:
If you have any ideas I could try, let me know. My best idea so far would be to reverse engineer the EFI firmware. But I didn't have time to look into it yet.

HP support told me that sensors are usually readable in the ILO GUI. All gen10 servers have ILO5 except this one doesn't have it at all. If we knew what protocol ILO was using to talk to the chip, we could try that too, maybe there's an unconnected ILO port here.

Otherwise HP support has been very unhelpful. I can't seem to get past L1, they are acting like idiots. If you know an HP engineer, maybe they could simply tell us the information too.

pekseg, 2018-07-23:
any news on this?

acooks, 2018-07-24:
@pekseg there's nothing more that can be done about this in software. The nct7491 doesn't seem to be connected to a bus that is usable by the main cpu.

MWolthuis, 2019-02-11:
Searching around the web I came across the following site:
https://redmine.ixsystems.com/issues/64077
It also mentions the missing temperature info in FreeNAS. At the end of the site there are some github commits. Might they be useful for solving this issue?

fejesjoco, 2019-02-11:
@MWolthuis that's only for CPU temp I think.