lm-sensors/lm-sensors #54: Recent kernels lack support of /dev/port (CONFIG_DEV_PORT has been dropped)

noci2012, 2018-01-17:
This makes sensors_detect fail:

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
/dev/port: No such file or directory

groeck, 2018-01-17:
You might want to consider enabling CONFIG_DEVPORT in your configuration. Looks like you have it explicitly disabled, since it is enabled by default.
Also, the problem has been fixed with lm-sensors 3.3.4, though the error message is expected. What version of lm-sensors are you running, and does it continue after the message ? If it does continue, what is your expectation ?

closed 2018-01-17 by groeck

noci2012, 2018-01-17:
Apearantly  something changed in the kernel config unintentionaly:
in kernel config 4.9.6 it was enabled, during the recent frienzy of updating to 4.14.12 it appearantly got lost, although the .config was copied..., no INTENTIONAL change.

3.4.0 is present on my system

noci2012, 2018-01-17:
needless to say I disagree with closing....

groeck, 2018-01-17:
groeck@server:~/src/linux-staging$ git describe
v4.15-rc8
groeck@server:~/src/linux-staging$ git grep DEVPORT
drivers/char/Kconfig:config DEVPORT

DEVPORT is in Device Drivers -> Character devices -> /dev/port character device

noci2012, 2018-01-17:
It didn't show as this was X32, with ISA disabled. 
The option didn't show up when in make menuconfig
do /DEVPORT
after enabling ISA save, and restart make menuconfig it the choice came back.
Not sure why ISA went away.

noci2012, 2018-01-17:
So yes it is still there and it is about a config error. Solved.