lm-sensors/lm-sensors #68: it8665 - sensors not honoring limit settings

wenzler, 2018-01-26:
Hey,

not sure what I’m doing wrong here.
Setting min max values following sensors -u notation,
the labels are used so the chip section is not ignored - yet no change on the limits. :/

chip "it8665-*"

    label temp1     "CPU Temp"
    label temp2     "SYS Temp"
    label temp3     "AUX Temp"
    label fan1     "CPU Fan"
    label fan2     "Top Fan1"
    label fan3     "Top Fan2"
    label fan4     "Bottom Fan"

    set temp1_min 1.000
    set temp1_max 50.000
    set temp2_min 1.000
    set temp2_max 100.000
    set temp3_min 1.000
    set temp3_max 50.000
    set fan1_min  300.000
    set fan2_min  300.000
    set fan3_min  300.000
    set fan4_min  300.000
    set fan5_min  300.000
    set fan6_min  300.000

it8665-isa-0290
Adapter: ISA adapter
in0:          +0.74 V  (min =  +2.04 V, max =  +2.54 V)
in1:          +2.50 V  (min =  +0.00 V, max =  +1.21 V)
in2:          +2.00 V  (min =  +0.69 V, max =  +2.10 V)
in3:          +2.01 V  (min =  +1.24 V, max =  +1.41 V)
in4:          +0.56 V  (min =  +0.32 V, max =  +1.19 V)
in5:          +0.45 V  (min =  +2.38 V, max =  +2.70 V)
in6:          +0.94 V  (min =  +0.79 V, max =  +2.71 V)
3VSB:         +3.27 V  (min =  +3.12 V, max =  +4.67 V)
Vbat:         +3.18 V  
+3.3V:        +3.27 V  
CPU Fan:      957 RPM  (min =   33 RPM)
Top Fan1:     774 RPM  (min =   34 RPM)
Top Fan2:     787 RPM  (min =   25 RPM)
Bottom Fan:   372 RPM  (min =   -1 RPM)
fan5:           0 RPM  (min =   -1 RPM)
fan6:           0 RPM  (min =   -1 RPM)
CPU Temp:     +40.0°C  (low  = +29.0°C, high =  -9.0°C)
SYS Temp:     +23.0°C  (low  = -101.0°C, high = +120.0°C)  sensor = thermistor
AUX Temp:     +30.0°C  (low  = +93.0°C, high = -101.0°C)  sensor = thermistor
intrusion0:  ALARM

This is an Asus X370-F ROG - current git versions for sensors and it87 used.

Thanks!

groeck, 2018-01-26:
I am sorry, I have no idea what you mean with "Setting min max values following sensors -u notation". Please elaborate.

wenzler, 2018-01-26:
Well - I rely on sensors -u to giving me the correct field names for using in sensors3.conf

sensors -u shows
> AUX Temp:
>   temp3_input: 27.000
>   temp3_max: 50.000
>   temp3_min: 1.000

So the adding this to sensors3.conf:
> chip "it8665-*"
>     label temp3     "AUX Temp"
>     set temp3_min 1.000
>     set temp3_max 50.000
> 

is imho supposed to change the sensors output from

> AUX Temp: +30.0°C (low = +93.0°C, high = -101.0°C) sensor = thermistor

To

> AUX Temp: +30.0°C (low = +1.0°C, high = 50.0°C) sensor = thermistor

But this does not happen.
And that’s where I scratch my head.

closed 2018-01-26 by wenzler