lm-sensors/lm-sensors #70: AMD Ryzen 1950X Incorrect Temperature Value

luisthedude, 2018-02-07:
`k10temp-pci-00cb
Adapter: PCI adapter
temp1:       +4294948.3°C  (high = +70.0°C)
 
nouveau-pci-4100
Adapter: PCI adapter
GPU core:     +0.90 V  (min =  +0.85 V, max =  +1.00 V)
temp1:        +39.0°C  (high = +95.0°C, hyst =  +3.0°C)
                       (crit = +105.0°C, hyst =  +5.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)
 
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
 
k10temp-pci-00c3
Adapter: PCI adapter
temp1:       +4294948.3°C  (high = +70.0°C)`

OS: Ubuntu 17.10
Kernel: Linux Kernel 4.15.1 Stable
CPU: AMD 1950X 
Motherboard: Asus Prime X399-A

groeck, 2018-02-07:
Yes, I have seen it, but I have no idea what is going on. The value suggests that the chip may report a temperature below 27 degrees C (around 19 degrees C if I interpret the numbers correctly), which would turn into those huge values. Question is why; I wonder if AMD changed the microcode and dumped or changed the temperature offset. I'll adjust the driver and report 0 degrees in that situation.

luisthedude, 2018-02-07:
I'm not sure how but this user has it working correctly: https://www.phoronix.com/scan.php?page=news_item&px=AMD-Zen-Temps-Hwmon-Next

groeck, 2018-02-07:
I know about that article, and obviously I know about the code since I wrote it, and since I am the upstream maintainer of the hwmon subsystem. Yet, here we have an instance where it does not or no longer work.

luisthedude, 2018-02-07:
If you would like we can chat and troubleshoot the issue, let me know what you think.

groeck, 2018-02-08:
Output from /proc/cpuinfo might help. Other than that, there isn't much to debug. I submitted a patch upstream to report 0 degrees C in that situation, but that won't really solve the problem. I'll ask around if that was seen elsewhere.

luisthedude, 2018-02-08:
Hopefully this helps, full output can be found here: https://pastebin.com/4PEifiMU

groeck, 2018-02-08:
Not really. Nothing unusual, same microcode version as the one running on my 1700X. Does the output change when the system runs hot ?

luisthedude, 2018-02-08:
When the system gets hot the last three digits change, for instance from +42949**48.3**°C to +42949**60.1**°C, so maybe you will only have to delete +42949.

groeck, 2018-02-08:
We can not just randomly add or remove significant bits, especially if an underflow is involved.
+4294948.3°C translates to ~8°C reported by the chip (unadjusted), and +4294960.1°C translates to ~19.8°C. None of those numbers makes any sense; the chip is supposed to report a _higher_ temperature than its real temperature, not a lower temperature. I wonder if the chip does just that - if I add 27°C instead of subtracting it, I calculate +35°C at idle and 46.8°C under load, which would make some sense. But I still have no idea why that would be the case.
Does the system have a Super-IO chip which could be used to compare the reported temperatures ?

luisthedude, 2018-02-08:
I did hear about the offset of 27°C so you might be right on this, the motherboard is an Asus Prime X399-A. How can I check for the Super-IO chip for this motherboard?

groeck, 2018-02-08:
sensors-detect would tell you. Still, it is really odd that the temperature offset would go the other way, unless it is really selected by the BIOS and Asus messed up. Unfortunately, I don't have access to the NDA documentation, so I have no idea if there is a means to configure the temperature offset in the chip.

luisthedude, 2018-02-08:
Do you need a complete output of sensors-detect as well?

groeck, 2018-02-08:
No, just if it detected any Super-IO chips, and the chip IDs.

groeck, 2018-02-10:
I received the following information:
"Tctl reporting can be influenced by changing the Sense MI Skew option in BIOS. This setting has a direct influence and ASUS BIOSes have this option. So I believe that the user has changed this setting and that's why the values reported are not correct."

Please confirm if you changed the BIOS configuration.

luisthedude, 2018-02-11:
Yes, both Sense MI Skew and Sense MI Offset are set to Auto, would I need to disable both of this options?

groeck, 2018-02-11:
I have no idea what the correct settings are. There is some information available online suggesting that it may depend on the BIOS version. Some posts suggest that skew has to be enabled and that offset would have to be either 272 or 283, but I have no idea what either value means. Unfortunately the actual chip register values are unpublished, so I have no means to detect the status and adjust for bad settings in the driver.

luisthedude, 2018-02-11:
After disabling  Sense MI Skew  only I can confirm that it displays the correct temperature from 35 to 65 under load. Quick question does this temperature already include the 27+ offset?

groeck, 2018-02-11:
Interesting, I thought Sense MI Skew would have to be _enabled_, not disabled. Anyway, yes, the displayed temperature includes the 27 degrees offset (so the chip really reports +35 + 27 degrees C).

luisthedude, 2018-02-12:
Looking good so far, thank's again for your assistance.

groeck, 2018-02-12:
Closing per above.

closed 2018-02-12 by groeck