lm-sensors/lm-sensors #71: aorus gigabyte gaming 7 and Threadripper 1950x

Goddard, 2018-02-10:
Finding no sensors using sensors detect and building from source.  Any help is appreciated here.  Let me know if I can provide additional information.

Thanks

groeck, 2018-02-10:
"Finding no sensors" is not actionable. Kind of similar to "My car does not start. Any help is appreciated". Building what from source ? Output from sensors-detect ? What version of sensors-detect ? What distribution ?

Goddard, 2018-02-10:
Ubuntu 16.04 with kernel 4.15.2

sudo sensors-detect 
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Gigabyte Technology Co., Ltd. X399 AORUS Gaming 7 [Default string]
# Kernel: 4.15.2-041502-generic x86_64
# Processor: AMD Ryzen Threadripper 1950X 16-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8686
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8733

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no):
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no):
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no):
Found unknown SMBus adapter 1022:790b at 0000:00:14.0.
Sorry, no supported PCI bus adapters found.

Next adapter: NVIDIA i2c adapter 1 at 41:00.0 (i2c-0)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 2 at 41:00.0 (i2c-1)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 4 at 41:00.0 (i2c-2)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 6 at 41:00.0 (i2c-3)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 7 at 41:00.0 (i2c-4)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 8 at 41:00.0 (i2c-5)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 9 at 41:00.0 (i2c-6)
Do you want to scan it? (yes/NO/selectively): yes

Sorry, no sensors were detected.
Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
http://www.lm-sensors.org/wiki/Devices for driver status.

groeck, 2018-02-10:
Install the it87 driver from https://github.com/groeck/it87. FWIW, you did not run sensors-detect from here; that would detect the chips.

Goddard, 2018-02-11:
ahh yeah you are right, I uninstalled the version of lm-sensors and it detected it fine.  Do I have to reboot for them to show up with sensors?

groeck, 2018-02-11:
https://github.com/groeck/it87/blob/master/README, last section

Goddard, 2018-02-11:
So this means the "sensors" command won't show any extra sensors?

sensors-detect showed successes, but using sensors displays.  I guess I was expecting to see some CPU sensors?

sensors
iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +40.0°C

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +43.4°C  (high = +70.0°C)

k10temp-pci-00cb
Adapter: PCI adapter
temp1:        +43.4°C  (high = +70.0°C)

# sensors-detect revision $Revision$
# System: Gigabyte Technology Co., Ltd. X399 AORUS Gaming 7 [Default string]
# Kernel: 4.15.2-041502-generic x86_64
# Processor: AMD Ryzen Threadripper 1950X 16-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8686E Super IO Sensors'                        Success!
    (address 0x240, driver `to-be-written')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8792E Super IO Sensors'                        Success!
    (address 0xa60, driver `it87')

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no):
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): yes
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): yes
Using driver `i2c-piix4' for device 0000:00:14.0: AMD KERNCZ SMBus

Next adapter: NVIDIA i2c adapter 1 at 41:00.0 (i2c-0)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 2 at 41:00.0 (i2c-1)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 4 at 41:00.0 (i2c-2)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 6 at 41:00.0 (i2c-3)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 7 at 41:00.0 (i2c-4)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 8 at 41:00.0 (i2c-5)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: NVIDIA i2c adapter 9 at 41:00.0 (i2c-6)
Do you want to scan it? (yes/NO/selectively): yes

Now follows a summary of the probes I have just done.
Just press ENTER to continue:

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

Driver `to-be-written':
  * ISA bus, address 0x240
    Chip `ITE IT8686E Super IO Sensors' (confidence: 9)

Driver `it87':
  * ISA bus, address 0xa60
    Chip `ITE IT8792E Super IO Sensors' (confidence: 9)

Note: there is no driver for ITE IT8686E Super IO Sensors yet.
Check http://www.lm-sensors.org/wiki/Devices for updates.

Do you want to generate /etc/sysconfig/lm_sensors? (yes/NO): yes
Copy prog/init/lm_sensors.service to /lib/systemd/system
and run 'systemctl enable lm_sensors.service'
for initialization at boot time.

Goddard, 2018-02-11:
Actually think I found the issue.

● lm_sensors.service - Initialize hardware monitoring sensors
   Loaded: loaded (/lib/systemd/system/lm_sensors.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-02-11 15:58:57 EST; 22s ago
  Process: 6333 ExecStart=/usr/bin/sensors -s (code=exited, status=203/EXEC)
  Process: 6327 ExecStart=/sbin/modprobe -qab $BUS_MODULES $HWMON_MODULES (code=exited, status=1/FAILURE)
 Main PID: 6333 (code=exited, status=203/EXEC)

Feb 11 15:58:57 99-x399 systemd[1]: Starting Initialize hardware monitoring sensors...
Feb 11 15:58:57 99-x399 systemd[1]: lm_sensors.service: Main process exited, code=exited, status=203/EXEC
Feb 11 15:58:57 99-x399 systemd[1]: Failed to start Initialize hardware monitoring sensors.
Feb 11 15:58:57 99-x399 systemd[1]: lm_sensors.service: Unit entered failed state.
Feb 11 15:58:57 99-x399 systemd[1]: lm_sensors.service: Failed with result 'exit-code'.

groeck, 2018-02-11:
It looks like you may have installed the most recent version of lm-sensors, but not the updated it87 driver I pointed to above. It might also be possible that there is an ACPI reserved address conflict. What the README file is saying is that updating sensors-detect will not have any impact on the actual sensor drivers, and vice versa.

You might see some error messages in dmesg.

Goddard, 2018-02-12:
Here is the dmesg error.

[67684.271451] it87: loading module not compiled with retpoline compiler.
[67684.272589] it87: it87 driver version v1.0-34-g850c17b
[67684.272663] it87: Found IT8686E chip at 0xa40, revision 2
[67684.272701] it87: Beeping is supported
[67684.272721] ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20170831/utaddress-247)
[67684.272730] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver

groeck, 2018-02-12:
Well, there you have the explanation. See module parameters for workarounds.

groeck, 2018-02-13:
Closing. Not an lm-sensors-issue but an issue with the it87 driver; not even an issue there either but needs module parameters. Reopen against it87 if there are still questions.

closed 2018-02-13 by groeck

xjcl, 2020-11-16:
So is this a correct summary from a user's point of view?:

- Can't use the distro's lm-sensors because it is too old and doesn't detect anything
- Can't use the master lm-sensors because the it87 driver version is incompatible
- Nothing one can do to get temp readings except install a newer distro

karlp, 2021-01-31:
is there an current source for the it87 driver? the github repo is deleted, with no information to link forwards.

davispuh, 2026-01-20:
> is there an current source for the it87 driver? the github repo is deleted, with no information to link forwards.

Looks like it's being maintained at https://github.com/frankcrawford/it87