lm-sensors/lm-sensors #72: ASRock Z97 cpu fan rpm not show

ilje, 2018-02-13:
Hi i have Alphacool Eisbaer 360 water cooling system and i have problem with lm-sensors
like you can see in my bios you have show rpm speed pump cpu fan2 and chassis fan 1 and 2 
but lm-sensors show only chassis fan without cpu fan2 (water pump) what is wrong here ?
Operating System: Fedora 27 (Workstation Edition)
CPE OS Name: cpe:/o:fedoraproject:fedora:27
Kernel: Linux 4.15.2_ilje.Fedora27
 Architecture: x86-64

[image]

groeck, 2018-02-13:
First of all, this is not an issue with the 'sensors' command or lm-sensors. 'sensors' only displays the information provided by the driver, here the nct6775 driver. 
Second, the most likely reason is that the board multiplexes two fan controls to a single pin on the NCT6776 chip; ASRock boards are known for doing this. It may well be that both CPU fans are multiplexed. "ASRock Z97" does not provide enough information to be able to determine if this is the case with this board; there are several different ASRock Z97 boards available; a quick search on the ASRock web page shows more than 20.

ilje, 2018-02-14:
hi sorry this is my mb Z97 Pro3 http://www.asrock.com/mb/Intel/Z97%20Pro3/

groeck, 2018-02-14:
The configuration file of ASRock's A-Tuning Utility confirms that both CPU fans are connected to the same fan count input on the NCT6776. The selection is done using GPIO pins (GPIO01 and GPIO16). The multiplexed fan inputs (connectors) are labeled fan2 and fan6 in the configuration file. The best option might be to connect the CPU fan to another fan connector.

groeck, 2018-02-21:
Closing. Notethat this is not an lm-sensors issue to start with.

closed 2018-02-21 by groeck