lm-sensors/lm-sensors #74: ASROCK - J4105-ITX, no fans

Reiner1210, 2018-02-21:
I have the following motherboard
https://www.asrock.com/mb/Intel/J4105-ITX/

sensors get the following output

acpitz-virtual-0
Adapter: Virtual device
temp1:        +51.0°C  (crit = +95.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +51.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +51.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +51.0°C  (high = +105.0°C, crit = +105.0°C)
Core 2:        +50.0°C  (high = +105.0°C, crit = +105.0°C)
Core 3:        +50.0°C  (high = +105.0°C, crit = +105.0°C)

I do not see the fans (CPU and chassis)

sensors-detect:

# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# Board: ASRock J4105-ITX
# Kernel: 4.15.3-1-default x86_64
# Processor: Intel(R) Celeron(R) J4105 CPU @ 1.50GHz (6/122/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): y
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd423
    (logical device B has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:31d4 at 0000:00:1f.1.
Sorry, no supported PCI bus adapters found.
Module i2c-dev loaded successfully.

Next adapter: SMBus I801 adapter at f040 (i2c-0)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No

Next adapter: i915 gmbus dpb (i2c-1)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpc (i2c-2)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus misc (i2c-3)
Do you want to scan it? (yes/NO/selectively): 

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no): 
Unloading i2c-dev... OK
Unloading cpuid... OK

groeck, 2018-02-21:
The Super-IO chip is an NCT6796D which is not yet supported by the kernel driver.
Note that this is _not_ an lm-sensors issue.

closed 2018-02-21 by groeck

groeck, 2018-02-21:
Added note: https://github.com/groeck/nct6775 does have experimental support for the NCT6796D, and support for detection of NCT6796D has been added to sensors-detect.

Reiner1210, 2018-02-23:
Do you know which kernel will support it in the future ?

Mit AquaMail Android
http://www.aqua-mail.com gesendet

groeck, 2018-02-23:
The nct6775 driver here (https://github.com/groeck/nct6775) supports the NCT6796D. If I get some test feedback in time, I'll submit the necessary changes into v4.17. Unfortunately, so far no one provided test feedback, otherwise it would already be available.

Reiner1210, 2018-02-27:
Do you have received my last emails. Concerning the tests I have done ?

Mit AquaMail Android
http://www.aqua-mail.com gesendet

groeck, 2018-02-27:
No, I have not received any e-mails with test results.

Reiner1210, 2018-02-27:
Can you give me your e mail address ?

Mit AquaMail Android
http://www.aqua-mail.com gesendet

Reiner1210, 2018-02-27:
You have now the email ?

groeck, 2018-02-27:
> You have now the email ?
Yes, I do. I'll have to compare with the datasheet tonight, but at first glance
everything looks good to me.

Voltages need to be scaled via /etc/sensors3.conf.

Guenter