lm-sensors/lm-sensors #77: NCT6106D on Supermicro X10SLV not reporting

reluctantuser, 2018-03-05:
Hardware: brand new Supermicro X10SLV motherboard with an Intel G3220 processor & 16GB ram

OS: Debian 10 "buster", 64-bit, fully updated

LM Senors is installed. Debian package info is:
libsensors4:amd64  1:3.4.0-4  amd64  library to read temperature/voltage/fan sensors
lm-sensors         1:3.4.0-4  amd64  utilities to read temperature/voltage/fan sensors

"sensors-detect" has been run and did discover the NCT6106D chip.
"lsmod" reports the "nct6775" module is loaded.

The output of "sensors" only reports the following "chips":
pch_haswell-virtual-0
acpitz-virtual-0
coretemp-isa-0000

I do have Supermicro's own "Superdoctor 5" loaded and it sees all of the fans, temperatures, and voltages.

Is it possible the Supermicro software or something else that I cannot yet identify has already grabbed the ports that the "lm sensors" software wants to use?

I suggest that question after seeing the following output in "dmesg":

[    2.368881] nct6775: Found NCT6106D or compatible chip at 0x2e:0x290
[    2.368959] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000290-0x000000000000029F (\_SB.PCI0.LPCB.SIO1.HMIO) (20170728/utaddress-247)
[    2.369078] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver

I can try uninstalling "Superdoctor 5" to see if it makes a difference.

My prior experience with "Superdoctor 2", the CLI only variant of Supermicro's sensors monitoring software, combined with "lm sensors" has been excellent; the 2 pieces of software do not interfere with each other.

groeck, 2018-03-05:
Sure, something else (BIOS/ACPI) has grabbed the resources. This is why you get the warning. You would have to boot with "acpi_enforce_resources=lax" command line option and accept the risk involved.

reluctantuser, 2018-03-05:
Yes, that boot parameter is exactly what was needed.

The "sensors" command" now displays the output of the "nct6106-isa-0290" chip when the "nct6775" module loaded and that boot parameter is used.

Here's the dump of the output:

nct6106-isa-0290
Adapter: ISA adapter
in0:                    +1.73 V  (min =  +1.26 V, max =  +2.00 V)
in1:                    +1.82 V  (min =  +1.55 V, max =  +2.02 V)
in2:                    +3.33 V  (min =  +2.90 V, max =  +3.66 V)
in3:                    +3.34 V  (min =  +2.83 V, max =  +3.66 V)
in4:                    +1.50 V  (min =  +1.12 V, max =  +1.72 V)
in5:                    +1.06 V  (min =  +0.90 V, max =  +1.22 V)
in6:                    +1.64 V  (min =  +1.45 V, max =  +1.83 V)
in7:                    +3.10 V  (min =  +2.50 V, max =  +3.60 V)
in8:                    +2.02 V  (min =  +0.00 V, max =  +4.08 V)
fan1:                   543 RPM  (min =  200 RPM)
fan2:                   952 RPM  (min =  200 RPM)
fan3:                     0 RPM  (min =  200 RPM)  ALARM
SYSTIN:                 +40.0°C  (high = +85.0°C, hyst = +80.0°C)
                                 (crit low = +127.0°C, crit = +127.0°C)  sensor = thermal diode
CPUTIN:                 +37.0°C  (high = +85.0°C, hyst = +80.0°C)
                                 (crit low = +127.0°C, crit = +127.0°C)  sensor = thermistor
AUXTIN:                 +23.0°C  (high = +80.0°C, hyst = +75.0°C)
                                 (crit low = +127.0°C, crit = +127.0°C)  sensor = thermistor
PECI Agent 0:           +43.0°C  (high = +98.0°C, hyst = +93.0°C)
                                 (crit low = +127.0°C, crit = +93.0°C)
PCH_CHIP_CPU_MAX_TEMP:  +50.0°C  (high = +95.0°C, hyst = +90.0°C)
                                 (crit low = +127.0°C, crit = +127.0°C)
PCH_CHIP_TEMP:           +0.0°C     TEMP9
PCH_CPU_TEMP:            +0.0°C     TEMP10
intrusion0:            ALARM
beep_enable:           enabled

The Supermicro X10SLV board does have 3 fan headers. All fan headers are the 4-pin type.

And here I had thought the days when I needed to use that boot parameter had long since past as I had forgotten about the  times when I used to use it.

closed 2018-03-05 by reluctantuser