lm-sensors/lm-sensors #80: Nuvoton NCT6795D hardwired inputs

maciejsszmigiero, 2018-03-11:
Hi,

I am wondering if Nuvoton NCT6795D has the same hardwired inputs as other members of its family,
and so should be added to etc/sensors.conf.default after "nct6791-*".

It seems to me (experimentally) that this is the case indeed, but can this be verified for example in the
datasheet?

Thanks.

groeck, 2018-03-11:
There are two variants of this chip, one built for a specific board vendor and one for everyone else. The datasheet of the chip built for the board vendor is held as close secret by bot the board vendor and not available. On top of that, there is no known means to distinguish the "special" from the "generic" chip. It is known that, for example, the SMBUSMASTER 0 input on the "special" chip is connected to the CPU temperature. Other than that, you know just as much as I do. Why don't you publish your findings ?
Note that we do publish configurations in the "configs" subdirectory. Please feel free to publish yours.

maciejsszmigiero, 2018-03-11:
So I guess an other than experimental verification of this fact is not possible as of now.

Since the values do seem to match and it would be a reasonable thing for Nuvoton to make these inputs consistent with earlier chips of this family I will open a PR in a moment to add this chip to its siblings in sensors.conf.default.

> Note that we do publish configurations in the "configs" subdirectory.

I will submit my board (MSI MS-7A34 or B350 TOMAHAWK) when I will be able to determine with
a reasonable accuracy all the important data: I am still, for example, not sure about 5V and 12V lines.

closed 2018-03-11 by maciejsszmigiero

groeck, 2018-03-11:
Try to match BIOS voltages and BIOS voltage jumps against voltages reported by the chip. Comparison with hwinfo data available on the web suggests
0 = Vcore
1 = 5V (*5)
2 = AVCC
3 = 3VCC
4 = 12V (* 12 ?)
5, 6 = ?
7 = 3VSB
8 = VBAT
9 = VTT
10 = ?
11 = CPU NB
12 = ?
13 = dimm (*2 ?)
14 = ?

maciejsszmigiero, 2018-03-11:
> 1 = 5V (x 5)

for +5V in4 also gives a good result in my case:
compute in4 5.040 * @, @ / 5.040

> 9 = VTT

It looks like +12V in my case:
compute in9 ((56 / 10) + 1) * @, @ / ((56 / 10) + 1)

So further verification is still necessary (I hope that a schematic of either this or a similar board
will finally show on some web site like it usually happened in the past), but thanks for this data anyway!

groeck, 2018-03-11:
A multiplication factor of 5.04 would be really difficult to accomplish with divider resistors and thus seems to be a bit unlikely. Someone else suggested in1=5V, in4=12V, in5=System Agent, in6=CPU Ring, in10=Gfx Vcore, and in14=VCCIN for B350M MORTAR. It is at least somewhat likely that boards in the same family would have similar voltages.

Do you have reason to believe that the data provided by hwinfo is wrong ?

maciejsszmigiero, 2018-03-11:
> A multiplication factor of 5.04 would be really difficult to accomplish with divider resistors and thus
> seems to be a bit unlikely.

You are right. A scaling factor of 5.0 seems more plausible in this regard.
So in1 is probably +5V as you wrote earlier.

> Do you have reason to believe that the data provided by hwinfo is wrong ?

I assume you mean a program from https://www.hwinfo.com/.

I don't really have an opinion on this since I don't know where they get their data from (maybe it is likewise of purely experimental nature - I just don't know).