lm-sensors/lm-sensors #84: fancontrol using hddtemp

JohnKis, 2018-03-13:
Hi there,

I've set up fancontrol to control my case fans based on hard drive temps.

I'm using the following script to save the average and maximum drive temps to a file:

```sh
#!/bin/bash -e
#
# Company:  PowerCraft Technology
# Author:   Copyright Jelle de Jong <jelledejong*******>
# Note:     Please send me an email if you enhanced the script
# Version:  0.0.4
# Date:     2010-07-04 / 2010-07-05

# This program is free software; you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation; either version 3 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with this program; if not, write to the Free Software
# Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston,
# MA 02110-1301, USA.

PATH=/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/bin/

LC_ALL=C

# set -x
# exec 2>>/var/log/pct-fancontrol-hddtemp.log
# date=$(date)

DISKS="/dev/sda /dev/sdb /dev/sdc"

# located in /var/lib/ so it will be there for fancontrol during boot
LOCATION=/var/lib/fancontrol-hddtemp
FILE_MAX="$LOCATION"/fancontrol-hddtemp-maximum
FILE_AVG="$LOCATION"/fancontrol-hddtemp-average

DEGUG=0

[[ -e $LOCATION ]] || mkdir "$LOCATION"

while true
    do
    temperature=0
    summation=0
    maximum=0
    average=0
    count=0
    for value in $(hddtemp -n $DISKS 2>/dev/null)
    do
        temperature="${value//[!0-9]}"
        echo "Temp: $temperature"
	((summation+=$temperature))
        ((count+=1))
        if [ $temperature -gt $maximum ]; then
            maximum=$temperature
        fi
        if [ $DEGUG = 1 ]; then
            average=$(echo "scale=1; $summation / $count" | bc)
            echo $summation
            echo $maximum
            echo $average
            echo $count
            echo "-----"
        fi
    done
    if [ $count -ne 0 ]; then
    	average=$(echo "scale=1; $summation / $count" | bc)
    fi
    # fancontrol output format
    echo "($maximum * 1000)/1" | bc > "$FILE_MAX"
    echo "($average * 1000)/1" | bc > "$FILE_AVG"
    sleep 45
done

exit
``` 

In `/etc/fancontrol` I've set FCTEMPS to use the file from above:
```
FCTEMPS=hwmon2/pwm4=/var/lib/fancontrol-hddtemp/fancontrol-hddtemp-maximum
```

It's working fine however from time to time, fancontrol stops. I checked the logs and found the following:
```
Mar 11 23:15:17 jkis-srv fancontrol[1286]: Error reading temperature from /sys/class/hwmon//var/lib/fancontrol-hddtemp/fancontrol-hddtemp-maximum
Mar 11 23:15:17 jkis-srv fancontrol[1286]: Aborting, restoring fans...
Mar 11 23:15:17 jkis-srv fancontrol[1286]: Verify fans have returned to full speed
Mar 11 23:15:17 jkis-srv systemd[1]: fancontrol.service: Main process exited, code=exited, status=1/FAILURE
Mar 11 23:15:17 jkis-srv systemd[1]: fancontrol.service: Unit entered failed state.
```

The log message doesn't seem to be accurate as it prepends `/sys/class/hwmon/` to every file path regardless whether the path is absolute. So the issue is likely to be a race condition whereby fancontrol tries to read the file at the same time it's written by the hddtemp script.

I appreciate that it isn't a fancontrol issue however I'd be grateful if you could give me some pointers on how to fix it.

jschwender, 2018-03-14:
Maybe it gets better if you reduce the time the file is open. If you replace bc calculation by 
`$(( maximum * 1000 )) > "$FILE_MAX`
the time the file status is open is reduced. I can't see what advantage you gain by utilizing bc, as you use scale=1, you have integer inputs, you divide large numbers by smaller numbers. bc is much more time and resource consuming and you don't want floating output. I would try to replace bc completely here and see if that helps.
greetings

JohnKis, 2018-03-14:
Thanks for the suggestions, I'll try it. Hopefully it'll make things better.

I found the script online and I haven't modified. I think bc is still needed, otherwise the value would be a float (e.g.: 19000.0 instead of 19000), however I can do the calculation before writing the value to the file.

I'll leave it running for a while and will report back.

groeck, 2018-03-14:
Fancontrol doesn't specifically support reading data such as temperatures from files. It expects values to be read from sysfs attributes. Trying to read values from a file may work, but that would be accidental. I would be happy to accept patches modifying the code to be more flexible. Only condition is that any changes have to be backward compatible.

leighbb, 2018-03-15:
Hello,

You can avoid the race by writing to a temporary file then using **mv** to move the file into position.  As **mv** is atomic, the problem should go away.

```
    echo "($maximum * 1000)/1" | bc > "$FILE_MAX.new"
    echo "($average * 1000)/1" | bc > "$FILE_AVG.new"
    mv "$FILE_MAX.new" "$FILE_MAX"
    mv "$FILE_AVG.new" "$FILE_AVG"
```

Hope that helps!

jschwender, 2018-03-15:
Good idea from @leighbb. Note: mv is not generally atomic. The man page does not state that mv is atomic. Though it is true for fs=ext4 beeing local under Linux, but not if source and target are under different mount points, and not necessarily on OSX or with NFS. Not sure about btrfs, zfs? I think you have to use -T option with mv to be atomic, and the target must exist, and i am not sure that is is atomic in any security context (SElinux...). However, it is still a good idea on a local fs like ext4, make sure the intermediate file is on the same local fs and the target exists when you call mv -T .

JohnKis, 2018-03-16:
Thanks @leighbb and @jschwender. It makes perfect sense. I've changed the script and so far so good.

@groeck - Even though it's not intentional, fancontrol seems to work just fine reading values from a file. With the race condition solved, it's working perfectly.

closed 2018-03-16 by JohnKis