lm-sensors/lm-sensors #88: Add ability to use DMI information to select pre-defined configuration

leighbb, 2018-03-15:
# Proposal

I propose to add to libsensors the ability to use the DMI information in sysfs (if present) to load a configuration file appropriate to the system.  This would be located in $PREFIXDIR/lib/dmi/_vendor_/_device_.conf.

Here are examples for the two motherboards I am currently using:

| Board | board\_vendor | board\_name | Proposed filename |
| ----- | ------------ | ---------- | ----------------- |
| ASUS P8B75-V | ASUSTeK COMPUTER INC. | P8B75-V | $PREFIXDIR/lib/dmi/ASUSTek COMPUTER INC./P8B75-V.conf |
| ASRock AB350 Pro4 | ASRock | AB350 Pro4 | $PREFIXDIR/lib/dmi/ASRock/AB350 Pro4.conf |

I have a work in progress patch (see below) that is working for me.  

## Benefits

This capability would allow the package to include pre-defined configurations that will automatically work on the appropriate hardware.  For people with standard hardware or motherboards, they would no longer have to configure lm-sensors manually in most cases after running sensors-detect.

## Packaging

I use Debian and I can envision a second package called lm-sensors-data that would contain contents of the /usr/lib/sensors/dmi directory.  The reason for the separation is that embedded and custom devices wouldn't want it.

## Potential issues

There are a couple of issues that need to be resolved:

### Order of reading the configuration files

The code I have added will read the board-specific configuration if the DMI information is present, prior to reading the standard configuration files, but _only_ if a specific configuration file is not passed to sensors\_init. Just want to confirm if this is the best way forward.

### Quality of DMI Data

It may be necessary to add validation of the DMI data.  It may also be the case that vendors have been in-consistent in the vendor name - this could be addressed with symbolic links.

### Variation in DMI Data

Whilst putting this issue together I've noticed that my HP Microserver does not provide *board\_vendor* or *board\_name* but does provide *sys\_vendor* and *chassis\_name* so I will probably have to enhance the code to check for the various possible combinations.

## WIP Patch

```
diff --git a/Makefile b/Makefile
index d6f2192c..8f38a7fd 100644
--- a/Makefile
+++ b/Makefile
@@ -153,11 +153,11 @@ endif
 ALL_CPPFLAGS += $(CPPFLAGS)
 ALL_CFLAGS += $(CFLAGS)
 
-PROGCPPFLAGS := -DETCDIR="\"$(ETCDIR)\"" $(ALL_CPPFLAGS)
+PROGCPPFLAGS := -DETCDIR="\"$(ETCDIR)\"" -DLIBDIR="\"$(LIBDIR)\"" $(ALL_CPPFLAGS)
 PROGCFLAGS := $(ALL_CFLAGS)
-ARCPPFLAGS := -DETCDIR="\"$(ETCDIR)\"" $(ALL_CPPFLAGS)
+ARCPPFLAGS := -DETCDIR="\"$(ETCDIR)\"" -DLIBDIR="\"$(LIBDIR)\"" $(ALL_CPPFLAGS)
 ARCFLAGS := $(ALL_CFLAGS)
-LIBCPPFLAGS := -DETCDIR="\"$(ETCDIR)\"" $(ALL_CPPFLAGS)
+LIBCPPFLAGS := -DETCDIR="\"$(ETCDIR)\"" -DLIBDIR="\"$(LIBDIR)\"" $(ALL_CPPFLAGS)
 LIBCFLAGS := -fpic -D_REENTRANT $(ALL_CFLAGS)
 
 .PHONY: all user clean install user_install uninstall user_uninstall
diff --git a/lib/init.c b/lib/init.c
index 33bee0f2..49e663c3 100644
--- a/lib/init.c
+++ b/lib/init.c
@@ -195,6 +195,19 @@ int sensors_init(FILE *input)
 		const char* name;
 
 		/* No configuration provided, use default */
+
+		/* Use the DMI board vendor and board name, if available */
+		name = sensors_get_dmi_config_file();
+		if (name) {
+			input = fopen(name, "r");
+			if (input) {
+				res = parse_config(input, name);
+				fclose(input);
+				if (res)
+					goto exit_cleanup;
+			}
+		}
+
 		input = fopen(name = DEFAULT_CONFIG_FILE, "r");
 		if (!input && errno == ENOENT)
 			input = fopen(name = ALT_CONFIG_FILE, "r");
diff --git a/lib/sysfs.c b/lib/sysfs.c
index dcc6af5a..6150ff23 100644
--- a/lib/sysfs.c
+++ b/lib/sysfs.c
@@ -915,3 +915,52 @@ int sensors_write_sysfs_attr(const sensors_chip_name *name,
 
 	return 0;
 }
+
+static char *sensors_read_dmi_board_vendor(void)
+{
+	char path[NAME_MAX];
+
+	if (snprintf(path, NAME_MAX, "%s/class/dmi/id",
+		     sensors_sysfs_mount) < 0)
+		return NULL;
+
+	return sysfs_read_attr(path, "board_vendor");
+}
+
+static char *sensors_read_dmi_board_name(void)
+{
+	char path[NAME_MAX];
+
+	if (snprintf(path, NAME_MAX, "%s/class/dmi/id",
+		     sensors_sysfs_mount) < 0)
+		return NULL;
+
+	return sysfs_read_attr(path, "board_name");
+}
+
+char *sensors_get_dmi_config_file(void)
+{
+	char path[NAME_MAX];
+	char *board_vendor;
+	char *board_name;
+	int res;
+
+	board_vendor = sensors_read_dmi_board_vendor();
+	if (!board_vendor)
+		return NULL;
+
+	board_name = sensors_read_dmi_board_name();
+	if (!board_name) {
+		free(board_vendor);
+		return NULL;
+	}
+
+	res = snprintf(path, NAME_MAX, "%s/sensors/dmi/%s/%s.conf",
+			LIBDIR, board_vendor, board_name);
+	if (res < 0 || res >= NAME_MAX)
+		return NULL;
+
+	free(board_vendor);
+	free(board_name);
+	return strdup(path);
+}
diff --git a/lib/sysfs.h b/lib/sysfs.h
index 3f4d83a2..790f647b 100644
--- a/lib/sysfs.h
+++ b/lib/sysfs.h
@@ -40,4 +40,7 @@ int sensors_write_sysfs_attr(const sensors_chip_name *name,
 			     const sensors_subfeature *subfeature,
 			     double value);
 
+/* Generate filename for config file based on DMI board vendor and board name */
+char *sensors_get_dmi_config_file(void);
+
 #endif /* def LIB_SENSORS_SYSFS_H */
```

jschwender, 2018-03-15:
I have also been playing around with dmi selection. It worked for me somehow. i found you need to differentiate hw version, BIOS version maybe BIOS release date. From what i have seen the data that vendors put in here is of low quality, but this will work for a certain number of boards. I have already placed some dmi info and the corresponding driver info in the config file (e.g.configs/Asus/Q1900-ITX.conf), and the detection is done by parsing these files. This way you can be more specific with the selection.

leighbb, 2018-03-15:
Hey, @jschwender, thanks for responding. Can you share the output of the following bash command on your Q1900?

```
(cd /sys/class/dmi/id && for i in *_* ; do echo -e "$i:\t$(cat $i)"; done)
```

I'm trying to collect as many examples as possible so I can understand how feasible my proposition is :-)

jschwender, 2018-03-15:
bios_date:	08/10/2015
bios_vendor:	American Megatrends Inc.
bios_version:	0403
board_asset_tag:	Default string
board_name:	H170-PLUS D3
board_serial:	150851441701418
board_vendor:	ASUSTeK COMPUTER INC.
board_version:	Rev X.0x
chassis_asset_tag:	Default string
chassis_serial:	Default string
chassis_type:	3
chassis_vendor:	Default string
chassis_version:	Default string
product_family:	Default string
product_name:	System Product Name
product_serial:	System Serial Number
product_uuid:	E1993D40-4DC8-11E5-9C78-305A3A0F3141
product_version:	System Version
sys_vendor:	System manufacturer

sorry right now i cannot access the q1900, i will do that tomorrow. This is from the machine i use rigth now.

groeck, 2018-03-16:
I am not in favor of creating a new configuration file hierarchy. Maybe we can find a means to include dmi information in existing configuration files. Either case, it would have to be dynamic; as was already noted, not all DMI parameters exist on all boards, and even if they exist they are not always set.

groeck, 2018-03-16:
bios_date:09/15/2017
bios_vendor:American Megatrends Inc.
bios_version:1.91
board_asset_tag:To be filled by O.E.M.
board_name:B350 TOMAHAWK (MS-7A34)
board_serial:H216518741
board_vendor:Micro-Star International Co., Ltd.
board_version:1.0
chassis_asset_tag:To be filled by O.E.M.
chassis_serial:To be filled by O.E.M.
chassis_type:3
chassis_vendor:Micro-Star International Co., Ltd.
chassis_version:1.0
product_family:To be filled by O.E.M.
product_name:MS-7A34
product_serial:To be filled by O.E.M.
product_uuid:00000000-0000-0000-0000-4CCC6AF43250
product_version:1.0
sys_vendor:Micro-Star International Co., Ltd.

bios_date:09/19/2017
bios_vendor:American Megatrends Inc.
bios_version:1.70
board_asset_tag:To be filled by O.E.M.
board_name:B350M MORTAR (MS-7A37)
board_serial:H416685934
board_vendor:MSI
board_version:1.0
chassis_asset_tag:To be filled by O.E.M.
chassis_serial:To be filled by O.E.M.
chassis_type:4
chassis_vendor:MSI
chassis_version:1.0
product_family:To be filled by O.E.M.
product_name:MS-7A37
product_serial:To be filled by O.E.M.
product_uuid:00000000-0000-0000-0000-309C2304DE10
product_version:1.0
sys_vendor:MSI

bios_date:12/01/2017
bios_vendor:American Megatrends Inc.
bios_version:F10
board_asset_tag:Default string
board_name:AB350M-Gaming 3-CF
board_serial:Default string
board_vendor:Gigabyte Technology Co., Ltd.
board_version:x.x
chassis_asset_tag:Default string
chassis_serial:Default string
chassis_type:3
chassis_vendor:Default string
chassis_version:Default string
product_family:Default string
product_name:AB350M-Gaming 3
product_serial:Default string
product_uuid:031B021C-040D-05E8-5D06-320700080009
product_version:Default string
sys_vendor:Gigabyte Technology Co., Ltd.

groeck, 2018-03-16:
"(cd /sys/class/dmi/id; sudo grep . *_*)" seems to be less complex and accomplish the same.
Also, you'll have to be careful to only use attributes which don't need root access.

jschwender, 2018-03-16:
bios_date:02/25/2014
bios_vendor:American Megatrends Inc.
bios_version:FA
board_asset_tag:To be filled by O.E.M.
board_name:C1037UN-EU
board_serial:To be filled by O.E.M.
board_vendor:Gigabyte Technology Co., Ltd.
board_version:x.x
chassis_asset_tag:To Be Filled By O.E.M.
chassis_serial:To Be Filled By O.E.M.
chassis_type:3
chassis_vendor:Gigabyte Technology Co., Ltd.
chassis_version:To Be Filled By O.E.M.
product_family:To be filled by O.E.M.
product_name:To be filled by O.E.M.
product_serial:To be filled by O.E.M.
product_uuid:03AA02FC-0414-0575-5206-670700080009
product_version:To be filled by O.E.M.
sys_vendor:Gigabyte Technology Co., Ltd.

bios_date:10/11/2012
bios_vendor:LENOVO
bios_version:6UET70WW (1.50 )
board_name:29123AG
board_vendor:LENOVO
board_version:Not Available
chassis_asset_tag:No Asset Information
chassis_type:10
chassis_vendor:LENOVO
chassis_version:Not Available
product_family:ThinkPad T410s
product_name:29123AG
product_uuid:E020ACF9-D8F6-E111-9EB6-B426B0AA384D
product_version:ThinkPad T410s
sys_vendor:LENOVO
_(serial numbers removed)_

bios_date:06/06/2008
bios_vendor:FUJITSU SIEMENS // Phoenix Technologies Ltd.
bios_version:4.06 R1.08.2440                 
board_name:D2440
board_serial:X72L-P00
board_vendor:FUJITSU SIEMENS
board_version:S26361-D2440
chassis_asset_tag:0x15                
chassis_serial:removed
chassis_type:23
chassis_vendor:FUJITSU SIEMENS
chassis_version:     
product_name:PRIMERGY RX330 S1   
product_serial:removed       
product_uuid:00000000-0000-0000-0000-0040D0C01B2D
product_version:GS02    
sys_vendor:FUJITSU SIEMENS                 
_( a very old server board, no sensors at all, uses IPMI BMC)_

bios_date:01/01/2007
bios_vendor:Seabios
bios_version:0.5.1
chassis_type:1
chassis_vendor:Red Hat
product_name:RHEV Hypervisor
product_serial:removed
product_uuid:603F0A15-C567-4703-8CE2-232139488D79
product_version:6.5-20150115.0.el6ev
sys_vendor:Red Hat

bios_date:05/23/2012
bios_vendor:American Megatrends Inc.
bios_version:090006 
board_name:Virtual Machine
board_vendor:Microsoft Corporation
board_version:7.0
chassis_type:3
chassis_vendor:Microsoft Corporation
chassis_version:7.0
product_name:Virtual Machine
product_version:7.0
sys_vendor:Microsoft Corporation

bios_date:06/18/2013
bios_vendor:Intel Corp.
bios_version:MUCDT10N.86A.0076.2013.0618.1644
board_asset_tag:Base Board Asset Tag
board_name:D2550MUD2
board_serial:removed
board_vendor:Intel Corporation
board_version:AAG81497-701
chassis_asset_tag:        
chassis_serial:        
chassis_type:3
chassis_vendor:        
chassis_version:        
product_name:        
product_serial:        
product_uuid:E728D33B-7F7F-11E3-B4AB-4437E6CC93DB
product_version:        
smbios_version:2.7
sys_vendor:        
_*driver: w83627dhg-isa-0290_

**Note** that if sensors gives data on voltages, the core voltages and RAM supply voltages depend on the specific Processor/RAM module installed on the board. This requires individual adjustments of the limits in the configs, depending on board+processor+ram combination.

jschwender, 2018-03-16:
bios_date:11/08/2016
bios_vendor:American Megatrends Inc.
bios_version:P1.80
board_asset_tag:        
board_name:Q1900-ITX
board_vendor:ASRock
board_version:        
chassis_asset_tag:To Be Filled By O.E.M.
chassis_serial:To Be Filled By O.E.M.
chassis_type:3
chassis_vendor:To Be Filled By O.E.M.
chassis_version:To Be Filled By O.E.M.
product_family:To Be Filled By O.E.M.
product_name:To Be Filled By O.E.M.
product_serial:To Be Filled By O.E.M.
product_version:To Be Filled By O.E.M.
sys_vendor:To Be Filled By O.E.M.
 
..this is the Q1900...

jschwender, 2018-03-16:
bios_date:01/08/2016
bios_vendor:American Megatrends Inc.
bios_version:V10.6
board_asset_tag:To be filled by O.E.M.
board_name:970A-G43 (MS-7693)
board_serial:To be filled by O.E.M.
board_vendor:MSI
board_version:3.0
chassis_asset_tag:To Be Filled By O.E.M.
chassis_serial:To Be Filled By O.E.M.
chassis_type:3
chassis_vendor:MSI
chassis_version:3.0
product_family:To be filled by O.E.M.
product_name:MS-7693
product_serial:To be filled by O.E.M.
product_version:3.0
sys_vendor:MSI

here is the output on another MSI board

leighbb, 2018-03-16:
Thanks for the output guys.  My goal is to help lm-sensors "just work" for the majority of people who don't want to mess about with configuring stuff.  Right now, sensors-detect does a great job of identifying the kernel modules etc. that you need to gain access to the sensors, but the output is generally mostly junk (in my experience, with the three boards I have access to).  That's why I think having the ability to include pre-canned configs that are automatically selected by the DMI information - if present - would be great.

I see there are a couple of ways to do this:
1. Include a database of configurations indexed by DMI information in some way and have the library automatically load the appropriate configuration (my initial suggestion) if it can find a match.
2. Include a database of configurations indexed by DMI information in some way and have sensors-detect automatically create the appropriate configuration.
3. Have a huge configuration file (or files) with extra syntax to select boards by DMI information.

I prefer option 1 but I could work with option 2 for now.  What you could also do with sensors-detect is if the DMI information is a perfect match (for some definition of perfect), then you could store in the configuration database the appropriate modules to load and offer to configure them directly instead of scanning.  That would be quick, simple and robust.

I think option 3 is a bad idea, as the parsing time would be excessive when the number of boards increases.

Finally, my initial suggestion (option 1) will always fall back to the current behaviour if no DMI information is present or a matching configuration cannot be found, which I think is important.

Anyway, I will now review the DMI data you've provided.

groeck, 2018-08-17:
No further action planned - see README.

olysonek, 2018-08-21:
> No further action planned - see README.

No longer relevant. Reopening. I'll review this issue as soon as I can.