lm-sensors/lm-sensors #92: ga-970a-d3p

pr0dukter, 2018-03-16:
chip "it8728-isa-0228"
    label  in0  "VCore"    # CPU Core Voltage
    label  in1  "VDDR"     # DDR2 1.8V
    label  in2  "+3.3V"    # VCC3
    label  in3  "+5V"      # VCC
    label  in4  "+12V"
    label  in5  "-12V"     # Not in the BIOS, maybe not really -12V
    label  in6  "-5V"      # Not in the BIOS, maybe not really -5V
    label  in7  "3.3VSB"     # VCCH
    label  in8  "VBat"

    label  fan1  "CPU Fan"
    label  fan2  "PWR Fan"
    label  fan3  "SYS Fan"

    label temp1  "NB Temp"
    label temp2  "CPU Temp"

    compute  in2  @ * (178/120), @ / (178/120)
    compute  in3  @ * (294/120), @ / (294/120)
    compute  in4  @ * (728/120), @ / (728/120)
    compute  in5 -@ * ((520/60) +1), -@ / ((520/60) +1)
    compute  in6 -@ * ((162/120) +1), -@ / ((162/120) + 1)
    compute  temp3 ((@ / 2.56) +36.4921875), -((@ / 2.56) +36.4921875)

    #compute 
    #compute  temp3 @ * ((95.01)*100.2)/100, @ / (95.01/27.43)*1.02
    #compute in6 (1+294/120)*@ - 4.096*294/120, (@ + 4.096*294/120)/(1+294/120)

   set in0_min  0.7
   set in0_max  1.5
   set in1_min  1.5   * 0.95
   set in1_max  1.5   * 1.05

   set in2_min 3.3 * 0.95
   set in2_max 3.3 * 1.05
   set in3_min 5.0 * 0.95
   set in3_max 5.0 * 1.05
   set in4_min 12 * 0.95
   set in4_max 12 * 1.05
   set in5_min (-1) * (12 * 0.95)
   set in5_max (-1) * (12 * 1.05)
   set in6_min (-5) * 0.95
   set in6_max (-5) * 1.05
   set in7_min  3     * 0.95
   set in7_max  3     * 1.05
   
   set temp1_min 0      # operates at significantly higher temps than temp1
   set temp1_max 65     # Default max fan speed reference point

   
   set temp2_min 0
   set temp2_max 55     # Tropical ambient temperature plus a bit

   
   #set temp3_min 0      # reports thermal margin, and reports -10C when idle
   #set temp3_max 65     # Bulldozer Published limit +3C
   #set temp3_type 5

groeck, 2018-03-17:
scaling factors are unusual for Gigabyte boards and would need some explanation.

pr0dukter, 2018-03-18:
i dont know what that means but all my values match the values in my bios
and without the conf file i can get only a couple readings and they are
horribly inaccurate, i used the math on the old old lm_sensors page from
archive.org to get those values, it teaches you how to make a conf file for
any chipset

theres no data sheet on the ISA bus, address *0x228* Chip `ITE IT8728F
Super IO

are you talking about the negavive voltage?  there was data in the raw
output so i made a very accurate average from different support forums on
the net  including some russia guys with many same equations, simply dont
use it if you dont want to, i just want to help other people with this
board get working fans

people who have this motherboard still cant get accurate voltage, temp
readings and therefore fancontrol, modify it however you want, you're in a
position to help those people  theres no thermal for many amd cpus either
so fans are a real nightmare on this board still in 2018

groeck, 2018-03-19:
I wasn't talking about the negative voltages. The question is why Gigabyte would choose factors such as 1.45 (178/120=29/20), 2.45 (294/120=49/20), or 6.06 (728/120=91/15) when simpler ones such as 1.5, 2.5, and 6.0 would do just as well. When the numbers are that unusual, I like to see the raw numbers (BIOS voltages, BIOS voltage changes, and raw voltage measures) to be able to confirm the calculations.
Note that the temp3 calculation looks wrong; I would have expected a multiplication on one side, and it is unusual to have the minus sign on one side only.

pr0dukter, 2018-03-21:
hey if you can improve it please help me man, i just did this by myself,
and i like i say it is accurate for me, please instruct me, do you know
what i am talking about when i refer to the archive . org lm sensors page
that said how to come up with these equations?

pr0dukter, 2018-05-22:
ok ive updated it, this should be a propper file

---begin
`chip "it8728-isa-0228"
    label  in0  "VCore"    # CPU Core Voltage
    label  in1  "VDDR"     # DDR2 1.8V
    label  in2  "+3.3V"    # VCC3
    label  in3  "+5V"      # VCC
    label  in4  "+12V"
    label  in5  "-12V"     # Not in the BIOS, maybe not really -12V
    label  in6  "-5V"      # Not in the BIOS, maybe not really -5V
    label  in7  "3.3VSB"     # VCCH
    label  in8  "VBat"

    label  fan1  "CPU Fan"
    label  fan2  "PWR Fan"
    label  fan3  "SYS Fan"

    label temp1  "NB Temp"
    label temp2  "CPU Temp"

    compute  in2  @ * (178/120), @ / (178/120)
    compute  in3  @ * (294/120), @ / (294/120)
    compute  in4  @ * (728/120), @ / (728/120)
    compute  in5 -@ * ((520/60) +1), -@ / ((520/60) +1)
    compute  in6 -@ * ((162/120) +1), -@ / ((162/120) + 1)
    compute  temp3 ((@ / 2.56) +36.4921875), -((@ / 2.56) +36.4921875)

   set in0_min  0.825 * 0.95
   set in0_max  1.425 * 1.05
   set in1_min  1.5   * 0.95
   set in1_max  1.5   * 1.05

   set in2_min 3.3 * 0.95
   set in2_max 3.3 * 1.05
   set in3_min 5.0 * 0.95
   set in3_max 5.0 * 1.05
   set in4_min 12 * 0.95
   set in4_max 12 * 1.05
   set in5_min (-12) * 0.95
   set in5_max (-12) * 1.05
   set in6_min (-5) * 0.95
   set in6_max (-5) * 1.05
   set in7_min  3 * 0.95
   set in7_max  3 * 1.05
   
   set temp1_min 0      # operates at significantly higher temps than temp2
   set temp1_max 65     # Default max fan speed reference point

   
   set temp2_min 0
   set temp2_max 55     # Tropical ambient temperature plus a bit

   
#   set temp3_min 0      # reports thermal margin, and reports -10C when idle triggering mobo alarm
#   set temp3_max 65     # Bulldozer Published limit +3C triggers alarm
#   set temp3_type 5

`

cooper151288, 2018-05-23:
@pr0dukter 
I have a similar board, GA-970a-DS3. Mine needs scaling. Ill test your values (with the adjustments to which temp input is which).
Out of interest, is there a fixed offset between it87's temp input and k10temp? I've used the debug mode for the CPU sensor and I get an offset of 12 "degrees" unless suspend has happened prior to boot

cooper151288, 2018-05-23:
results are in: 
this config file has some matches with GA-970A-DS3 (im using in-tree it87 and k10temp) but:

temp1 is (presumably) a thermistor which measures the case air and can't be configured (it controls the max value of the fan speed and (i think) the PSU fan) 

My temp3 is a constant 12 "degrees" over Tctl from k10temp unless you set debug mode in d18f3 0xa4, 17:16 =11b and set it in the 70s+ using bits 31:21 (might be emergency over-temp logic)

No idea if temp2 is right, my BIOS has a "system" and "CPU" temperature

my 5v has different scaling, i get 8V on that scaling.

I suspect the -12V has something to do with my systems "hidden" fan control, as it dips when the fan really cranks up. Either that or the PSU is creaking under load, only to need to make the fans (CPU and PSU) spin faster. Would not surprise me if they made the fan run partially +12 to -12 when very hot.

Might be worth adding that the it8728f on my board wouldn't let me set auto_channels_temp (though they showed up in sysfs) or any of the "newer" fancontrol (which didn't show) until late in linux 3.x (think it was when 8732 was added - or otherwise, definitely early 2016) so its probably a funky implementation (never, ever buy a cheap mobo!)

In short, 2 "similar" boards with the same superIO have wildly differing implementations. Thanks Gigabyte!

cooper151288, 2018-05-24:
Whatever it was, it was fixed when a different chip was added to it87. Is there a way of putting offsets/scaling to thermistors (that the firmware sees ie not a compute)? My temp1 is definitely a thermistor (shows -ve values constantly when set as diode). Would like to know this so I can confirm what I think is true of this mobos thermal management.
If it was a bad driver before, it was the same autopwm I saw in BIOS and windows. My pwm1 and pwm3 are on the same fan  (1 voltage 3 actual pwm), pwm1 tied to temp3 (cpu). Pwm3 and pwm2 (again actally a voltage) are by default tied to this thermistor.
when temp1 gets up to about 33+, fan1 can do up tk 7k rpm, wheras at 25, it'll do 3.5k rpm

cooper151288, 2018-05-25:
The offsets work on temp 2 and 3 (believe these to be motherboard/socket and AMD-TSI @ Tctl + 12) respectively. Temp1 will ignore anything fed as offset while set as thermistor.

If temp1 is set as diode, it is always negative. The offset then woks but an overflow occurs at -128 and 0, which keeps it negative. In this mode it becomes more negative when the CPU gets hotter, with much more response than a thermistor.

If temp1 is set as PECI/AMD-TSI it does not update.

groeck, 2018-05-25:
Confirmed with the datasheet that the temperature offsets only work for diodes. Kind of odd, but it is what it is. I don't find any other offset registers in the datasheet.

cooper151288, 2018-05-25:
yeah I didnt see any other offsets. Offset works as expected on my thermal diode and also as unexpected on the cpu temperature (it87 calls it a PECI unless i say otherwise and its not bothered what type I assign to it. 
I did see the tacho based closed loop fan control- might be fun to play with.
I can actually set offsets on my thermal diode and "PECI"/TSI/firmware based PCI reader (18:3.0xa4 confirms this measurement is Tctl+12 on my board) even if i call them thermistors first!
These registers appear to be reserved rather than read only if thermistor so it may be the joys of undefined behaviour in firmware (seen a bad flash make a stepper stage drive its motors backwards today). Or theres more that the chip knows, im not sure. Without non-evaluation datasheets (8728) and datasheets that are referenced in the BKDG (15 power and thermal?) all i can really do is speculate.

cooper151288, 2018-05-25:
whoops. temp2 (proably diode) will not apply an offset if i call it a thermistor.
temp3 (tctl+12) will apply offset regardless of what you call it

groeck, 2018-05-26:
What is the default offset on temp3 ? Is it 12 degrees C ? Also, what is the default reported type ?

cooper151288, 2018-05-26:
default offset is 0 on all 3. Presume temp2 is "motherboard temp" in BIOS. Temp3 will set the CPU warning temperature beep if the BIOS option is set for it. Looking at AMDs thermal design guide for FS1b (doc 52121) temp 1 is probably a control temperature- looks like the aim of this is to have the same level of performance all year.
temp3 appears as PECI as default, but always seems to report sane values (value k10temp reports + 12 + offset, with ~ 1.5s lag)

groeck, 2018-05-26:
Hmmm .. since you have an AMD CPU, this should show AMDTSI (type 5). Can you send me a register dump, by any chance ?

cooper151288, 2018-05-27:
yeah which regs should i dump? ive got the superiotool from coreboot which supports my chip if it helps

groeck, 2018-05-27:
See README file for the it87 driver how to create dumps, though the output of superiotool would be even better. I would need register dumps for the configuration register space as well as the environmental controller register space.

pr0dukter, 2018-05-27:
sorry to interrupt you guys, on pwn mode my fan2 and fan3 hit a max of 800rpm,  is there a way to make it go faster in pwm? i tried slope, would i have to go to voltage to increase speed?  they should be hooked to 12v on mine but are plugged into the 5v on board 3 and 4 pin spots,  question is way to increase rpm without spliting wire and running power to 12v? if not no biggie

pr0dukter, 2018-05-27:
also my temp_3 is always a negative # on POST so sets off alarm if i put min/max values in, same on yours @cooper151288 cooper151288?

groeck, 2018-05-27:
Unfortunately those are only the configuration registers. Any chance to also send a dump of the environmental controller registers ?
For temp3, does it match the temperature shown with k10temp ? Either case, older AMD CPUs have really bad temperature measurement accuracy, especially at low temperatures.

cooper151288, 2018-05-27:
the ec reg just says:
BRAM 0x0000 (i think).
Ill try and do it with the info from it87. Ive tried getting ec info other ways and havent had any luck

temp3 = d18f3 0xa4 31:21 rounded to nearest degree + 12 degrees + the configured offset

CPU is an FX-8350, fam15h model 2

groeck, 2018-05-27:
The sensors commend tells you the address to look for. Examples:
```
it8686-isa-0a40
it8792-isa-0a60
```
The isadump command then gives you the data I am looking for. Example:
```
sudo isadump 0xa45 0xa46
```

cooper151288, 2018-05-27:
ok ill grab that when im by my computer. ill see what happens when coming back from suspend too, see if anything interesting happens. might make sense to check some of the CPUs PCI space while im at it

cooper151288, 2018-05-27:
is this the right data?
it8728-isa-0228
Adapter: ISA adapter
VCore:        +0.86 V  (min =  +0.00 V, max =  +1.50 V)
VDDR:         +1.46 V  (min =  +1.43 V, max =  +1.57 V)
+3.3V:        +3.06 V  (min =  +3.13 V, max =  +3.47 V)
+5V:          +8.17 V  (min =  +0.00 V, max = +14.99 V)
+12V:        +12.08 V  (min = +11.43 V, max = +12.59 V)
-12V:        -11.72 V  (min = -11.37 V, max = -12.64 V)
-5V:          -5.25 V  (min =  -4.74 V, max =  -5.25 V)
3.3VSB:       +3.38 V  (min =  +2.86 V, max =  +3.14 V)
VBat:         +3.17 V
CPU Fan:     1018 RPM  (min =    0 RPM)
System Fan:   894 RPM  (min =    0 RPM)
SYS Fan 2:      0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
Case Temp:    +27.0°C  (low  =  +0.0°C, high = +40.0°C)  sensor = thermistor
NB Temp:      +29.0°C  (low  =  +0.0°C, high = +70.0°C)  sensor = thermal diode
CPU Temp:     +15.0°C  (low  =  +0.0°C, high = +85.0°C)  sensor = Intel PECI
intrusion0:  OK

radeon-pci-0100
Adapter: PCI adapter
temp1:        +32.0°C  (crit = +120.0°C, hyst = +90.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +15.9°C  (high = +70.0°C)
                       (crit = +90.0°C, hyst = +87.0°C)

radeon-pci-0500
Adapter: PCI adapter
temp1:        +29.5°C  (crit = +120.0°C, hyst = +90.0°C)

matt@GA-970A-DS3:~$ su
Password:
root@GA-970A-DS3:/home/matt# isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 87 28 01 08 00 00 73 10 00 80 00 40 01 00 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 00 00 02 20 00 00 00 00 00 00 00 00 00 00 00 00
70: 00 00 00 38 00 04 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 20 10 00 00 40 00 00
c0: 00 43 00 00 00 00 00 00 00 22 06 00 00 40 00 00
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 00 00 07 07 07 00 00 00 7e
f0: 00 00 00 00 00 00 1c 00 00 01 00 00 80 00 00 00
root@GA-970A-DS3:/home/matt# isadump 0x22d 0x22e
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x22d and data register 0x22e.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 11 00 c4 00 ff ff 00 37 ff 87 48 0f 27 83 d2 ff
10: ff ff ff 73 d7 82 80 7f 02 02 ff ff ff ff ff ff
20: 48 7a ac 8b a6 65 ba 8d 84 1c 20 13 ba cc c3 c3
30: 7d 00 83 77 c3 b0 ff 00 ad 9d 6d 62 ba a8 83 77
40: 28 00 46 00 55 00 5f 40 ad 6a d4 00 ff ff ff ff
50: ff ce 7f 7f 7f c0 00 fb 90 f4 3e 12 e4 00 00 00
60: 00 14 41 4e 20 03 0f ff 00 14 41 46 20 03 0f ff
70: 00 14 41 40 20 03 0f ff ff ff ff ff ff ff ff ff
80: 00 00 ff ff 00 00 ff ff 00 00 00 00 00 ea 00 73
90: ff 00 00 00 ff 00 00 00 42 99 00 00 01 e4 f2 1f
a0: 15 7d 7d 7d 7d 7d 7d ff ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

cooper151288, 2018-05-27:
and here is my sysfs for it, after my sensors3.conf is applied:
root@GA-970A-DS3:/sys/class/hwmon/hwmon2# grep -r . | sort
grep: power/autosuspend_delay_ms: Input/output error
alarms:0
fan1_alarm:0
fan1_beep:0
fan1_input:1323
fan1_min:0
fan2_alarm:0
fan2_beep:0
fan2_input:979
fan2_min:0
fan3_alarm:0
fan3_beep:0
fan3_input:0
fan3_min:0
fan5_alarm:0
fan5_beep:0
fan5_input:0
fan5_min:0
in0_alarm:0
in0_beep:0
in0_input:864
in0_max:1500
in0_min:0
in1_alarm:0
in1_beep:0
in1_input:1464
in1_max:1572
in1_min:1428
in2_alarm:0
in2_beep:0
in2_input:2064
in2_max:2340
in2_min:2112
in3_alarm:0
in3_beep:0
in3_input:3336
in3_label:+3.3V
in3_max:6120
in3_min:0
in4_alarm:0
in4_beep:0
in4_input:1992
in4_max:2076
in4_min:1884
in5_alarm:0
in5_beep:0
in5_input:1212
in5_max:1308
in5_min:1176
in6_alarm:0
in6_beep:0
in6_input:2232
in6_max:2232
in6_min:2016
in7_alarm:0
in7_beep:0
in7_input:3384
in7_label:3VSB
in7_max:3144
in7_min:2856
in8_input:3168
in8_label:Vbat
intrusion0_alarm:0
name:it8728
power/async:disabled
power/control:auto
power/runtime_active_kids:0
power/runtime_active_time:0
power/runtime_enabled:disabled
power/runtime_status:unsupported
power/runtime_suspended_time:0
power/runtime_usage:0
pwm1:78
pwm1_auto_channels_temp:4
pwm1_auto_point1_temp:0
pwm1_auto_point1_temp_hyst:-3000
pwm1_auto_point2_temp:20000
pwm1_auto_point3_temp:65000
pwm1_auto_slope:32
pwm1_auto_start:78
pwm1_enable:2
pwm1_freq:11718
pwm2:70
pwm2_auto_channels_temp:1
pwm2_auto_point1_temp:0
pwm2_auto_point1_temp_hyst:-3000
pwm2_auto_point2_temp:20000
pwm2_auto_point3_temp:65000
pwm2_auto_slope:32
pwm2_auto_start:70
pwm2_enable:2
pwm2_freq:23437
pwm3:64
pwm3_auto_channels_temp:1
pwm3_auto_point1_temp:0
pwm3_auto_point1_temp_hyst:-3000
pwm3_auto_point2_temp:20000
pwm3_auto_point3_temp:65000
pwm3_auto_slope:32
pwm3_auto_start:64
pwm3_enable:0
pwm3_freq:11718
temp1_alarm:0
temp1_beep:1
temp1_input:29000
temp1_max:40000
temp1_min:0
temp1_offset:0
temp1_type:4
temp2_alarm:0
temp2_beep:1
temp2_input:32000
temp2_max:70000
temp2_min:0
temp2_offset:-5000
temp2_type:3
temp3_alarm:0
temp3_beep:1
temp3_input:19000
temp3_max:85000
temp3_min:0
temp3_offset:-12000
temp3_type:6

groeck, 2018-05-28:
Ok, I am a bit at loss. The register contents tell me that temp3_type should be reported as 5, not 6. No idea how it can report a value of 6. I'll have to sleep over that and maybe even write a simulator.

cooper151288, 2018-05-28:
If i load your upstream module, i get temp3_type=3 (which would be consistent with offsets behaving). In-tree module gives 6.

cooper151288, 2018-05-28:
No it87 loaded (but has been loaded)  
   0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 11 00 c4 00 ff ff 00 37 ff 87 48 0f 27 c3 96 ff
10: ff ff ff 73 d7 82 80 7f 01 02 ff ff ff ff ff ff
20: 48 7a ac 8b a6 65 ba 8d 84 1e 20 13 ba cc c0 c0
30: 7d 00 83 77 c3 b0 ff 00 ad 9d 6d 62 ba a8 83 77
40: 28 00 46 00 55 00 5f 40 ad 6a d4 00 ff ff ff ff
50: ff ce 7f 7f 7f c0 00 fb 90 f4 3e 12 e4 00 00 00
60: 00 14 41 4e 20 03 0f ff 00 14 41 46 20 03 0f ff
70: 00 14 41 40 20 03 0f ff ff ff ff ff ff ff ff ff
80: 00 00 ff ff 00 00 ff ff 00 00 00 00 00 1d 00 05
90: ff 00 00 00 ff 00 00 00 42 99 00 00 01 e4 f2 1f
a0: 15 7d 7d 7d 7d 7d 7d ff ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

In-tree it87
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 11 00 c4 00 ff ff 00 37 ff 87 48 0f 27 be 96 ff
10: ff ff ff 73 d7 82 80 7f 01 02 ff ff ff ff ff ff
20: 48 7a ac 8b a6 65 ba 8d 84 1e 20 13 ba cc c3 c3
30: 7d 00 83 77 c3 b0 ff 00 ad 9d 6d 62 ba a8 83 77
40: 28 00 46 00 55 00 5f 40 ad 6a d4 00 ff ff ff ff
50: ff ce 7f 7f 7f c0 00 fb 90 f4 3e 12 e4 00 00 00
60: 00 14 41 4e 20 03 0f ff 00 14 41 46 20 03 0f ff
70: 00 14 41 40 20 03 0f ff ff ff ff ff ff ff ff ff
80: 00 00 ff ff 00 00 ff ff 00 00 00 00 00 ea 00 7f
90: ff 00 00 00 ff 00 00 00 42 99 00 00 01 e4 f2 1f
a0: 15 7d 7d 7d 7d 7d 7d ff ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

Your later it87  
   0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 11 00 c4 00 ff ff 00 37 ff 87 48 0f 27 b4 96 ff
10: ff ff ff 73 d7 82 80 7f 01 02 ff ff ff ff ff ff
20: 48 7a ac 8b a6 65 ba 8d 84 1e 20 13 b9 cc c3 c3
30: 7d 00 83 77 c3 b0 ff 00 ad 9d 6d 62 ba a8 83 77
40: 28 00 46 00 55 00 5f 40 ad 6a d4 00 ff ff ff ff
50: ff ce 7f 7f 7f c0 00 fb 90 f4 3e 12 e4 00 00 00
60: 00 14 41 4e 20 03 0f ff 00 14 41 46 20 03 0f ff
70: 00 14 41 40 20 03 0f ff ff ff ff ff ff ff ff ff
80: 00 00 ff ff 00 00 ff ff 00 00 00 00 00 1c 00 65
90: ff 00 00 00 ff 00 00 00 42 99 00 00 01 e4 f2 1f
a0: 15 7d 7d 7d 7d 7d 7d ff ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

groeck, 2018-05-29:
Found the bug. Updated the driver; it now reports type 5 as expected for temp3. In all register dumps, temp3_offset is set to -12 degrees C, which may explain the difference to Tctl.

groeck, 2018-05-29:
The upstream driver doesn't know about AMDTSI and always reports PECI.

cooper151288, 2018-05-29:
My -12 offset is due to my sensors3.conf. My bad, probably should have mentioned that or attached my conf. Take it you got -5 on my temp2 as well?

If we still get -12 from that info, we can be resonably convinced that: 

my sensor will accept values when inputted into the "thermal diode" offset register and work with that.

my prediction of the it8728f implementation in GA-970A-DS3 is correct. ie temp3=round(tctl)+12+offset, unless tctl is excessive (80+), where it really cranks up. should be straightforward to graph that as long as I bear in mind the fact that it may go to THERMTRIP and shut down within 500ms at some point (supposed to be at 90)

I'll give your new driver version a go when i've got a bit more time. it should work even better if you've fixed something.
I'll also have to see what happens when I come back from suspend, as some weird things happen:

Tctl_max changes from 90 to 70
temp3 loses its 12 degree additional offset.

This boards ACPI support is awful (a very buggy 1.0b!) so thats probably not helping matters but it looks like something is being stored somewhere incorrectly. may try a BIOS flash w/ cleared DMI + CMOS.
Other than that, i might just get a better system, as fun as messing about with this may be.

groeck, 2018-05-29:
Only difference with my latest driver will be that it reports temperature type 5 instead of 6 for temp3. That was due to a a missing "else" statement in the previous driver version.

Yes, I did see -5 on temp2_offset.
 
temp1_alarm:0
temp1_beep:1
temp1_input:28000
temp1_max:40000
temp1_min:0
temp1_offset:0
temp1_type:4
temp2_alarm:0
temp2_beep:1
temp2_input:32000
temp2_max:70000
temp2_min:0
temp2_offset:-5000
temp2_type:3
temp3_alarm:0
temp3_beep:1
temp3_input:19000
temp3_max:85000
temp3_min:0
temp3_offset:-12000
temp3_type:5

It would be really important to get unmodified (no sensors -s) register dumps from right after booting and after a suspend/resume cycle.

cooper151288, 2018-05-30:
yeah no problems I'll get that done when I'm at my PC. If I blacklist it87, revert sensors3.conf to default and reboot will that be what you need to see?

groeck, 2018-05-30:
Output from "superiotool -da" and "isadump 0x22d 0x22e". Please make sure it is a cold reboot (power cycle) for the first reboot. Thanks!

cooper151288, 2018-06-06:
I'll do the same with a suspend and a true cold boot (ie power off, cord out, leave it a min for caps to discharge) over the coming days if i can't do it today.

I've raised a support ticket with Gigabyte over my suspend/ACPI issues and, despite using an unsupported CPU on an unsupported OS on a mobo that is 3 years out of warranty, they actually seem interested so i'll send them a load of register dumps and my ACPI tables and see if they can provide anything useful.

Was my issue with temp3_type reflected in the ec dumps when I tried the upstream it87 or was the driver simply reporting the wrong value?

cooper151288, 2018-06-06:
This is after loading the updated it87.ko

root@GA-970A-DS3:/btrfs/git/it87# insmod ./it87.ko
root@GA-970A-DS3:/btrfs/git/it87# sensors
fam15h_power-pci-00c4
Adapter: PCI adapter
power1:       28.82 W  (crit = 125.19 W)

it8728-isa-0228
Adapter: ISA adapter
in0:          +0.86 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +1.49 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.06 V  (min =  +0.00 V, max =  +3.06 V)
+3.3V:        +3.34 V  (min =  +0.00 V, max =  +6.12 V)
in4:          +1.99 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.19 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +2.23 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.38 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.17 V
fan1:        2213 RPM  (min =    0 RPM)
fan2:        1004 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermal diode
temp3:        +26.0°C  (low  = +127.0°C, high = +90.0°C)  sensor = AMD AMDSI
intrusion0:  OK

radeon-pci-0100
Adapter: PCI adapter
temp1:        +38.0°C  (crit = +120.0°C, hyst = +90.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +14.5°C  (high = +70.0°C)
                       (crit = +90.0°C, hyst = +87.0°C)

radeon-pci-0500
Adapter: PCI adapter
temp1:        +36.5°C  (crit = +120.0°C, hyst = +90.0°C)

root@GA-970A-DS3:/btrfs/git/it87# superiotool -da
superiotool r4.8-370-gbce6742a9d
Found ITE IT8728F (id=0x8728, rev=0x1) at 0x2e
Register dump:
idx   val    def
0x02: 0x00   (NA)
0x07: 0x07   (NA)
0x20: 0x87   (0x87)
0x21: 0x28   (0x28)
0x22: 0x01   (0x01)
0x23: 0x08   (0x00)
0x24: 0x00   (0x00)
0x2b: 0x40   (MM)
0x2e: 0x00   (0x00)
0x2f: 0x00   (0x00)

LDN 0x00 (Floppy)
idx   val    def
0x30: 0x00   (0x00)
0x60: 0x00   (0x03)
0x61: 0x00   (0xf0)
0x70: 0x00   (0x06)
0x74: 0x04   (0x02)
0xf0: 0x00   (0x00)
0xf1: 0x80   (0x00)

LDN 0x01 (COM1)
idx   val    def
0x30: 0x00   (0x00)
[... 252 lines cut ...]
pwm3_auto_slope:32
pwm3_auto_start:64
pwm3_enable:0
pwm3_freq:11718
temp1_alarm:0
temp1_beep:1
temp1_input:30000
temp1_max:127000
temp1_min:127000
temp1_offset:0
temp1_type:4
temp2_alarm:0
temp2_beep:1
temp2_input:33000
temp2_max:127000
temp2_min:127000
temp2_offset:0
temp2_type:3
temp3_alarm:0
temp3_beep:1
temp3_input:26000
temp3_max:90000
temp3_min:127000
temp3_offset:0
temp3_type:5

cooper151288, 2018-06-06:
after a true cold boot:
superiotool r4.8-370-gbce6742a9d                                                                                              
Found ITE IT8728F (id=0x8728, rev=0x1) at 0x2e
Register dump:
idx   val    def
0x02: 0x00   (NA)
0x07: 0x07   (NA)
0x20: 0x87   (0x87)
0x21: 0x28   (0x28)
0x22: 0x01   (0x01)
0x23: 0x08   (0x00)
0x24: 0x00   (0x00)
0x2b: 0x40   (MM)
0x2e: 0x00   (0x00)
0x2f: 0x00   (0x00)

LDN 0x00 (Floppy)
idx   val    def
0x30: 0x00   (0x00)
0x60: 0x00   (0x03)
0x61: 0x00   (0xf0)
0x70: 0x00   (0x06)
0x74: 0x04   (0x02)
0xf0: 0x00   (0x00)
0xf1: 0x80   (0x00)

LDN 0x01 (COM1)
idx   val    def
0x30: 0x00   (0x00)
0x60: 0x00   (0x03)
0x61: 0x00   (0xf8)
0x70: 0x00   (0x04)
0xf0: 0x00   (0x00)
0xf1: 0x50   (0x50)

LDN 0x02 (COM2)
idx   val    def
0x30: 0x00   (0x00)
0x60: 0x00   (0x02)
0x61: 0x00   (0xf8)
0x70: 0x00   (0x03)
0xf0: 0x00   (0x00)
0xf1: 0x50   (0x50)

LDN 0x03 (Parallel port)
idx   val    def
0x30: 0x00   (0x00)
0x60: 0x00   (0x03)
0x61: 0x00   (0x78)
0x62: 0x00   (0x07)
0x63: 0x00   (0x78)
0x70: 0x00   (0x07)
0x74: 0x04   (0x03)
0xf0: 0x08   (0x03)

LDN 0x04 (Environment controller)
idx   val    def
0x30: 0x01   (0x00)
0x60: 0x02   (0x02)
0x61: 0x28   (0x90)
0x62: 0x00   (0x02)
0x63: 0x00   (0x30)
0x70: 0x00   (0x09)
0xf0: 0x00   (0x00)
0xf1: 0x00   (0x00)
0xf2: 0x0e   (0x00)
0xf3: 0x00   (0x00)
0xf4: 0x8a   (0x00)
0xf5: 0x3e   (MM)
0xf6: 0x1b   (MM)
[... 209 lines cut ...]
pwm3_auto_slope:32
pwm3_auto_start:64
pwm3_enable:0
pwm3_freq:11718
temp1_alarm:0
temp1_beep:1
temp1_input:30000
temp1_max:127000
temp1_min:127000
temp1_offset:0
temp1_type:4
temp2_alarm:0
temp2_beep:1
temp2_input:41000
temp2_max:127000
temp2_min:127000
temp2_offset:0
temp2_type:3
temp3_alarm:0
temp3_beep:1
temp3_input:49000
temp3_max:90000
temp3_min:127000
temp3_offset:0
temp3_type:6

pr0dukter, 2018-06-20:
personally i tell arch what order to load my fam15h_power k10temp and it87 because if i dont my fan control values in /sys are always jumbled, i guess this is due to a race condition? anyhow thats why why i have statements telling them to load, i dont use lmsensors.service from systemd either because it doesnt seem to fix the out of order problem only the /etc/modules and /etc/modprobe  settings guarantee for me

also i am forced to amd_iommu=on and iommu=pt or various other iommu= flags for my amdgpu to not give errors at boot, somthing about -10 -20 spirv amd something

groeck, 2018-06-20:
@pr0dukter: Maybe I am missing the context, but there is no guarantee that hwmon device numbering is consistent across boots. libsensors is supposed to take that into account. Using hwmon attributes other than through libsensors is not supported.

cooper151288, 2018-06-22:
Ive started translating the dumped settings using the datasheet. Gigabyte "can't reproduce" any issue i told them about and told me to use windows instead. I'll get some information out of them regarding how BIOS sets a few registers. i think my "after resume from suspend" is correct as my temp3 matches k10temp and it makes sense for HTC to kick in before THERMTRIP.
Can you think of any registers I should ask about? Knowing what BIOS does (or is supposed to do) may be useful for drivers

pr0dukter, 2018-06-22:
i just compiled sensorsio, want my data as well, can you tell me the commands to run

cooper151288, 2018-07-02:
My ticket with Gigabyte has been referred to their BIOS team now as I've asked specific questions as to how BIOS configures registers on it8728f and fam15h cpu, so they should report back with info on BIOS quirks on GA-970A-DS3 rev 1.1

groeck, 2018-08-17:
No further action planned - see README.

olysonek, 2018-08-21:
> No further action planned - see README.

No longer relevant. Reopening. I'll review this issue as soon as I can.

akontsevich, 2020-01-05:
Where to take config for Gigabyte 970A-DS3P?

0rbadvent, 2020-01-05:
What exactly are you asking?

Sent from ProtonMail mobile

-------- Original Message --------

akontsevich, 2020-01-05:
Config for this chip/board to understand sensors meanings/names.

forgetso, 2020-11-16:
@akontsevich Try using pwmconfig: https://askubuntu.com/questions/22108/how-to-control-fan-speed

I have the same board and I was able to get it working with the following config.

```
Settings of hwmon0/pwm1:
  Depends on hwmon0/temp2_input
  Controls 
  MINTEMP=26
  MAXTEMP=60
  MINSTART=150
  MINSTOP=100
  MINPWM=0
  MAXPWM=255
```

akontsevich, 2020-11-16:
> ```
> Settings of hwmon0/pwm1:
>   Depends on hwmon0/temp2_input
>   Controls 
>   MINTEMP=26
>   MAXTEMP=60
>   MINSTART=150
>   MINSTOP=100
>   MINPWM=0
>   MAXPWM=255
> ```

@forgetso where to put this and how to enable?

forgetso, 2020-11-17:
@akontsevich Go to the link and follow the instructions for using `pwmconfig`. This will help create the config for your machine.

If you simply want to use my config, you can create the file `/etc/fancontrol` and enable with `sudo service fancontrol start`

0rbadvent, 2020-11-17:
So do i

Sent from ProtonMail mobile

-------- Original Message --------