lm-sensors/lm-sensors #93: Need help to understand sensors info

berillions, 2018-03-17:
Hi,

I have an Asus PRIME X370-Pro + Ryzen 5 1600. I need to compile it87 module from git to see all info.

The sensors output is like this : 
``babaorheum@debian-desktop:~$ sensors
it8665-isa-0290
Adapter: ISA adapter
in0:          +0.36 V  (min =  +0.24 V, max =  +1.17 V)
in1:          +2.46 V  (min =  +2.47 V, max =  +2.54 V)
in2:          +1.97 V  (min =  +0.68 V, max =  +1.30 V)
in3:          +1.98 V  (min =  +2.06 V, max =  +1.85 V)
in4:          +0.03 V  (min =  +2.08 V, max =  +1.73 V)
in5:          +0.03 V  (min =  +1.73 V, max =  +2.71 V)
in6:          +0.03 V  (min =  +0.60 V, max =  +2.51 V)
3VSB:         +3.23 V  (min =  +2.03 V, max =  +2.75 V)
Vbat:         +3.16 V  
+3.3V:        +3.25 V  
fan1:        1313 RPM  (min =   11 RPM)
fan2:        1002 RPM  (min =   24 RPM)
fan3:           0 RPM  (min =   26 RPM)  ALARM
fan5:           0 RPM  (min =   -1 RPM)  ALARM
fan6:        1303 RPM  (min =   -1 RPM)
temp1:        +30.0°C  (low  = -35.0°C, high = -41.0°C)
temp2:        +37.0°C  (low  = +78.0°C, high = -99.0°C)  sensor = thermistor
temp3:        +32.0°C  (low  = +94.0°C, high = -22.0°C)  sensor = thermistor
intrusion0:  ALARM

amdgpu-pci-2800
Adapter: PCI adapter
fan1:        1086 RPM
temp1:        +35.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +30.0°C  (high = +70.0°C)

amdgpu-pci-2900
Adapter: PCI adapter
fan1:         777 RPM
temp1:        +36.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM``

I would like to know if it's possible to have the real name for in0 to in3. When i search on Google, i found that in0=VCore. Why i have not this name ?

"k10temp-pci-00c3" info is for my Ryzen CPU but about temp{1,2,3} in "it8665-isa-0290", it's for what ?

In fact, i need help to understand correctly these informations.

Thanks
Maxime

groeck, 2018-03-17:
You can enter configuration data for your board in /etc/sensors3.conf or better in /etc/sensors.d/<your_name>. Something like

chip "it8665-*"
    label temp1 "MB CPU Temp"
    label in0 "Vcore"

would be a start.

leighbb, 2018-03-18:
Hi @berillions.  It is useful to go into the BIOS and take photos of the configured voltages, temperatures and fan speeds so that you can compare them with the output you are seeing.  Let's say you look at the page that displays the temperature readings and it displays: `M/B Temperature: 30degC`.  You could then add a line like this into the config:
```
    label temp1 "M/B Temperature"
```
Similarly, if the BIOS display something like `Chassis Fan 1: 1002 RPM` you can add the following:
```
    label fan2 "Chassis Fan 1"
```
If you can't work out which fan is which, you could temporarily disconnect one at a time.  If you know there is an unused fan connector on your motherboard, then you could temporarily move a fan to that connector.  This will allow you to determine which fan connectors correspond to which fan displayed in the sensors output.  You can then label them appropriately and create ignore lines for the ones which are definitely not used.

If you cannot correlate a particular input, or temperature, it's probably best to add an appropriate ignore line for the time being.

closed 2018-04-10 by groeck