lm-sensors/lm-sensors #99: lm-sensors interrupts on Dell XPS 13 take too long

vbotka, 2018-04-08:
The problem manifests itself with mouse cursor lags and jumps on desktop. I experience such a lag on all input devices. The problem disappeared when I disabled lm-sensor provider in psensor http://wpitchoune.net/psensor/. Probably this is the reason for [1,2].

[1] Mouse cursor lags and jumps about erratically on desktop
https://ubuntuforums.org/showthread.php?t=2381522

[2] Dell XPS 13 - Jumping Mouse (Ubuntu)
https://www.reddit.com/r/Dell/comments/7rtdmz/dell_xps_13_jumping_mouse_ubuntu/

Thank you,
                      -vlado

groeck, 2018-04-10:
This is not an lm-sensors problem. The problem is that Dell doesn't support Linux, and that the Dell SMM API to read sensor data slows down the system significantly, especially when reading fan speeds. See https://bugzilla.kernel.org/show_bug.cgi?id=195751 for a severe example. There is nothing we can do about this; the only available remedy is to not load the dell-smm-hwmon driver on affected systems.

ajbogh, 2018-08-06:
@groeck I had a similar problem on a custom built computer (specs below). lm-sensors and system-monitor were the culprit and uninstalling them solved the problem. I'd love for it to be fixed, or perhaps fail fast on the problem driver so that it doesn't block the UI.

Memory: 62.8 GiB GSkill Ripjaws X  
MB: ASRock X99  
CPU: Intel® Core™ i7-5930K CPU @ 3.50GHz × 12  
GPU: GeForce GTX 1080/PCIe/SSE2  
Gnome: 3.28.2  
OS Type: 64-bit  
Disk1: 503.0 GB Samsung 960 Pro SSD  
Disk2: 3.0TB WD  
Disk3: 4.0TB WD  
Disk4: 3.0TB Seagate

groeck, 2018-08-06:
@ajbogh: I'll be more than happy to review and if appropriate apply any patches needed to solve the problem. Unfortunately, your problem description is so vague that providing patches (or even trying to analyze the problem and identify the root cause) will be all but impossible for anyone but yourself.

ajbogh, 2018-08-06:
Thanks @groeck. I'd be happy to help you identify the problem if you're willing to provide a method of doing so. Is there a command that could identify the lag, or do you recommend compiling lm-sensors and debugging it manually? 

What I wanted to make apparent was that this bug is not specific to Dell. There must be something that is causing the mouse lag due to lm-sensors, and it's affecting multiple systems. While that doesn't give you specific information, hopefully it can reopen the issue and bring new light to the problem.

groeck, 2018-08-06:
Guess you already identified the culprit as to be lm-sensors, even though lm-sensors by itself doesn't really do anything unless explicitly executed. As such, you are way ahead of me. You obviously also already determined that the problem is not caused by psensors (which presumably you are using) or by its means to read sensor data.
I would have expected a problem with a kernel driver, but you already determined that this is not the case. I also have no idea how psensors accesses sensors data except that it has an "lm-sensor provider", whatever that means. Given all that, no, sorry, I have no idea how to track this down further.

groeck, 2018-08-06:
lm-sensors doesn't use psensors. psensors, presumably, uses lm-sensors (or, rather, an lm-sensor provider, whatever that is). You have made it clear that you assume that the problem lies with lm-sensors and not with psensors. As such, you _did_ make a very clear assessment.

Please note that, while I "own" the lm-sensors repository, that is not by choice. It is because the original repository got lost when its server died, and I happened to have a copy. I do not consider myself to be the maintainer of lm-sensors. I review and apply patches to it as time permits. However, I may consider taking it down at some point in the near future, since I am unable to provide the (free) support that is expected nowadays.

f3rr, 2018-08-07:
The statement from @groeck that dell doesnt support linux wonders me and in my opinion its simple not true. I have the same lagging issue since years now on my Dell XPS 9333 Developer Edition (Ubuntu) notebook. Its an Ubuntu Developer Notebook, so i assume linux is supported.

i have the problem with xsensros, and with psensor (and not psensor**s**) and with sensors
I can confirm once disabling lm-sensors in psensor, and restarting it, the lag problem doesn't exists anymore, nether any temp reading in pensor. So this isnt a fix.

This repo has only lm-sensors in focus, so lets focus on that one:
lets run:` while true; do sensors; sleep 0.5; done`
and after every half a second mouse laggs.

@groeck 
/usr/bin/sensors is an lm-sensors problem, so please reopen this issue.
I will just ignore you last comment cos its a joke.

groeck, 2018-08-07:
@f3rr: Please see README.

olysonek, 2018-08-21:
> @f3rr: Please see README.

No longer relevant. Reopening. I'll review this issue as soon as I can.

aurel32, 2018-08-23:
I really think it is a kernel issue. To confirm that the best would be to read the corresponding files in `/sys/class/hwmon` with cat. If the problem can be reproduced that way it's very likely the bug pointed by @groeck above.

olysonek, 2018-08-24:
Great, thanks for tracking this down. Can we close this then?

f3rr, 2018-08-25:
Yes from my side.