lm-sensors/lm-sensors #107: lm-sensors/prog/pwm/fancontrol bug when accessing a file

salamdamai, 2018-05-20:
Hi
 I am testing fancontrol v0.71. I keep getting this error message:

Enabling PWM on fans...
Starting automatic fan control...
/usr/sbin/fancontrol: line 538: read: read error: 0: Input/output error
Error reading PWM value from /sys/class/hwmon/hwmon0/pwm2

File pwm2 exists and it has a integer value in line 1, 255. I am familiar with shell script but I am not familiar what this script is doing.

groeck, 2018-05-20:
The error is reported by the kernel driver. There is nothing wrong with fancontrol. All it does is to report the error seen when reading the atttribute. Unfortunately you don't provide any information about the kernel driver or the chip it supports, nor the kernel version. As such, we have no means to find out if and when the driver would report an I/O error, and there is nothing we can do (and, again, it is not a fancontrol problem to start with).

salamdamai, 2018-05-21:
This is the kernel:

Linux dell5500 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1 (2018-04-29) x86_64 GNU/Linux
The CPU is intel Xeon X5660.

Thanks for getting back.

groeck, 2018-05-22:
Please consider providing additional information, such as the failing driver, the output of sensors-detect, and any relevant kernel log messages.

salamdamai, 2018-05-23:
Hi
This is the output from sensors-detect (only the one that is able to detect):

Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): yes
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     Yes
Found `SMSC SCH5504-NS Super IO'                            
    (no hardware monitoring capabilities)
To load everything that is needed, add this to /etc/modules:
#----cut here----
# Chip drivers
coretemp
#----cut here----

This is the output from dmesg:
[dmesg.log](https://github.com/groeck/lm-sensors/files/2033201/dmesg.log)

groeck, 2018-07-09:
No further information provided.

closed 2018-07-09 by groeck