lm-sensors/lm-sensors #109: Error in prog/pwmm/fancontrol due to empty variable

lisaev, 2018-05-21:
In function UpdateFanSpeeds() there is a conditional
---
if [ $pwmpval -eq 0 -o $min_fanval -eq 0 ]
---
apparently, $min_fanval can be empty or contain whitespaces, as shown in an old Arch Linux bug  https://bugs.archlinux.org/task/34187 (reported against lm_sensors 3.4.0 but this code hasn't changed since then). This causes an error "[: too many arguments".

If I understand correctly, $min_fanval must be essentially integer, so whitespaces should be treated as hard errors, so I suggest the attached patch to deal with cases when $min_fanval is empty.

On a related note, I have patches that cleanup a bit the fancontrol script (quoting variables, replacing [ with builtin [[ and arithmetic evaluation environment ((, and reducing number of external calls to cat(1), grep(1), sed(1), etc.). Is there an interest in taking them?

Thanks for considering!

[0001-fancontrol-fix-an-error-in-UpdateFanSpeeds-when-min_.txt](https://github.com/groeck/lm-sensors/files/2024092/0001-fancontrol-fix-an-error-in-UpdateFanSpeeds-when-min_.txt)

groeck, 2018-05-21:
I don't immediately see how min_fanval can be a non-integer. Instead of blindly taking your patch, I would prefer to understand what exactly is going on.
Besides, I don't understand how the patch can work,
           if (( pwmpval == 0 || min_fanval == 0 ))
given that the variable names do not start with $.

lisaev, 2018-05-21:
I think that $min_fanval can rather be empty instead of non-integer. But both these situations will lead to test (i.e. [) to error out with the same message.

Regarding variable names not starting with $, in bash (( ... )) environment it is not necessary. Per bash manpage:
"Shell  variables  are  allowed as operands; parameter expansion is performed before the expression is evaluated.  **Within an expression, shell variables  may  also  be referenced by name without using the parameter expansion syntax.**  A shell variable that is null or unset evaluates  to 0 when referenced by name without using the parameter expansion syntax. The value of a variable is evaluated as an arithmetic  expression  when it  is  referenced, or when a variable which has been given the integer
attribute using declare -i is assigned a value.  A null value evaluates to  0.   A shell variable need not have its integer attribute turned on to be used in an expression."

So, if/when min_fanval is non-integer, there will be an error. And if it's empty, it will be treated the same as 0... or are you saying this latter case is wrong?

lisaev, 2018-05-21:
And to further illustrate:
-takahe-16:32-~: x=1; y=2; (( x == 1 && y == 2 )) && echo yes
yes
-takahe-16:32-~: x=1; y=2; (( x == 1 && y != 2 )) && echo yes
-takahe-16:32-~: x=1; y=-2; (( x == 1 && y != 2 )) && echo yes
yes

In fact, my patch can be further simplified by using bash's C-like convention that an integer variable which is non-zero evaluates to true, and zero -- to false, namely (( !  pwmpval || ! min_fanval )), but it is more obscure imho....

groeck, 2018-05-21:
Unless I am missing something, for min_fanval to be non-numeric, one_fanval must be non-numeric. If so, changing the if statement may just hide some other problem. I would rather understand that other problem and fix it.

lisaev, 2018-05-26:
Yes, you are right. Digging through git history I found commit af447db31acad7ad8e43afa717e0646daf66721b (fancontrol: Fix shell error when FCFANS is not set) from 2014 which fixed this. Before that commit,  min_fanval could indeed be empty which caused flood of error messages from fancontrol. I was confused by the custom patch which we still apply in the Arch Linux package and didn't realize it is actually against version 3.3.x.

Sorry for the confusion.