lm-sensors/lm-sensors #111: Nuvoton NCT6796D Not Identified

reluctantuser, 2018-05-31:
New ASrock J5005-ITX board with NCT6796D (based on visual identification, "I looked at the board")

OS: Debian Linux "buster" 64-bit, fully updated as of 30-May-2018

"sensors-detect" fails to identify the sensor chip even though support has been added previously for the NCT6796D per this URL:

[https://github.com/groeck/lm-sensors/commit/df96b0bb4d36e0fed4d15b3777be287ba7e92ba0](url)

Here is a snip of my "sensors-detect" output:

``
sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
Board: ASRock J5005-ITX
Kernel: 4.16.0-1-amd64 x86_64
Processor: Intel(R) Pentium(R) Silver J5005 CPU @ 1.50GHz (6/122/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd423
    (logical device B has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
``

Adding "acpi_enforce_resources=lax" does not help.

I would guess that a driver needs to be changed/updated to support the new ID. Hopefully it will be that simple.

groeck, 2018-05-31:
The log output you provided shows that you are running a version of sensors-detect from 2015. The commit you refer to is from February 2018. 
sensors-detect is a user space application. Kernel command line parameters do not affect it, and do not change its output. Also, since it is a user space application, kernel driver updates do not affect its output either.
The Linux kernel will support NCT6796D starting with v4.17. This has nothing to do with the ability of sensors-detect to identify the chip.

reluctantuser, 2018-05-31:
The version of "lm-sensors" that I am running is as current as whatever version the Debian packagers are tossing into their "buster" repository, but as you seem to suggest that isn't important here.

In this case my version of "lm sensors" is 3.4.0-4:

``
ii  libsensors4:amd64                 1:3.4.0-4                    amd64        library to read temperature/voltage/fan sensors
``
``
ii  lm-sensors                        1:3.4.0-4                    amd64        utilities to read temperature/voltage/fan sensors
``

Good to know that support is coming in 4.17, but the detect ID number still concerns me. What was found on my system differs from what was in the commit for support, or perhaps there is another commit that I cannot find via web search?

groeck, 2018-05-31:
You seem to be concerned that the sensors program does not detect the chip, but you don't seem to be concerned about actual support for the chip in the Linux kernel. I am curious - can you explain why it is so important for you that sensors-detect identifies the chip ?
If you want a version of lm-sensors which detects the chip, you will have to download and install it from here. There has not been a release of lm-sensors for a few years, and it is unlikely that there will be one soon.
To repeat what I said above - an updated version of lm-sensors won't help to get the chip supported in the Linux kernel. For that, you'll need an up-to-date version of the nct6775 kernel driver, which will be released with Linux v4.17.

groeck, 2018-07-09:
Not a problem observed with the package here.

closed 2018-07-09 by groeck