lm-sensors/lm-sensors #112: sensors3.conf for Supermicro X11SAE

mlampe, 2018-06-05:
Here's my config file for the Supermicro X11SAE.
[sensors3.txt](https://github.com/groeck/lm-sensors/files/2070571/sensors3.txt)

rigred, 2018-08-19:
This config is not really applicable to the lm-sensors repo.
You are welcome to submit it to my git repo here instead:
https://github.com/rigred/sensor_config

I created this to keep track of custom sensor config files for various mainboards.

olysonek, 2018-08-20:
> This config is not really applicable to the lm-sensors repo.

@rigred Why not? lm-sensors has sample configs in the `configs/` directory.

rigred, 2018-08-20:
At the time the repo was unmaintained, as groeck decided.
Now that you are the new maintainer, such commits can be considered again.

rigred, 2018-08-20:
As such I would have 3 configs from my repo that I would like to add:
https://github.com/rigred/sensor_config

olysonek, 2018-08-20:
OK, thanks. I'll take a look as soon as I can.

groeck, 2018-08-20:
The only configuration options I commonly rejected were those with, say, disabled fans even though a given board had a respective fan header. The only reason I closed this issue is because I had no intention to create a patch out of it myself, not because I rejected the configuration. If I had received a pull request or patch e-mail instead I would have had it accepted and integrated.

Come on folks, you already got me to drop support for this package. Please refrain for putting false statements into my mouth.

rigred, 2018-08-20:
Not my intent, nor do I believe did I put false statements in anyones mouth. When I made the original comment I assumed the repo was dead. 

Will make a pull request in the morning.