lm-sensors/lm-sensors #118: Request: Support for AIO water coolers such as Corsair/Asetek

nick-s-b, 2018-08-05:
Hello!
I recently built a new computer and opted for a water cooler to better deal with a power-hungry AMD processor. `lm-sensors` picked up every sensor available except for the water cooler-related ones. 

A water cooler is basically composed of a pump, which circulates the water around, a radiator, which dissipates the heat, and a fan/fans, which cool down the radiator. 

To control the speeds of the water pump and fans, I use this awesome piece of code: https://github.com/audiohacked/OpenCorsairLink/  which can control all kinds of OEM pumps that are sold under many different brand names.

Does anyone know how to add fan and pump speed info from `OpenCorsairLink` to `lm-sensors`? I think the relevant section is this: https://github.com/audiohacked/OpenCorsairLink/tree/testing/protocol  It's all C code. If `lm-sensors` could grab that data then every program that uses `lm-sensors` could graph it, track it and issue alerts.

Thanks!

groeck, 2018-08-05:
I see two options: Someone could write a kernel driver for the devices in question, or someone could enhance libsensors to support user space plugins, and then implement such a plugin for the water coolers in question. Patches welcome.

groeck, 2018-08-17:
WontFix - repository will go offline by August 1.

olysonek, 2018-08-21:
> WontFix - repository will go offline by August 1.

No longer relevant. Reopening. I'll review this issue as soon as I can.

jameshilliard, 2019-02-03:
So looks like the Asetek based Corsair coolers aren't USB HID based, only the Coolit ones are. Seems it's a custom vendor binary protocol.

nick-s-b, 2019-02-04:
@jameshilliard Have you looked at this? 
https://github.com/audiohacked/OpenCorsairLink/blob/testing/lowlevel/asetek.c
and the protocol itself:
https://github.com/audiohacked/OpenCorsairLink/blob/testing/protocol/asetek/pump.c

I haven't seen it documented anywhere else.

jameshilliard, 2019-02-04:
Yeah, seems that's the one, weird thing with my h115i is that it's recognized as [this](https://github.com/audiohacked/OpenCorsairLink/blob/3c0374c25676210bc5cfb9f53a600d721a8824b3/device.c#L220-L229) and not [this](https://github.com/audiohacked/OpenCorsairLink/blob/3c0374c25676210bc5cfb9f53a600d721a8824b3/device.c#L268-L277). Seems the pro is a different protocol from the non-pro.

nick-s-b, 2019-02-04:
@jameshilliard Hmm... I have the exact same issue but I haven't noticed it before. I have a H115i (non-pro version) but OpenCorsairLink reports it as H110i GT V2. This is a new bug because it did report it as H115i in the past.

jameshilliard, 2019-02-04:
@nick-s-b I think the bug is due to it picking the device name up based off of the productid(when it should probably only pick the protocol based on the productid), think there's a USB device name or something that's accurate which can be read off from the device.

nick-s-b, 2019-02-04:
I filed the bug with OLC. Feel free to add your own `lsusb -d 1b1c: -v` output so the author can get more data.

JAicewizard, 2020-10-04:
I tried to get a linux driver into the kernel, but failed with the biggest issue the fact that there is no known way to "reset" the fan speed.
There are a couple options I could take before I send in a new patch, 
one is to just, not allow setting fan speed,
the other will be to find a way to reset the fan speed/profile.
I can reset the fan speed to the default fan curve, but this fan-curve needs to be defined in the kernel and the maintainers dont like this.