lm-sensors/lm-sensors #121: missing quote on wlan off

cyrinux, 2018-08-22:
JSON is break with "ERROR: Can't get value of subfeature temp1_input: Can't read" without quotes :)

{code}

❯ sensors -v
sensors version 3.4.0+SVN with libsensors version 3.4.0+SVN

❯ sensors -j
{
   "iwlwifi-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
ERROR: Can't get value of subfeature temp1_input: Can't read

      }
   },
   "pch_skylake-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
         "temp1_input": 56.500
      }
   },
   "acpitz-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
         "temp1_input": 25.000,
         "temp1_crit": 107.000
      }
   },
   "dell_smm-virtual-0":{
      "Adapter": "Virtual device",
      "Processor Fan":{
         "fan1_input": 5985.000
      },
      "CPU":{
         "temp1_input": 67.000
      },
      "Ambient":{
         "temp2_input": 60.000
      },
      "SODIMM":{
         "temp3_input": 45.000
      },
      "Other":{
         "temp4_input": 37.000
      }
   },
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 88.000,
         "temp1_max": 100.000,
         "temp1_crit": 100.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 88.000,
         "temp2_max": 100.000,
         "temp2_crit": 100.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 87.000,
         "temp3_max": 100.000,
         "temp3_crit": 100.000,
         "temp3_crit_alarm": 0.000
      }
   }
}
{code}

Thanks you!

lasers, 2018-08-22:
This should throw an error code too.

cyrinux, 2018-08-22:
Not sure about throw an error, maybe simply not display the device. Because I must be able to disable/enable my wireless card and get sensors ...

lasers, 2018-08-22:
It depends on what they want to do here with `-j` output. I was saying that if it wasn't going to be a valid JSON, it should throw an error code... Otherwise, print a valid JSON with either `0.000` or having invalid sensors not a part of `-j` output. I think we shouldn't put in quoted error messages.

lasers, 2018-08-22:
@su8 This breaks my [lm_sensors](https://github.com/ultrabug/py3status/pull/1294) module for @cyrinux. Can you fix it? :rofl:

su8, 2018-08-22:
I can't reproduce the JSON error with latest git commit and v3.4.0.

```json
{
   "atk0110-acpi-0":{
      "Adapter": "ACPI interface",
      "Vcore Voltage":{
         "in0_input": 1.356,
         "in0_min": 0.800,
         "in0_max": 1.600
      },
      "+3.3V Voltage":{
         "in1_input": 3.344,
         "in1_min": 2.970,
         "in1_max": 3.630
      },
      "+5V Voltage":{
         "in2_input": 5.040,
         "in2_min": 4.500,
         "in2_max": 5.500
      },
      "+12V Voltage":{
         "in3_input": 12.278,
         "in3_min": 10.200,
         "in3_max": 13.800
      },
      "CPU Fan Speed":{
         "fan1_input": 1068.000,
         "fan1_min": 600.000,
         "fan1_max": 7200.000
      },
      "Chassis Fan Speed":{
         "fan2_input": 2257.000,
         "fan2_min": 600.000,
         "fan2_max": 7200.000
      },
      "CPU Temperature":{
         "temp1_input": 41.000,
         "temp1_max": 60.000,
         "temp1_crit": 95.000
      },
      "MB Temperature":{
         "temp2_input": 38.000,
         "temp2_max": 45.000,
         "temp2_crit": 75.000
      }
   },
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "temp1":{
         "temp1_input": 29.250,
         "temp1_max": 70.000,
         "temp1_crit": 83.500,
         "temp1_crit_hyst": 81.500
      }
   }
}
```

su8, 2018-08-22:
I can submit a patch that quotes the error messages, so you don't get stuck right in the middle of parsing json. So **ERROR: Can't get value of subfeature temp1_input: Can't read** will become `"ERROR: Can't get value of subfeature temp1_input: Can't read"`

lasers, 2018-08-22:
@su8 You have all good readings. He have one bogus reading.

If he run `sensors -j`, he get https://pastebin.com/raw/yb3GNzRw and no error code.
If he run `sensors -u`, he get same ERROR line in same place and **and an error code**.

The proper fix for him is to add this config to ignore that bogus reading.
```
| /etc/sensors.d
| ------------------
| chip "iwlwifi-virtual-*"
|     ignore temp1
```
For more information, see `man -P 'less +/"IGNORE STATEMENT"' sensors.conf`.

After doing that, he now have http://ix.io/1kXg.  Looking good, right?  No, There still is an empty line in same place that shouldn't be there.

Basically, just fix an error code for `-j` too and remove that empty line.

su8, 2018-08-22:
That's the offending line

https://github.com/lm-sensors/lm-sensors/blob/2ddf3e5ad5ded5ee8857f50cf79d4c6ff7a1c8db/prog/sensors/chips.c#L115

Just comment it in and you get no empty line, but be aware that some **{** or **}** will be nested with others just for the missing chip.

su8, 2018-08-22:
Here's a proper patch:

```c
  if (0 != cnt) {
    printf("\n");
  }
```

lasers, 2018-08-22:
What about error code? I don't want users with bogus readings to think my module is a :wastebasket::fire:.

su8, 2018-08-22:
With the **ignore** thing does it throw error code in the terminal/py3status module ?

lasers, 2018-08-22:
I don't know...  I have `try/except` to catch failed command from `sensors -j` for users still using older `sensors` versions since it was not long ago that `sensors` got `-j`.  @cyrinux had current git and it was still telling him to upgrade because `sensors -j` didn't throw an error with that **ERROR** line in json.

@cyrinux Do `sensors -u` throw an error code too with ignore in?

lasers, 2018-08-22:
@su8 Add **error code*** for `-j` too. :yellow_heart:

lasers, 2018-08-22:
@cyrinux Can you test #122?

su8, 2018-08-23:
So to recap, you'd like to have a error code (-1) on the programs exit when she stumbles upon a error that can be checked via `echo $!` ?

lasers, 2018-08-23:
@su8 Yes. `sensors` already does that for `sensors -u`, but not `sensors -j`. With `ignore` added, `sensors` stopped printing error on `sensors -u`. I'm clarifying that it should happen to `sensors -j` too.

olysonek, 2018-08-23:
So I've only glanced over this so far, but:

> JSON is break with "ERROR: Can't get value of subfeature temp1_input: Can't read" without quotes :)

It's not broken if you separate standard output (which is where the JSON is written) and standard error output (which is where the error is written). If you run it like this, you should get a correct JSON and the error goes to the file `errors`:
`sensors -j 2>errors`

So perhaps no fix is necessary.

lasers, 2018-08-23:
@su8 How do you diff? `git diff`? This patch is bad.

lasers, 2018-08-23:
```json
{
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 55.000,
         "temp1_max": 81.000,
         "temp1_crit": 91.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 48.000,
         "temp2_max": 81.000,
         "temp2_crit": 91.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 51.000,
         "temp3_max": 81.000,
         "temp3_crit": 91.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
,   <---- Look at this little shit.
         "temp4_max": 81.000,
         "temp4_crit": 91.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 55.000,
         "temp5_max": 81.000,
         "temp5_crit": 91.000,
         "temp5_crit_alarm": 0.000
      },
      "Core 4":{
         "temp6_input": 50.000,
         "temp6_max": 81.000,
         "temp6_crit": 91.000,
         "temp6_crit_alarm": 0.000
      },
      "Core 5":{
         "temp7_input": 52.000,
         "temp7_max": 81.000,
         "temp7_crit": 91.000,
         "temp7_crit_alarm": 0.000
      }
   }
}
```

su8, 2018-08-23:
Can you **ignore** it in sensors.conf, e.g:

```bash
chip "coretemp-isa-0000"
ignore temp4_input
```

su8, 2018-08-23:
Add `subCnt--;` on line 104 in `lm-sensors/prog/sensors/chips.c`

su8, 2018-08-23:
And execute `lm-sensors/prog/sensors/sensors -j 2>/dev/null`

lasers, 2018-08-23:
```json
{
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 55.000         "temp1_max": 81.000         "temp1_crit": 91.000         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 48.000         "temp2_max": 81.000         "temp2_crit": 91.000         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 51.000         "temp3_max": 81.000         "temp3_crit": 91.000         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_max": 81.000         "temp4_crit": 91.000         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 55.000         "temp5_max": 81.000         "temp5_crit": 91.000         "temp5_crit_alarm": 0.000
      },
      "Core 4":{
         "temp6_input": 50.000         "temp6_max": 81.000         "temp6_crit": 91.000         "temp6_crit_alarm": 0.000
      },
      "Core 5":{
         "temp7_input": 52.000         "temp7_max": 81.000         "temp7_crit": 91.000         "temp7_crit_alarm": 0.000
      }
   }
}
```

su8, 2018-08-23:
Are you sure that you added it right below `sensors_strerror(err)); ` on line 99 ?

su8, 2018-08-23:
Here's what I get with and without this variable:

```json
{
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "temp1":{
         "temp1_input": 33.250,
         "temp1_max": 70.000,
         "temp1_crit": 83.500,
         "temp1_crit_hyst": 81.500
      }
   },
   "atk0110-acpi-0":{
      "Adapter": "ACPI interface",
      "Vcore Voltage":{
         "in0_min": 0.800,
         "in0_max": 1.600
      },
      "+3.3V Voltage":{
         "in1_input": 3.344,
         "in1_min": 2.970,
         "in1_max": 3.630
      },
      "+5V Voltage":{
         "in2_input": 5.070,
         "in2_min": 4.500,
         "in2_max": 5.500
      },
      "+12V Voltage":{
         "in3_input": 12.278,
         "in3_min": 10.200,
         "in3_max": 13.800
      },
      "CPU Fan Speed":{
         "fan1_input": 1165.000,
         "fan1_min": 600.000,
         "fan1_max": 7200.000
      },
      "Chassis Fan Speed":{
         "fan2_input": 2265.000,
         "fan2_min": 600.000,
         "fan2_max": 7200.000
      },
      "CPU Temperature":{
         "temp1_input": 44.000,
         "temp1_max": 60.000,
         "temp1_crit": 95.000
      },
      "MB Temperature":{
         "temp2_input": 39.000,
         "temp2_max": 45.000,
         "temp2_crit": 75.000
      }
   }
}
```

without this variable:

```json
{
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "temp1":{
         "temp1_input": 33.250,
         "temp1_max": 70.000,
         "temp1_crit": 83.500,
         "temp1_crit_hyst": 81.500
      }
   },
   "atk0110-acpi-0":{
      "Adapter": "ACPI interface",
      "Vcore Voltage":{
,
         "in0_min": 0.800,
         "in0_max": 1.600
      },
      "+3.3V Voltage":{
         "in1_input": 3.344,
         "in1_min": 2.970,
         "in1_max": 3.630
      },
      "+5V Voltage":{
         "in2_input": 5.070,
         "in2_min": 4.500,
         "in2_max": 5.500
      },
      "+12V Voltage":{
         "in3_input": 12.278,
         "in3_min": 10.200,
         "in3_max": 13.800
      },
      "CPU Fan Speed":{
         "fan1_input": 1165.000,
         "fan1_min": 600.000,
         "fan1_max": 7200.000
      },
      "Chassis Fan Speed":{
         "fan2_input": 2265.000,
         "fan2_min": 600.000,
         "fan2_max": 7200.000
      },
      "CPU Temperature":{
         "temp1_input": 44.000,
         "temp1_max": 60.000,
         "temp1_crit": 95.000
      },
      "MB Temperature":{
         "temp2_input": 39.000,
         "temp2_max": 45.000,
         "temp2_crit": 75.000
      }
   }
}
```

lasers, 2018-08-23:
Lol... It was line 104. You said line 104 at first. :smile:

This works perfectly on line **99**. Cheers.

su8, 2018-08-23:
Do I have to add it to #122 ?

lasers, 2018-08-23:
I think we need @cyrinux to test this comma patch first. And that line might be stderr?

olysonek, 2018-08-23:
> Add subCnt--; on line 99 (just above else) in lm-sensors/prog/sensors/chips.c

That seems very hacky to me. Better move `subCnt++` to the two `else` branches.

BTW, use tabs for indentation in your patches, as that's what the source files use.

cyrinux, 2018-08-26:
I'm here, can I have a branch to test for this issue ? Thx

olysonek, 2019-05-05:
Fixed in 2c8cca3d6cd60, except for the exit code.

arigit, 2019-05-12:
@olysonek  thanks for this fix. Hoping a new lm-sensor version will be released to address the stray comma/invalid json problem.

Here's a related discussion around an issue with a gnome extension that got impacted by this lm-sensors problem - just for reference, https://github.com/UshakovVasilii/gnome-shell-extension-freon/issues/114