lm-sensors/lm-sensors #128: Feature request: follow curve while increase pwm on fancontrol

skjnldsv, 2018-09-04:
Hey!
Thanks a lot for this awesome software!

I'm using very high rpm fans (12k+) and the slightest change on pwm can get very high  very quickly.
Additionally, the temps provided my the cpu are often bumped very high very quickly on use. For example, default temp is 30, I start a big task, 5sec later the temp state 60°.

This increase the pwm very quickly. And for short tasks that just create a small spike in the temps, the fan can have its pwm value define for the whole delay (10s of 12k rpm can get very noisy)

My request would be to only increase the pwm progressively. e.g.
`MINPWM=100`, `MAXPWM=200`, `MAXTEMP=70`
Temp go from `20` to `70°C`. Currently, the fancontrol would set PWM at `200` directly.
My suggestion is that it slowly increase from `(200-100)/(70-20)=20` every $DELAY.
That way if it's a small spike, the pwm will only be updated to 120 and won't go from 0 to 120 and to 0 again :)

Thanks! :wave:

cooper151288, 2018-11-18:
Check your processor manual- you may have slew control available- this would do filtering on the temperature signal. a lot of AMD 15h/16h have it available

zdzichu, 2020-03-12:
Also, newer version of `fancontrol` have `AVERAGE=` settings, which would take last x readings and smooth them. It deals with peaks.

skjnldsv, 2020-03-12:
AVERAGE  do the job yes, thx!

closed 2020-03-12 by skjnldsv