lm-sensors/lm-sensors #130: lm_sensors-3.4 get 4.29 MW Megawatts ?but it is 2U server

wjwcan10482, 2018-09-05:
## rpm url
```
http://mirror.centos.org/centos/7/os/x86_64/Packages/lm_sensors-3.4.0-4.20160601gitf9185e5.el7.x86_64.rpm
```
## host info
```
[root@host2886758669 ~]# cat /etc/redhat-release 
CentOS Linux release 7.2.1511 (Core) 
[root@host2886758669 ~]# uname -a
Linux host2886758669 3.10.0-327.36.3.el7.x86_64 #1 SMP Mon Oct 24 16:09:20 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```
 
[root@host2886758669 ~]# sensors
coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +37.0°C  (high = +82.0°C, crit = +92.0°C)
Core 0:         +37.0°C  (high = +82.0°C, crit = +92.0°C)
Core 1:         +35.0°C  (high = +82.0°C, crit = +92.0°C)
Core 2:         +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 3:         +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 4:         +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 5:         +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 6:         +36.0°C  (high = +82.0°C, crit = +92.0°C)
Core 8:         +36.0°C  (high = +82.0°C, crit = +92.0°C)
Core 9:         +36.0°C  (high = +82.0°C, crit = +92.0°C)
Core 10:        +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 11:        +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 12:        +33.0°C  (high = +82.0°C, crit = +92.0°C)
Core 13:        +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 14:        +35.0°C  (high = +82.0°C, crit = +92.0°C)

coretemp-isa-0001
Adapter: ISA adapter
Physical id 1:  +37.0°C  (high = +82.0°C, crit = +92.0°C)
Core 0:         +37.0°C  (high = +82.0°C, crit = +92.0°C)
Core 1:         +37.0°C  (high = +82.0°C, crit = +92.0°C)
Core 2:         +36.0°C  (high = +82.0°C, crit = +92.0°C)
Core 3:         +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 4:         +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 5:         +33.0°C  (high = +82.0°C, crit = +92.0°C)
Core 6:         +37.0°C  (high = +82.0°C, crit = +92.0°C)
Core 8:         +37.0°C  (high = +82.0°C, crit = +92.0°C)
Core 9:         +35.0°C  (high = +82.0°C, crit = +92.0°C)
Core 10:        +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 11:        +35.0°C  (high = +82.0°C, crit = +92.0°C)
Core 12:        +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 13:        +34.0°C  (high = +82.0°C, crit = +92.0°C)
Core 14:        +35.0°C  (high = +82.0°C, crit = +92.0°C)

power_meter-acpi-0
Adapter: ACPI interface
power1:        **4.29 MW (interval =   1.00 s)**

Please help me~ Thks. is it bug from lm_sensors-3.4 ???

rigred, 2018-09-11:
This looks like it's stuck at maxval for a 32bit unsigned integer. **4'294'967'295**

olysonek, 2018-09-11:
But the problem could be in lower layers, it's not necessarily the driver. What motherboard is this? Does the BIOS or other software report useful values?

groeck, 2018-09-11:
The value is retrieved from the ACPI _PMM handle, which reports an unsigned long long (64 bit) value. See drivers/hwmon/acpi_power_meter.c:update_meter(). I would suggest to check the DSDT of the affected system. I would also suggest to check the kernel log because any errors are reported there and not to user space.

wjwcan10482, 2018-09-12:
@rigred  @olysonek @groeck  Thank you ！
it is normal after rebbot (Because the server moves another place)

```
[root@host2886758669 ~]# sensors
power_meter-acpi-0
Adapter: ACPI interface
power1:      262.00 W  (interval =   1.00 s)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +46.0°C  (high = +82.0°C, crit = +92.0°C)
Core 0:         +42.0°C  (high = +82.0°C, crit = +92.0°C)
Core 1:         +40.0°C  (high = +82.0°C, crit = +92.0°C)
Core 2:         +40.0°C  (high = +82.0°C, crit = +92.0°C)
Core 3:         +40.0°C  (high = +82.0°C, crit = +92.0°C)
Core 4:         +43.0°C  (high = +82.0°C, crit = +92.0°C)
Core 5:         +46.0°C  (high = +82.0°C, crit = +92.0°C)
Core 6:         +41.0°C  (high = +82.0°C, crit = +92.0°C)

```

```
#  I guess it is normal too  ？！
[root@host2886758669 ~]# for i in /sys/class/hwmon/hwmon*/{,device/}name; do
>         if [ "$(cat $i)" = power_meter ]; then
>                 dir=$(dirname $i)
>                 file="$dir/power1_input"
>                 if [ -f "$file" ]; then
>                         cat "$file"
>                         break
>                 fi
>                 file="$dir/power1_average"
>                 if [ -f "$file" ]; then
>                         cat "$file"
>                         break
>                 fi
>         fi
> done
cat: /sys/class/hwmon/hwmon*/name: No such file or directory
245000000

```

```
# It's message log after reboot  about apci
5546449:Sep 12 14:53:58 host2886758669 kernel: BIOS-e820: [mem 0x000000006ca52000-0x000000006d416fff] ACPI NVS
5546477:Sep 12 14:53:58 host2886758669 kernel: ACPI: RSDP 00000000000f05b0 00024 (v02 ALASKA)
5546478:Sep 12 14:53:58 host2886758669 kernel: ACPI: XSDT 000000006ca520b8 000F4 (v01 ALASKA    A M I 01072009 AMI  00010013)
5546479:Sep 12 14:53:58 host2886758669 kernel: ACPI: FACP 000000006ca9ab18 00114 (v06 ALASKA   A M I  01072009 INTL 20091013)
5546480:Sep 12 14:53:58 host2886758669 kernel: ACPI BIOS Warning (bug): FADT (revision 6) is longer than ACPI 5.0 version, truncating length 276 to 268 (20130517/tbfadt-311)
5546481:Sep 12 14:53:58 host2886758669 kernel: ACPI Error: Gpe0Block - 32-bit FADT register is too long (32 bytes, 256 bits) to convert to GAS struct - 255 bits max, truncating (20130517/tbfadt-202)
5546482:Sep 12 14:53:58 host2886758669 kernel: ACPI: DSDT 000000006ca52248 488CA (v02 ALASKA   A M I  01072009 INTL 20091013)
5546483:Sep 12 14:53:58 host2886758669 kernel: ACPI: FACS 000000006d415080 00040
5546484:Sep 12 14:53:58 host2886758669 kernel: ACPI: FPDT 000000006ca9ac30 00044 (v01 ALASKA   A M I  01072009 AMI  00010013)
5546485:Sep 12 14:53:58 host2886758669 kernel: ACPI: FIDT 000000006ca9ac78 0009C (v01 ALASKA    A M I 01072009 AMI  00010013)
5546486:Sep 12 14:53:58 host2886758669 kernel: ACPI: SPMI 000000006ca9ad18 00041 (v05 ALASKA   A M I  00000000 AMI. 00000000)
5546487:Sep 12 14:53:58 host2886758669 kernel: ACPI: UEFI 000000006ca9ad60 0005C (v01  INTEL RstUefiV 00000000      00000000)
5546488:Sep 12 14:53:58 host2886758669 kernel: ACPI: UEFI 000000006ca9adc0 00042 (v01 ALASKA   A M I  01072009      01000013)
5546489:Sep 12 14:53:58 host2886758669 kernel: ACPI: MCFG 000000006ca9ae08 0003C (v01 ALASKA    A M I 01072009 MSFT 00000097)
5546490:Sep 12 14:53:58 host2886758669 kernel: ACPI: HPET 000000006ca9ae48 00038 (v01 ALASKA   A M I  00000001 INTL 20091013)
5546491:Sep 12 14:53:58 host2886758669 kernel: ACPI: APIC 000000006ca9ae80 016DE (v03 ALASKA   A M I  00000000 INTL 20091013)
5546492:Sep 12 14:53:58 host2886758669 kernel: ACPI: MIGT 000000006ca9c560 00040 (v01 ALASKA   A M I  00000000 INTL 20091013)
5546493:Sep 12 14:53:58 host2886758669 kernel: ACPI: MSCT 000000006ca9c5a0 00090 (v01 ALASKA   A M I  00000001 INTL 20091013)
5546494:Sep 12 14:53:58 host2886758669 kernel: ACPI: NFIT 000000006ca9c630 18028 (v01                 00000000      00000000)
5546495:Sep 12 14:53:58 host2886758669 kernel: ACPI: PCAT 000000006cab4658 00048 (v01 ALASKA   A M I  00000002 INTL 20091013)
5546496:Sep 12 14:53:58 host2886758669 kernel: ACPI: PCCT 000000006cab46a0 0006E (v01 ALASKA   A M I  00000002 INTL 20091013)
5546497:Sep 12 14:53:58 host2886758669 kernel: ACPI: RASF 000000006cab4710 00030 (v01 ALASKA   A M I  00000001 INTL 20091013)
5546498:Sep 12 14:53:58 host2886758669 kernel: ACPI: SLIT 000000006cab4740 00030 (v01 ALASKA   A M I  00000001 INTL 20091013)
5546499:Sep 12 14:53:58 host2886758669 kernel: ACPI: SRAT 000000006cab4770 02830 (v03 ALASKA   A M I  00000002 INTL 20091013)
[... 537 lines cut ...]
5547274:Sep 12 14:53:58 host2886758669 kernel: acpi PNP0A08:09: _OSC: OS now controls [PCIeHotplug PME PCIeCapability]
5547279:Sep 12 14:53:58 host2886758669 kernel: pci 0000:d7:00.0: System wakeup disabled by ACPI
5547281:Sep 12 14:53:58 host2886758669 kernel: acpi PNP0A08:09: Disabling ASPM (FADT indicates it is unsupported)
5547282:Sep 12 14:53:58 host2886758669 kernel: ACPI: PCI Interrupt Link [LNKA] (IRQs 3 4 5 6 7 10 *11 12 14 15)
5547283:Sep 12 14:53:58 host2886758669 kernel: ACPI: PCI Interrupt Link [LNKB] (IRQs 3 4 5 6 7 *10 11 12 14 15)
5547284:Sep 12 14:53:58 host2886758669 kernel: ACPI: PCI Interrupt Link [LNKC] (IRQs 3 4 5 6 10 *11 12 14 15)
5547285:Sep 12 14:53:58 host2886758669 kernel: ACPI: PCI Interrupt Link [LNKD] (IRQs 3 4 5 6 10 *11 12 14 15)
5547286:Sep 12 14:53:58 host2886758669 kernel: ACPI: PCI Interrupt Link [LNKE] (IRQs 3 4 5 6 7 10 *11 12 14 15)
[... cut ...]
5548272:Sep 12 14:54:02 host2886758669 kernel: ipmi_si: probing via ACPI
5548274:Sep 12 14:54:02 host2886758669 kernel: ipmi_si: Adding ACPI-specified kcs state machine
5548281:Sep 12 14:54:02 host2886758669 kernel: ipmi_si: Trying ACPI-specified kcs state machine at i/o address 0xca2, slave address 0x0, irq 0
5548282:Sep 12 14:54:02 host2886758669 kernel: power_meter ACPI000D:00: Found ACPI power meter.
5548283:Sep 12 14:54:02 host2886758669 kernel: power_meter ACPI000D:00: Ignoring unsafe software power cap!
```

olysonek, 2018-09-12:
Thanks for the pointers @groeck 

> I guess it is normal too?
> ...
> 245000000

Yes, the value is in microwatts.

If it happens again, please attach the kernel log from the boot during which the problem happened. Alternatively, you can attach the log from the previous boot, if you have it stored.

Also, please attach the DSDT (hopefully github will allow that):
```
cat /sys/firmware/acpi/tables/DSDT > dsdt.dat
```

olysonek, 2018-09-20:
I'm going to close this for now. If you run into the problem again, please reopen the issue and provide the requested data. Thanks!

closed 2018-09-20 by olysonek