lm-sensors/lm-sensors #133: acpitz-virtual-0 is gone in newer kernel(s) such as 4.19-rc5 when compared to 4.18.9

constantoverride, 2018-09-27:
dmesg diff:
```patch
-Linux version 4.18.0-rc8-g1ffaddd029c8 (fastnstable@tempshade) (gcc version 8.1.1 20180531 (GCC)) #8 SMP PREEMPT Mon Aug 6 04:26:35 CEST 2018
+Linux version 4.19.0-rc5-g6bf4ca7fbc85 (fastnstable@tempshade) (gcc version 8.2.1 20180831 (GCC)) #9 SMP PREEMPT Tue Sep 25 21:55:49 CEST 2018
-ACPI: Core revision 20180531
+ACPI: Core revision 20180810
+ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
-ACPI: Thermal Zone [TZS0] (59 C)
-ACPI: Thermal Zone [TZS1] (47 C)
-ACPI: Battery Slot [BAT0] (battery present)
+ACPI: Thermal Zone [TZS0] (55 C)
+ACPI: Thermal Zone [TZS1] (43 C)
+battery: ACPI: Battery Slot [BAT0] (battery present)
```

`sensors` in 4.18.0-rc8:
```
k10temp-pci-00c3
Adapter: PCI adapter
CPU temp:     +48.1°C  (high = +70.0°C)
                       (crit = +100.0°C, hyst = +99.0°C)

acpitz-virtual-0
Adapter: Virtual device
CPU temp:     +48.0°C  (crit = +98.0°C)
MOBO temp:    +44.0°C  (crit = +126.0°C)

radeon-pci-0008
Adapter: PCI adapter
gfx card:     +48.0°C  (crit = +120.0°C, hyst = +90.0°C)

total 0
drwxr-xr-x 3 root root    0 27.09.2018 04:20 ./
drwxr-xr-x 3 root root    0 27.09.2018 04:20 ../
drwxr-xr-x 2 root root    0 27.09.2018 04:21 power/
-r--r--r-- 1 root root 4096 27.09.2018 04:21 name
lrwxrwxrwx 1 root root    0 27.09.2018 04:20 subsystem -> ../../../../class/hwmon/
-r--r--r-- 1 root root 4096 27.09.2018 04:21 temp1_crit
-r--r--r-- 1 root root 4096 27.09.2018 04:21 temp1_input
-r--r--r-- 1 root root 4096 27.09.2018 04:21 temp2_crit
-r--r--r-- 1 root root 4096 27.09.2018 04:21 temp2_input
-rw-r--r-- 1 root root 4096 27.09.2018 04:21 uevent

```

`sensors` in 4.19-rc5 with the below patch:
```
k10temp-pci-00c3
Adapter: PCI adapter
CPU temp:     +44.9°C  (high = +70.0°C)
                       (crit = +100.0°C, hyst = +99.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +44.0°C  (crit = +98.0°C)
temp2:        +42.0°C  (crit = +126.0°C)

radeon-pci-0008
Adapter: PCI adapter
gfx card:     +49.0°C  (crit = +120.0°C, hyst = +90.0°C)

$ ls -la /sys/class/hwmon/hwmon0/
total 0
drwxr-xr-x 3 root root    0 27.09.2018 04:26 ./
drwxr-xr-x 4 root root    0 27.09.2018 04:26 ../
drwxr-xr-x 2 root root    0 27.09.2018 04:42 power/
lrwxrwxrwx 1 root root    0 27.09.2018 04:26 device -> ../../thermal_zone0/
-r--r--r-- 1 root root 4096 27.09.2018 04:26 name
lrwxrwxrwx 1 root root    0 27.09.2018 04:26 subsystem -> ../../../../../class/hwmon/
-r--r--r-- 1 root root 4096 27.09.2018 04:38 temp1_crit
-r--r--r-- 1 root root 4096 27.09.2018 04:38 temp1_input
-r--r--r-- 1 root root 4096 27.09.2018 04:38 temp2_crit
-r--r--r-- 1 root root 4096 27.09.2018 04:38 temp2_input
-rw-r--r-- 1 root root 4096 27.09.2018 04:26 uevent

[...]

lrwxrwxrwx 1 root root 0 27.09.2018 02:59 /sys/class/hwmon/hwmon0/device/device/subsystem -> ../../../../bus/acpi/
lrwxrwxrwx 1 root root 0 27.09.2018 02:59 /sys/class/hwmon/hwmon0/device/subsystem -> ../../../../class/thermal/

```

poc naive (but working) patch:
```patch
diff --git a/lib/sysfs.c b/lib/sysfs.c
index 6a3c8fa8..b10eae4f 100644
--- a/lib/sysfs.c
+++ b/lib/sysfs.c
@@ -666,6 +666,10 @@ static int sensors_read_one_sysfs_chip(const char *dev_path,
 		subsys = strrchr(subsys_path, '/') + 1;
 	}
 
+  if (0 == strncmp(subsys,"thermal",8)) {
+    snprintf(linkpath, NAME_MAX, "%s/device", dev_path); // add another "/device" ie. /sys/class/hwmon/hwmon0/device/device/subsystem->../../../../bus/acpi/ is good but /sys/class/hwmon/hwmon0/device/subsystem->../../../../class/thermal/ isn't
+    return sensors_read_one_sysfs_chip(linkpath, dev_name, hwmon_path);
+  } else
 	if ((!subsys || !strcmp(subsys, "i2c")) &&
 	    sscanf(dev_name, "%hd-%x", &entry.chip.bus.nr,
 		   &entry.chip.addr) == 2) {
```

olysonek, 2018-09-27:
Hm, apparently "accessing attributes reached by a symlink pointing to another device, like the "device"-link, is a bug in the application":
https://github.com/torvalds/linux/blob/7876320f88802b22d4e2daf7eb027dd14175a0f8/Documentation/admin-guide/sysfs-rules.rst

But lm_sensors already does that. I have a feeling that fixing this is going to be a lot of fun...

Thanks for the report.

olysonek, 2018-09-27:
Ok, I think I know how to fix this. I'll do that during the weekend probably. The key is the "Position of devices along device chain can change." bullet in that document.

groeck, 2018-10-07:
Problem is caused by commit f6b6b52ef7a5 ("thermal_hwmon: Pass the originating device down to hwmon_device_register_with_info") in the upstream thermal subsystem code. The thermal subsystem now passes the thermal device as parent to the hwmon subsystem.
lm-sensors needs to know the hwmon parent device type to determine the bus and chip address. The poc above doesn't do that, which could result in problems if there is more than one thermal device in the system. Otherwise it is a step into the right direction.

olysonek, 2018-10-08:
Right. I've hacked up a poc patch for this last week, but I haven't got around to refactor it yet. Sorry for not updating this issue with that.
https://github.com/lm-sensors/lm-sensors/commit/b24abd4cc9fb6888ea47e75d71b9865df40b10a9

kukabu, 2018-10-25:
I have the same issue with Raspberry Pi (BCM2835_THERMAL module).
@olysonek, your patch doesn't work for me.
now I use the workaround:
```
--- sysfs.c.orig        2018-10-25 08:28:35.055617851 +0400
+++ sysfs.c     2018-10-25 08:29:15.905213226 +0400
@@ -717,6 +717,10 @@
                /* As of kernel 2.6.32, the hid device names don't look good */
                entry.chip.bus.nr = bus;
                entry.chip.addr = id;
+       } else if (subsys && !strcmp(subsys, "thermal")) {
+               entry.chip.bus.type = SENSORS_BUS_TYPE_VIRTUAL;
+               entry.chip.bus.nr = 0;
+               entry.chip.addr = 0;
        } else {
                /* Ignore unknown device */
                err = 0;
```

olysonek, 2018-10-31:
@kukabu Sorry for the late reply. Did you use the whole branch or just the patch I linked? There's one more patch you need: https://github.com/lm-sensors/lm-sensors/commit/420c38fd8ac9f37522b0fd11a84f0ecf0ddca596

If you used that patch as well, can you please run the following script to find out what the underlying subsystem is? Change `/sys/class/hwmon/hwmon4` to the correct hwmon chip.
```
dev=/sys/class/hwmon/hwmon4
while true; do
	if [ -L "$dev/subsystem" ]; then
		subsystem=$(basename $(readlink "$dev/subsystem"))
		echo "$dev: $subsystem"
	fi
	if [ -L "$dev/device" ]; then
		dev=$(readlink -f "$dev/device")
	else
		break
	fi
done
```

Btw, what kernel are you running? I don't get any hwmon chips exported on my RPi, even with older kernels (I do have `CONFIG_THERMAL_HWMON` enabled).

olysonek, 2018-10-31:
> I tried branch wip_fix_sysfs_access, it works for me. Thanks.

Good, I'll use the idea of that patch then. I'll just try to refactor it.

olysonek, 2018-11-06:
The following branch contains the fix I'd like to use (the last two commits being most significant). I'd appreciate any testing or review.
https://github.com/lm-sensors/lm-sensors/tree/WIP_fix_bus_identification

xftroxgpx, 2018-11-12:
> The following branch contains the fix I'd like to use (the last two commits being most significant). I'd appreciate any testing or review.
> https://github.com/lm-sensors/lm-sensors/tree/WIP_fix_bus_identification

Didn't look at the code, but tested it to work for me with kernel `v4.20-rc1`
Cheers!

<details>

```

$ sensors
k10temp-pci-00c3
Adapter: PCI adapter
CPU temp:     +53.6°C  (high = +70.0°C)
                       (crit = +100.0°C, hyst = +99.0°C)

acpitz-acpi-0
Adapter: ACPI interface
CPU temp:     +54.0°C  (crit = +98.0°C)
MOBO temp:    +48.0°C  (crit = +126.0°C)

radeon-pci-0008
Adapter: PCI adapter
gfx card:     +51.0°C  (crit = +120.0°C, hyst = +90.0°C)

-----------

$ which sensors
/usr/bin/sensors
-----------

$ pacman -Qo /usr/bin/sensors
/usr/bin/sensors is owned by lm_sensors 3.4.0+5207+778964e8-1
-----------
$ uname -a
Linux z5 4.20.0-rc1-g651022382c7f #19 SMP PREEMPT Mon Nov 5 13:39:05 CET 2018 x86_64 GNU/Linux

```
```
commit 778964e85f2f94367ece31ffddf5927a029d503d (HEAD -> makepkg, origin/WIP_fix_bus_identification)

```

</details>

commit d4cb9325 (2018-11-23): libsensors: Consider all ancestor devices when determining bus type

Go through all ancestor devices of a hwmon device until we find a device
that resides on a bus that we recognize. This change is meant to address
hwmon devices that don't reside directly on a bus, but have another
device in the middle of the hiearchy. For example:
ACPI device -> thermal device -> hwmon device

These kind of hwmon devices started appearing in the 4.19 kernel
due to commit f6b6b52ef7a54160c0. It was not reported as a kernel
regression and fixed in the kernel, because according to
Documentation/admin-guide/sysfs [... cut]
closed 2018-11-23 by olysonek