lm-sensors/lm-sensors #134: sensors show CPU FAN 0 RPM for ASUS ROG STRIX X470-I MB

NTMan, 2018-10-03:
```
# sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:       +0.77 V
fan1:         678 RPM
temp1:        +56.0°C  (crit = +89.0°C, hyst = -273.1°C)
power1:       18.00 W  (cap = 165.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +67.2°C  (high = +70.0°C)
Tctl:         +77.2°C

# sensors -u
asus-isa-0000
Adapter: ISA adapter
cpu_fan:
  fan1_input: 0.000

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:
  in0_input: 0.768
fan1:
  fan1_input: 678.000
temp1:
  temp1_input: 56.000
  temp1_crit: 89.000
  temp1_crit_hyst: -273.150
power1:
  power1_average: 20.000
  power1_cap: 165.000

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:
  temp1_input: 66.875
  temp1_max: 70.000
Tctl:
  temp2_input: 76.875

# uname -r
4.19.0-0.rc5.git3.1.fc30.x86_64
```

olysonek, 2018-10-05:
Did you run sensors-detect? It seems the board has an ITE Super IO chip which controls the fans. The chip should probably be handled by the it87 driver, but that driver won't be loaded automatically. sensors-detect should take care of that. After running it, sensors should print values from the chip.

If that doesn't help, please report the exact part number of the ITE chip. It's located near the bottom left corner in this picture:
https://im9.cz/iR/importprodukt-orig/fc2/fc2edcaa90c8932e3b8713b4b1e10989.jpg

I don't know what the asus-isa-0000 chip reports, it may help if you identify the driver responsible. I wrote a script for that:
https://github.com/lm-sensors/lm-sensors/blob/master/prog/debug/find-driver

Run it as follows and report its output:
$ ./find-driver asus-isa-0000

NTMan, 2018-10-05:
> Did you run sensors-detect? It seems the board has an ITE Super IO chip which controls the fans. The chip should probably be handled by the it87 driver, but that driver won't be loaded [...]

<details><summary>Yes I run sensors-detect, but it not hepls.</summary>

```
$ sudo sensors-detect 
[sudo] password for mikhail: 
# sensors-detect revision $Revision$
# Board: ASUSTeK COMPUTER INC. ROG STRIX X470-I GAMING
# Kernel: 4.19.0-0.rc5.git3.1.fc30.x86_64 x86_64
# Processor: AMD Ryzen 7 2700X Eight-Core Processor (23/8/2)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8665E Super IO Sensors'                        Success!
    (address 0x290, driver `to-be-written')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): YES
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): yes
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
[... 66 lines cut ...]
Note: there is no driver for ITE IT8665E Super IO Sensors yet.
Check http://www.lm-sensors.org/wiki/Devices for updates.

No modules to load, skipping modules configuration.

Unloading i2c-dev... OK
```
</details>

> If that doesn't help, please report the exact part number of the ITE chip. It's located near the bottom left corner in this picture:
> https://im9.cz/iR/importprodukt-orig/fc2/fc2edcaa90c8932e3b8713b4b1e10989.jpg

`IT8665E`
[image]

> I don't know what the asus-isa-0000 chip reports, it may help if you identify the driver responsible. I wrote a script for that:
> https://github.com/lm-sensors/lm-sensors/blob/master/prog/debug/find-driver
> 
> Run it as follows and report its output:
> $ ./find-driver asus-isa-0000
```
$ ./find-driver asus-isa-0000
Driver: eeepc-wmi
Module: asus_wmi
```

p-herbert, 2018-10-08:
I have the same issue for a `ROG STRIX Z370-E` motherboard.

```
$ sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +31.0°C  (high = +82.0°C, crit = +100.0°C)
Core 0:        +30.0°C  (high = +82.0°C, crit = +100.0°C)
Core 1:        +31.0°C  (high = +82.0°C, crit = +100.0°C)
Core 2:        +29.0°C  (high = +82.0°C, crit = +100.0°C)
Core 3:        +27.0°C  (high = +82.0°C, crit = +100.0°C)
Core 4:        +31.0°C  (high = +82.0°C, crit = +100.0°C)
Core 5:        +29.0°C  (high = +82.0°C, crit = +100.0°C)

$ sensors -u
asus-isa-0000
Adapter: ISA adapter
cpu_fan:
  fan1_input: 0.000

acpitz-virtual-0
Adapter: Virtual device
temp1:
  temp1_input: 27.800
  temp1_crit: 119.000
temp2:
  temp2_input: 29.800
  temp2_crit: 119.000

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:
  temp1_input: 31.000
  temp1_max: 82.000
  temp1_crit: 100.000
  temp1_crit_alarm: 0.000
Core 0:
  temp2_input: 29.000
  temp2_max: 82.000
  temp2_crit: 100.000
  temp2_crit_alarm: 0.000
Core 1:
  temp3_input: 31.000
  temp3_max: 82.000
  temp3_crit: 100.000
  temp3_crit_alarm: 0.000
Core 2:
  temp4_input: 29.000
  temp4_max: 82.000
  temp4_crit: 100.000
  temp4_crit_alarm: 0.000
Core 3:
  temp5_input: 27.000
  temp5_max: 82.000
  temp5_crit: 100.000
  temp5_crit_alarm: 0.000
Core 4:
  temp6_input: 30.000
  temp6_max: 82.000
  temp6_crit: 100.000
  temp6_crit_alarm: 0.000
Core 5:
  temp7_input: 29.000
  temp7_max: 82.000
  temp7_crit: 100.000
  temp7_crit_alarm: 0.000
```

olysonek, 2018-10-09:
I don't know what the `cpu_fan` reading reported by the `asus_wmi` driver is actually measuring. I suggest you try asking on the platform-driver-x86 mailing list (the acpi4asus-user list seems to be dead) or ask Corentin Chary, the maintainer of the driver:
````
[~/git/linux]$ ./scripts/get_maintainer.pl drivers/platform/x86/asus-wmi.c
Corentin Chary <corentin.chary (at) gmail.com> (maintainer:ASUS NOTEBOOKS AND EEEPC ACPI/WMI EXTRAS DRIVERS)
Darren Hart <dvhart (at) infradead.org> (maintainer:X86 PLATFORM DRIVERS)
Andy Shevchenko <andy (at) infradead.org> (maintainer:X86 PLATFORM DRIVERS)
acpi4asus-user (at) lists.sourceforge.net (open list:ASUS NOTEBOOKS AND EEEPC ACPI/WMI EXTRAS DRIVERS)
platform-driver-x86 (at) vger.kernel.org (open list:ASUS NOTEBOOKS AND EEEPC ACPI/WMI EXTRAS DRIVERS)
linux-kernel (at) vger.kernel.org (open list)
```

@p-herbert And have you run sensors-detect? From the pictures it seems to me the board has a Nuvoton chip. What part number is it?

olysonek, 2018-10-09:
@groeck Do you happen to know if support for ITE IT8665E is in the works?

groeck, 2018-10-09:
re IT8665E: I had an out-of-tree driver supporting it on github, but marked it as private because I got harassed about taking it upstream which I didn't have the time for (much less anyone seriously reviewing it, or helping to kink out its bugs, or chip/board vendor support). There should be various clones of this driver (named it87) on github. There are no efforts I know of to take the code upstream. 

The IT8665E is often used in conjunction with IT8686E on the same board, and it is often accessed from an EC on the board. This causes interactions which I have not been able to sufficiently address. While that was somewhat acceptable for an out-of-tree driver, the resulting instabilities would affect lots of people if the code was upstreamed, which in my opinion is unacceptable. Anyone who wants the code upstream should discuss with ITE and board vendors to get the necessary support.

olysonek, 2018-10-10:
Ok, thanks.

@p-herbert Reposting in case you missed it in my ill-formated comment: And have you run sensors-detect? From the pictures it seems to me the board has a Nuvoton chip. What part number is it?

p-herbert, 2018-10-10:
Hey @olysonek,

Sorry for my late replay. The Nuvoton chip part number is `NCT6793D`. Here is the output from `sensors-detect`:

```
$ sudo sensors-detect
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# Board: ASUSTeK COMPUTER INC. ROG STRIX Z370-E GAMING
# Kernel: 4.15.0-36-generic x86_64
# Processor: Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz (6/158/10)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd121
    (logical device B has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): YES
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): YES
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): YES
Found unknown SMBus adapter 8086:a2a3 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.

Next adapter: i915 gmbus dpc (i2c-0)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: i915 gmbus dpb (i2c-1)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: i915 gmbus dpd (i2c-2)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: DPDDC-B (i2c-3)
Do you want to scan it? (yes/NO/selectively): yes

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)
```

olysonek, 2018-10-12:
@p-herbert The NCT6793D chip should already be supported by the nct6775 driver in your kernel. Try running `modprobe nct6775` and then run `sensors`.

You seem to have an old version of lm_sensors. Please try running the current version of sensors-detect. It should identify the NCT6793D just fine and insert the correct kernel module:
https://github.com/lm-sensors/lm-sensors/blob/master/prog/detect/sensors-detect

p-herbert, 2018-10-14:
Hey @olysonek,

I upgraded lm_sensors and `sensors-detect` was able to successfully identify the NCT6793D chip. However, I still am not receiving a reading from it.

```
$ sudo sensors-detect
# sensors-detect revision $Revision$
# Board: ASUSTeK COMPUTER INC. ROG STRIX Z370-E GAMING
# Kernel: 4.15.0-36-generic x86_64
# Processor: Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz (6/158/10)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT6793D Super IO Sensors'                   Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): YES
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): yes
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No
[... 78 lines cut ...]
  temp3_input: 26.000
  temp3_max: 82.000
  temp3_crit: 100.000
  temp3_crit_alarm: 0.000
Core 2:
  temp4_input: 25.000
  temp4_max: 82.000
  temp4_crit: 100.000
  temp4_crit_alarm: 0.000
Core 3:
  temp5_input: 23.000
  temp5_max: 82.000
  temp5_crit: 100.000
  temp5_crit_alarm: 0.000
Core 4:
  temp6_input: 25.000
  temp6_max: 82.000
  temp6_crit: 100.000
  temp6_crit_alarm: 0.000
Core 5:
  temp7_input: 25.000
  temp7_max: 82.000
  temp7_crit: 100.000
  temp7_crit_alarm: 0.000
```

groeck, 2018-10-14:
lm-sensors and the sensors program are user space applications. Updating them won't make the kernel driver do anything different. It might be useful to check the kernel log for any nct6775 related messages.

p-herbert, 2018-10-14:
Hi @groeck,

I checked the kernel logs but did not find any nct6775 related logs. The modules appear to be loaded:

```
$ lsmod | grep "nct6775\|coretemp"
nct6775                57344  0
hwmon_vid              16384  1 nct6775
coretemp               16384  0
``` 

Here is the info for the `NCT6775` module:

```
$ modinfo nct6775
filename:       /lib/modules/4.15.0-36-generic/kernel/drivers/hwmon/nct6775.ko
license:        GPL
description:    Driver for NCT6775F and compatible chips
author:         Guenter Roeck <linux@roeck-us.net>
srcversion:     4B451E70227835BB9DA8DFC
depends:        hwmon-vid
retpoline:      Y
intree:         Y
name:           nct6775
vermagic:       4.15.0-36-generic SMP mod_unload 
signat:         PKCS#7
signer:         
sig_key:        
sig_hashalgo:   md4
parm:           force_id:Override the detected device ID (ushort)
parm:           fan_debounce:Enable debouncing for fan RPM signal (ushort)
```

Does anything look incorrect?

olysonek, 2018-10-16:
So `dmesg | grep nct6775` outputs nothing? Does removing and re-inserting the module result in any output (`modprobe -r nct6775; modprobe nct6775`)? Anything in the kernel log after that?

What is the output of the following?
`ls /sys/class/hwmon/; cat /sys/class/hwmon/*/name`

Are you running a distro kernel? Which distro is it?

groeck, 2018-10-16:
Yes, I find this confusing. Either modprobe should fail, or there should be a kernel log message. Also, if modprobe fails, the driver should not be listed as loaded.

p-herbert, 2018-10-22:
Hey @olysonek,

Here is the output you requested:

```
$ dmesg | grep nct6775 
[   33.654619] nct6775: Enabling hardware monitor logical device mappings.
[   33.654647] nct6775: Found NCT6793D or compatible chip at 0x2e:0x290
[  343.868857] nct6775: Found NCT6793D or compatible chip at 0x2e:0x290
[  794.816108] nct6775: Found NCT6793D or compatible chip at 0x2e:0x290
[ 1613.721396] nct6775: Found NCT6793D or compatible chip at 0x2e:0x290
[ 1987.786701] nct6775: Found NCT6793D or compatible chip at 0x2e:0x290
```

```
$ ls /sys/class/hwmon/; cat /sys/class/hwmon/*/name
hwmon0  hwmon1  hwmon2
acpitz
coretemp
asus
```

```
$ uname -a
Linux desktop 4.15.0-36-generic #39-Ubuntu SMP Mon Sep 24 16:19:09 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

Sorry for the late reply!

andresvensson, 2018-11-27:
I had similar problem but solved it from info at another [thread.](https://ubuntuforums.org/showthread.php?t=2401422) Though I might share.

**1, Find your model name:**
`sudo dmidecode > sensors.txt`
open file and search

(my sensor is IT8665E)
> Management Device
	Description: ITE IT8665E
	Type: Other
	Address: 0x00000295
	Address Type: I/O Port

**2, Confirm that yours are listed at [https://github.com/a1wong/it87](url) and then:**
`git clone git clone https://github.com/a1wong/it87.git`
`cd it87/`
`make clean`
`make`
`sudo make install`

**3, Get required program and add the drivers:**
`echo "it87" | sudo tee -a /etc/modules`           #if you haven't done this already
#you may have some of these already 
`sudo apt-get install git dkms build-essential linux-headers-generic`
`sudo make dkms`
`sensors`

**4, Boom!**

kalj, 2019-05-19:
I can confirm that the procedure from @andresvensson solves the problem for an Asus PRIME X470-PRO motherboard. Thanks a lot!

bulak, 2019-05-31:
So, even though IT8665E (like @andresvensson reported) is not in the list of supported chips at https://github.com/a1wong/it87, the it87 works well on this chip? Is that what you had also, @kalj ? Is there any other developments going on with this issue? Because, out of the box, the 4.19.0-4-amd64 x86_64 kernel does not provide any support.

romanalo, 2019-06-10:
The procedure described by Ander Svensson also works on my Asus Rog Strix X370-i Gaming board.
Thank you very much to all

LeYvan, 2019-06-30:
I had to add "ignore_resource_conflict=1" to make modprobe load the compiled driver with an Asus crosshair hero 7. But fans are missing and it's not clear what voltages are reported.

sensor-detect report my chip as "ITE IT8665E Super IO Sensors" and sensors report it as "it8665-isa-0290". 

So even though the a1wong branch of it87 will load and report data, it is unknown if the data is accurate at all and not really usable.

andresvensson, 2019-07-12:
> I had to add "ignore_resource_conflict=1" to make modprobe load the compiled driver with an Asus crosshair hero 7. But fans are missing and it's not clear what voltages are reported.
>
> sensor-detect [...]

Yes and my readings stopped working after a kernel update (running Fedora now). I found a possible replacement [driver](https://github.com/electrified/asus-wmi-sensors) but it wont load correctly in my system and my motherboard is listed as "Currently don't work / unknown". Well, I give up this time. Guess I have to wait for a proper sensor driver development.

anko, 2019-07-19:
I'm also using the it87 driver for a Prime X370-Pro motherboard, with an *ITE IT8665E Super IO Sensors*, and I still get readings on kernel 5.0.21.

@andresvensson What kernel release are you running where it doesn't work? For me on 5.0.21 recently, it87 would `make` and `modprobe` fine, but gave no readings.  I had to `rm .cache.mk` and re-`make` to make it work again.

xilwen, 2019-07-20:
@p-herbert Recently got an ASUS X570-PRO which have a NCT6798D, and `sensors` also prints `asus-isa-0000 ... cpu_fan: 0 rpm` and readings from the graphic card only.
Followed the [mailing list](http://lists.lm-sensors.org/pipermail/lm-sensors/2014-September/042745.html) and added `acpi_enforce_resources=lax` to boot parameter, reboot then `modprobe -v nct6775` make `sensors` output works (displayed as nct6798-isa-0290).
Distro is ubuntu 18.04, running hwe-edge 5.0.0-20 kernel.

p-herbert, 2019-07-22:
@xilwen Thanks! I'll give it a try!

p-herbert, 2019-08-06:
@xilwen it worked!

unixb0y, 2019-08-06:
> @xilwen it worked!

Hmmm, I added that boot parameter in /etc/default/grub like this:

```
GRUB_DEFAULT=0
GRUB_TIMEOUT_STYLE=hidden
GRUB_TIMEOUT=0
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash acpi_enforce_resources=lax"
GRUB_CMDLINE_LINUX=""
```

and my `sudo sensors-detect` output is still:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): y
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT6793D Super IO Sensors'                   Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
```

And `sensors` outputs fan RPM 0.

groeck, 2019-08-06:
The output of sensors-detect is unrelated to the existence of kernel drivers, or to the kernel command line. As such, it is not entirely surprising that its output does not change after changing the kernel command line.
It would be interesting to know if "sudo update-grub" was run after changing /etc/default/grub. Output of /proc/cmdline as well as relevant output from dmesg might also be helpful. Re-running sensors-detect won't make a difference.

p-herbert, 2019-08-06:
@unixb0y after editing /etc/default/grub you need to:

1. Run `sudo update-grub`
2. Restart your machine
3. Run `sudo modprobe -v nct6775`

Then `sensors` should see the NCT6798D chip.

unixb0y, 2019-08-06:
@groeck Yes I did run the update grub command.
I will post those outputs later.

unixb0y, 2019-08-06:
@p-herbert Pretty sure that’s exactly what I did :/
But I’ll retry and let you know. Thanks for the quick reply!

unixb0y, 2019-08-06:
@p-herbert indeed, not working.
@groeck 
```
$ dmesg | grep nct6775
[ 1352.798572] nct6775: Enabling hardware monitor logical device mappings.
[ 1352.798600] nct6775: Found NCT6793D or compatible chip at 0x2e:0x290
```
```
$ cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz-5.0.0-23-generic root=UUID=48eaecbb-047b-4370-9244-8df879842c61 ro acpi_enforce_resources=lax quiet splash acpi_enforce_resources=lax vt.handoff=1
```

LHozzan, 2019-09-07:
@xilwen Many thanks! It worked for me on Debian10!
```
$ uname -a
Linux cube 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2 (2019-08-28) x86_64 GNU/Linux
```
```
$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +55.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +54.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +55.0°C  (high = +105.0°C, crit = +105.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +105.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:            N/A  

nct6776-isa-0a00
Adapter: ISA adapter
Vcore:                  +1.82 V  (min =  +0.00 V, max =  +1.74 V)  ALARM
in1:                    +1.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.36 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.36 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in6:                    +0.20 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.34 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.30 V  (min =  +2.70 V, max =  +3.63 V)
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  3580 RPM  (min =    0 RPM)
SYSTIN:                 +60.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                 +48.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = CPU diode
AUXTIN:                 +55.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = CPU diode
PCH_CHIP_CPU_MAX_TEMP:  +55.0°C  (high = +80.0°C, hyst = +75.0°C)
PECI Agent 0:           +54.0°C  (high = +80.0°C, hyst = +75.0°C)
                                 (crit = +105.0°C)
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            OK
intrusion1:            OK
beep_enable:           disabled
```
```
$ sudo dmidecode -t 2
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: Intel Corporation
	Product Name: NUC5i5RYB
	Version: H40999-502
	Serial Number: GERY505005FE
	Asset Tag:                                  
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: To be filled by O.E.M.
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0
```
```
$ sudo dmidecode -t 0
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: Intel Corporation
	Version: RYBDWi35.86A.0381.2019.0807.1532
	Release Date: 08/07/2019
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 6528 kB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		BIOS ROM is socketed
		EDD is supported
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 5.6
```
Unfortunatelly, with upgrade to newer version I update BIOS too. Dont know, where exactly problem occured, because on previous version Debian9 this module working out-of-the box.

jonesman, 2019-10-20:
Thanks everyone, worked aswell on an ASUS P10S WS

HarlemSquirrel, 2019-12-11:
https://github.com/electrified/asus-wmi-sensors

chankruze, 2020-01-26:
I'll try on my ASUS TUF FX-504 and report.

sokunsamnang, 2020-02-05:
> I'll try on my ASUS TUF FX-504 and report.

Does it work on your ASUS TUF FX-504 ?

UNIm95, 2020-04-25:
I can confirm message of @andresvensson 
I have used same repository and now i got sensors reading on ASUS TUF B450M-Pro gaming.
They are have not proper low/high warnings.

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

radeon-pci-0700
Adapter: PCI adapter
temp1:        +34.0°C  (crit = +120.0°C, hyst = +90.0°C)

it8655-isa-0290
Adapter: ISA adapter
in0:          +1.05 V  (min =  +2.76 V, max =  +2.14 V)  ALARM
in1:          +2.52 V  (min =  +2.00 V, max =  +1.74 V)  ALARM
in2:          +2.00 V  (min =  +2.04 V, max =  +2.66 V)  ALARM
in3:          +1.98 V  (min =  +1.47 V, max =  +2.73 V)
in4:          +2.00 V  (min =  +1.16 V, max =  +0.20 V)  ALARM
in5:          +1.92 V  (min =  +0.59 V, max =  +2.06 V)
in6:          +1.98 V  (min =  +2.76 V, max =  +2.62 V)  ALARM
3VSB:         +3.33 V  (min =  +5.32 V, max =  +4.12 V)  ALARM
Vbat:         +3.25 V  
+3.3V:        +3.38 V  
fan1:         654 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =  124 RPM)  ALARM
fan3:           0 RPM  (min =   22 RPM)  ALARM
temp1:        +38.0°C  (low  = +59.0°C, high = -72.0°C)  ALARM
temp2:        +35.0°C  (low  = -105.0°C, high =  +3.0°C)  ALARM  sensor = disabled
temp3:        +39.0°C  (low  = +63.0°C, high = -124.0°C)  ALARM  sensor = disabled
temp4:        +39.0°C  (low  = +109.0°C, high = +115.0°C)  sensor = disabled
temp5:        +39.0°C  (low  = +95.0°C, high = -25.0°C)  ALARM  sensor = disabled
temp6:        +39.0°C  (low  = +17.0°C, high = -69.0°C)  ALARM  sensor = disabled
intrusion0:  ALARM

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +38.8°C  (high = +70.0°C)
Tctl:         +38.8°C  
Before i had reading only for k10temp-pci-00c3 asus-isa-0000 and radeon-pci-0700(my videocard)

So i think it should be bit checked for readings and added to lm-sensors.

pichaosdev, 2020-09-01:
> I'll try on my ASUS TUF FX-504 and report.

did you try it?

chankruze, 2020-10-13:
> > I'll try on my ASUS TUF FX-504 and report.
> 
> Does it work on your ASUS TUF FX-504 ?

> > I'll try on my ASUS TUF FX-504 and report.
> 
> did you try it?

Nope ;(

P-IH, 2020-11-28:
@xilwen. 
Your solution worked on a TUF Gaming B550M-Plus (WI-FI)
`p-i@pi-TUF-B550M:~$ sensors
amdgpu-pci-0a00
Adapter: PCI adapter
vddgfx:      775.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 2970 RPM)
edge:         +44.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +44.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +54.0°C  (crit = +105.0°C, hyst = -273.1°C)
                       (emerg = +110.0°C)
power1:       34.00 W  (cap = 180.00 W)

nct6798-isa-0290
Adapter: ISA adapter
in0:                      296.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                      1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                      824.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                      216.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                      912.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                     408.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                     408.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                     984.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                       1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                      813 RPM  (min =    0 RPM)
fan2:                      768 RPM  (min =    0 RPM)
fan3:                      801 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +36.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +26.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +90.5°C    sensor = thermistor
AUXTIN1:                   +36.0°C    sensor = thermistor
AUXTIN2:                   +36.0°C    sensor = thermistor
AUXTIN3:                   +25.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +26.5°C  
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
PCH_CHIP_TEMP:              +0.0°C  
PCH_CPU_TEMP:               +0.0°C  
intrusion0:               ALARM
intrusion1:               ALARM
beep_enable:              disabled

nvme-pci-0500
Adapter: PCI adapter
Composite:    +32.9°C  (low  =  -0.1°C, high = +74.8°C)
                       (crit = +79.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +28.0°C  

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +30.6°C  
Tdie:         +30.6°C  

nvme-pci-0100
Adapter: PCI adapter
Composite:    +30.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +30.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)

p-i@pi-TUF-B550M:~$ 
`

liuhe36, 2020-12-07:
asus prime x570 pro work~

root@pve:~# sensors
nct6798-isa-0290
Adapter: ISA adapter
in0:                       +0.93 V  (min =  +0.00 V, max =  +1.74 V)
in1:                       +1.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                       +3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                       +3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                       +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                       +0.90 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                       +0.99 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                       +3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                       +3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                       +0.91 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                      +0.45 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                      +0.54 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                      +1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                      +0.98 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                      +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     1035 RPM  (min =    0 RPM)
fan2:                      942 RPM  (min =    0 RPM)
fan3:                        0 RPM  (min =    0 RPM)
fan4:                        0 RPM  (min =    0 RPM)
fan5:                        0 RPM  (min =    0 RPM)
fan6:                        0 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +31.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +27.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +26.0°C    sensor = thermistor
AUXTIN1:                   +31.0°C    sensor = thermistor
AUXTIN2:                   +21.0°C    sensor = thermistor
AUXTIN3:                   +25.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +26.5°C  
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
PCH_CHIP_TEMP:              +0.0°C  
PCH_CPU_TEMP:               +0.0°C  
intrusion0:               ALARM
intrusion1:               ALARM
beep_enable:              disabled

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +34.8°C  (high = +70.0°C)
Tctl:         +34.8°C

Engnation, 2020-12-16:
I also have an ASUS Prime X570 Pro running a Ryzen 9 3900x with Ubuntu 18.04. Using the solution from @xilwen, I was able to set GRUB_CMDLINE_LINUX="acpi_enforce_resources=lax" in the etc/default/grub.txt file and was also able to view the full set of sensors one time. However, (perhaps after a reboot) it stopped working. I have tried toggling acpi_enforce_rsources=lax on and off but no luck.

Does anyone have any ideas as to what might be wrong? Also is anyone else having this same issue?

Engnation, 2020-12-16:
I also have an ASUS Prime X570 Pro running a Ryzen 9 3900x with Ubuntu 18.04. 

Thanks to @xilwen I was also able to set GRUB_CMDLINE_LINUX="acpi_enforce_resources=lax" in the etc/default/grub.txt file and was also able to view the full set of sensors working. 

I also needed to: 
    Run sudo update-grub --> Restart your machine --> Run sudo modprobe -v nct6775
Thank to @p-herbert for this tip as well.

nty-t, 2021-05-27:
> Did you run sensors-detect? It seems the board has an ITE Super IO chip which controls the fans. The chip should probably be handled by the it87 driver, but that driver won't be loaded [...]

Driver: asus-nb-wmi
Module: asus_wmi

quickfur, 2021-10-03:
For the sake of future google searches ending up here, just want to drop a note here to thank @xilwen for the tip about `acpi_enforce_resources=lax`. I have an ASUS Prime B550M-A motherboard that has an NCT6798-R chip, which is supported by the nct6775 kernel driver. However, due to an ACPI conflict (kernel reports `ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000290-0x0000000000000299 (\AMW0.SHWM) (20210604/utaddress-204)`), the driver gets disabled during boot and so the sensors don't show up.

Adding `acpi_enforce_resources=lax` to my kernel boot cmdline in /etc/default/grub fixed this problem, and now I'm getting sensor readings from the board.  Thanks a ton!!

whoisltd, 2022-08-10:
I was doing it following comments but my system can't find the sensor
```
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +47.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +64.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +63.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +61.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +61.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +60.0°C  (high = +100.0°C, crit = +100.0°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          11.85 V  

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

nvme-pci-e100
Adapter: PCI adapter
Composite:    +39.9°C  (low  = -273.1°C, high = +82.8°C)
                       (crit = +84.8°C)
Sensor 1:     +41.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +65.0°C  (crit = +103.0°C)

```