lm-sensors/lm-sensors #137: Can't get min and max to work right on it8792

KonomiKitten, 2018-10-18:
I'm currently working with this config:

````
# The temp3 temperature offset depends on the CPU type and needs to be
# adjusted. Compare against the output of k10temp for the correct value.

# The temp3 temperature offset depends on the CPU type and needs to be
# adjusted. Compare against the output of k10temp for the correct value.

chip "it8686-isa-0a40"
  label temp1 "System 1"
  label temp2 "Chipset"
  label temp3 "CPU Socket"
  label temp4 "PCI-EX16"
  label temp5 "VRM MOS"
  label temp6 "vSOC MOS"
  label in0 "CPU Vcore"
  label in1 "+3.3v"
  label in2 "+12v"
  label in3 "+5v"
  label in4 "CPU Vcore SOC"
  label in5 "CPU Vddp"
  label in6 "DRAM A/B"
  label in7 "3VSB"
  label in8 "Battery"
  label fan1 "CPU_FAN"
  label fan2 "SYS_FAN1"
  label fan3 "SYS_FAN2"
  label fan4 "SYS_FAN3_PUMP"
  label fan5 "CPU_OPT"

  compute temp3 @+15,@/15
  compute in1 @*1.650,@/1.650
  compute in2 @*6,@/6
  compute in3 @*2.5,@/2.5

  set in0_min 0.6
  set in0_max 1.45
  set in1_min 3.135
  set in1_max 3.465
  set in2_min 11.400
  set in2_max 12.600
  set in3_min 4.750
  set in3_max 5.250
  set in6_min 1.2 * 0.97
  set in6_max 1.2 * 1.2
  set in7_min 3.3 * 0.97
  set in7_max 3.3 * 1.05

chip "it8792-isa-0a60"
  label temp1 "PCI-EX8"
  label temp2 "Temp 2"
  label temp3 "System 2"
  label in0 "CPU Vcore"
  label in1 "DDR VTT"
  label in2 "Chipset Core"
  label in3 "VIN3"
  label in4 "CPU Vdd18"
  label in5 "DDR Vpp"
  label in6 "3VSB"
  label in7 "Battery"

  compute in6 @*1.20,@/1.20

  set in0_min 0.6
  set in0_max 1.45
  set in6_min 3.135
  set in6_max 3.465
  set in7_min 3.3 * 0.97
  set in7_max 3.3 * 1.03

  ignore fan1
  ignore fan2
  ignore fan3
  ignore in8
````

After doing `sudo sensors -s` and then doing `sensors it8792-isa-0a60` I get the following:

````
it8792-isa-0a60
Adapter: ISA adapter
CPU Vcore:     +1.23 V  (min =  +0.60 V, max =  +1.45 V)
DDR VTT:       +0.86 V  (min =  +0.00 V, max =  +2.78 V)
Chipset Core:  +0.77 V  (min =  +0.00 V, max =  +2.78 V)
VIN3:          +3.33 V  (min =  +0.00 V, max =  +5.56 V)
CPU Vdd18:     +1.01 V  (min =  +0.00 V, max =  +2.78 V)
DDR Vpp:       +0.86 V  (min =  +0.00 V, max =  +2.78 V)
3VSB:          +3.34 V  (min =  +3.14 V, max =  +3.34 V)  ALARM
Battery:       +3.31 V  (min =  +3.21 V, max =  +3.40 V)
PCI-EX8:       +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
Temp 2:        +49.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
System 2:      +52.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:   ALARM
````

As far as I know `3VSB` should not be in ALARM anymore but `set in6_min 3.135` and `set in6_max 3.465` seem to be completely ignored. What am I doing wrong here?

olysonek, 2018-10-19:
I see nothing wrong with the config, apart from the fact that I think you wanted to use '-' instead of '/' here: `compute temp3 @+15,@/15`. But that's completely unrelated.

Let's find out whether the problem lies in lm_sensors or in the it87 driver. First find out which of the HW monitoring chips exported in `/sys/class/hwmon/` is `it8792-isa-0a60`. You can identify it by looking at the contents of `/sys/class/hwmon/hwmon*/name` - it should be `it8792`.

Now let's say you've identified the chip to be `/sys/class/hwmon/hwmon1`. Try setting the desired value manually and see if it changes. You need to give the value as a whole number in milivolts:
`$ echo 3135 > /sys/class/hwmon/hwmon1/in6_min
$ cat /sys/class/hwmon/hwmon1/in6_min`

If the value changes to the desired value, the problem is somewhere in lm_sensors. If not, the problem is in the driver.

olysonek, 2018-10-19:
Apparently I keep forgetting the syntax for multi-line code blocks :).
```
$ echo 3135 > /sys/class/hwmon/hwmon1/in6_min
$ cat /sys/class/hwmon/hwmon1/in6_min
```

KonomiKitten, 2018-10-20:
Looks like it is the driver, annoying.

````
bash ~
# cat /sys/class/hwmon/hwmon1/name 
it8792
bash ~
# cat /sys/class/hwmon/hwmon1/in6_min
2780
bash ~
# echo 3135 > /sys/class/hwmon/hwmon1/in6_min
bash ~
# cat /sys/class/hwmon/hwmon1/in6_min
2780
bash ~
# 
````

olysonek, 2018-10-20:
Ok, let me know what you find. Have you tried setting a value that's closer to the original 2780? And can you please give a link to the modified driver?

cooper151288, 2018-11-17:
looks like there is some rudimentary support for 8686 here but it looks like a special case. i didn't write the driver but i'm willing to look into this. what I could do with is datasheets. don't think i have 8686 datasheet. If you can dump the EC registers, that may help- we can find out which chips are similar and add better support
Have you tried the force_id parameter?- I see nothing about the 8686 here just 8792
There is also an ignore conflicts param.
What chip(s) are physically there on the mobo?
does dmesg | grep it87 say anything useful?- might find out what is being detected then.

KonomiKitten, 2018-11-18:
> If you can dump the EC registers

Would be happy to do this if instructed how.

> What chip(s) are physically there on the mobo?

ITE IT8792E
ITE IT8686E

> Does dmesg | grep it87 say anything useful?

````
$ sudo dmesg | grep it87
[   34.014597] it87: loading out-of-tree module taints kernel.
[   34.014934] it87: it87 driver version v1.0-50-g5515f5b
[   34.015000] it87: Found IT8686E chip at 0xa40, revision 2
[   34.015040] it87: Beeping is supported
[   34.016308] it87: Found IT8792E/IT8795E chip at 0xa60, revision 3

$ sudo journalctl -b0 | grep "kernel: it87:"
Nov 17 21:13:45 user kernel: it87: loading out-of-tree module taints kernel.
Nov 17 21:13:45 user kernel: it87: it87 driver version v1.0-50-g5515f5b
Nov 17 21:13:45 user kernel: it87: Found IT8686E chip at 0xa40, revision 2
Nov 17 21:13:45 user kernel: it87: Beeping is supported
Nov 17 21:13:45 user kernel: it87: Found IT8792E/IT8795E chip at 0xa60, revision 3
````

cooper151288, 2018-11-18:
Looks like the driver detects the chips correctly. Might be a driver bug. ITE's documentation is shoddy for most chips.

sensors-detect seems to be useful for some ITE chips at finding the config registers

superiotool (use the coreboot one or a more up to date version) can also be very useful here. 

/proc/ioports shows my EC reg location correctly but i cant see the config registers:
root@GA-970A-DS3:/home/matt# grep it87 /proc/ioports
    022d-022e : it87
      022d-022e : it87

From here, I do isadump 0x22d 0x22e and it gives me an output like:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 11 00 00 00 ff ff 00 37 ff 87 48 0f 27 f2 6f ff
10: ff ff ff 73 d7 82 80 7f 01 03 ff ff ff ff ff ff
20: 4b 7c ac 8b a6 63 ba 8d 86 18 1c 15 b9 ff 0c 0c
30: ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00
40: 7f 7f 7f 7f 5a 7f 5f 40 ad 6a d4 00 ff ff ff ff
50: ff ce 7f 7f 7f c0 00 00 90 00 3e 12 64 00 00 00
60: 00 14 41 4e 20 03 0f ff 00 14 41 46 20 03 0f ff
70: 00 14 41 40 20 03 0f ff ff ff ff ff ff ff ff ff
80: 00 00 ff ff 00 00 ff ff 00 00 00 00 00 ea 00 72
90: ff 00 00 00 ff 00 00 00 42 99 00 00 01 e4 f2 15
a0: 15 7d 7d 7d 7d 7d 7d ff ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

This is my EC dump from an IT8728F. From this and a datasheet, you can see all the settings (and more, driver doesn't expose everything) and set things up your own way (isaset can be used to set the values here, provided the chip allows it). I'll see if I can find a datasheet and look through the driver code to see if I can help.
Out of interest, your sensors3.conf mentions a CPU-specific offset. Mine does too. Mine also changes (from +12 to 0) when I wake from suspend and is not reflected in it8728f registers. what does your board do. is it a gigabyte board?

KonomiKitten, 2018-11-18:
> Is it a gigabyte board?

[GA-AB350N-Gaming WIFI](https://www.gigabyte.com/Motherboard/GA-AB350N-Gaming-WIFI-rev-10#sp)

cooper151288, 2018-11-18:
Is that as root? I get all zeroes if I'm not root. It's clearly seeing 2 chips there, which is good.
is the issue you are having with writes being silently dropped? 
Do the other interfaces work ie autopwm, manual pwm (I tend to poke around in sysfs rather than lm-sensors and isaset if I don't get what I want). I'll look around for documentation and see if anythings wrong.
Are you using the master branch on my repo?

KonomiKitten, 2018-11-18:
> Is that as root?

I didn't think to try root, silly me.

````
$ sudo grep it87 /proc/ioports
    0a45-0a46 : it87
      0a45-0a46 : it87
  0a65-0a66 : it87
    0a65-0a66 : it87
````
````
$ sudo isadump 0a45 0a46
Error: Invalid address!
Syntax for I2C-like access:
  isadump [OPTIONS] [-k V1,V2...] ADDRREG DATAREG [BANK [BANKREG]]
Syntax for flat address space:
  isadump -f [OPTIONS] ADDRESS [RANGE [BANK [BANKREG]]]
Options:
  -k    Super-I/O configuration access key
  -f    Enable flat address space mode
  -y    Assume affirmative answer to all questions
  -W    Read and display word (16-bit) values
  -L    Read and display long (32-bit) values

````

````
$ sudo isadump 0a65 0a66
Error: Invalid address!
Syntax for I2C-like access:
  isadump [OPTIONS] [-k V1,V2...] ADDRREG DATAREG [BANK [BANKREG]]
Syntax for flat address space:
  isadump -f [OPTIONS] ADDRESS [RANGE [BANK [BANKREG]]]
Options:
  -k    Super-I/O configuration access key
  -f    Enable flat address space mode
  -y    Assume affirmative answer to all questions
  -W    Read and display word (16-bit) values
  -L    Read and display long (32-bit) values
````

> Are you using the master branch on my repo?

````
$ aptitude show lm-sensors
Package: lm-sensors                      
Version: 1:3.4.0-4
State: installed
Automatically installed: no
Priority: extra
Section: utils
Maintainer: Aurelien Jarno <aurel32@debian.org>
Architecture: amd64
Uncompressed Size: 394 k
Depends: sed (>= 4.0.5-1), lsb-base (>= 3.2-13), init-system-helpers (>= 1.18~), libc6 (>= 2.4), libsensors4 (>=
         1:3.1.1), perl:any
Suggests: fancontrol, read-edid, i2c-tools
Conflicts: lm-sensors:i386
Description: utilities to read temperature/voltage/fan sensors
 Lm-sensors is a hardware health monitoring package for Linux. It allows you to access information from temperature,
 voltage, and fan speed sensors. It works with most newer systems. 
 
 This package contains programs to help you set up and read data from lm-sensors.
Homepage: http://www.lm-sensors.org
Tags: admin::kernel, admin::monitoring, implemented-in::c, interface::commandline, role::program, scope::utility,
      use::monitor
````

cooper151288, 2018-11-18:
you need the 'x' in there ie:

isadump 0xa45 0xa46

(otherwise it thinks its octal)

I'm not sure what this will bring up as these have bank support but it may still be useful- given the sheet for 8728, it looks like these will be the EC registers but I may be wrong.

for the versions, i should have been more specific, I meant the it87 driver. 
I can't seem to find datasheets for either the 8792 or 8686 (though they seem to be common now) although most ITE chips seem fairly similar.

Does anything controlled by the 8792 allow setting values or is absolutely everything ignored?

I ask this mostly because I have an older Gigabyte board (GA-970A-DS3) and there are plenty of gremlins in the firmware, BIOS bugs, wonky ACPI, the lot and it has rather unusual behavour.

cooper151288, 2018-11-18:
Can you raise the issue the it87 driver page? itll make it easier to track the issue, as it definitely looks to be either a driver bug or an oddity of Gigabyte's designs. Either way it's not an lm-sensors issue
The issue may possibly be related to the address the 8792 is located at but i'm still not sure. The device ID also looks wrong in the driver code.

cooper151288, 2018-11-19:
do it on mine, that way i'll see it and try to see whats going on. the code hinds at issues with your board and compatibility. it may be for a good reason. if it is I won't be able to help a whole lot.

KonomiKitten, 2018-11-20:
> @cooper151288 Do it on mine

Your repository doesn't have issues allowed, can you please enable them?

KonomiKitten, 2018-11-21:
Issue moved to https://github.com/cooper151288/it87/issues/1 as this is not a bug with lm-sensors.

closed 2018-11-21 by KonomiKitten

cooper151288, 2018-11-23:
The config here has 2 3VSB reading. ITE chips tend to use in7 and have internal scaling (on later chips)- you might want to check it's the correct label.The highest accepted value of in6_max/min/input is 2780 (0xff times 109 tenths of an mV). Next lowest is 2769 (0xfe * 109 tenths mV). Posting this here because this relates to the configuration of lm-sensors.

KonomiKitten, 2018-11-23:
> The config here has 2 3VSB reading. ITE chips tend to use in7 and have internal scaling (on later chips)- you might want to check it's the correct label.

I have a really bad feeling I've been using the wrong config file all along, I have a `GA-AB350N-Gaming WIFI` but I've been using the `GA-AB350-GAMING3.conf` from the repo due to Debian have an old copy of lm-sensors. Has this been the problem all along? Extremely sorry if it is...

cooper151288, 2018-11-24:
Finding bad config files is never a waste of time.
Do you get an offset between the temp reported in k10temp and the temp on the AMDTSI sensor reported by the it87 driver?

Just asking because I get one which isnt in my it87 registers (12 degrees) which disappears on waking from suspend (a bunch of other things change too, think its firmware/BIOS/ACPI/gigabyte's bizarre design).

KonomiKitten, 2018-11-24:
> Do you get an offset between the temp reported in k10temp and the temp on the AMDTSI sensor reported by the it87 driver?

I believe these two are lined up (judging from tests I could run, aka cpu stressing):

````
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:
  temp1_input: 36.625
  temp1_max: 70.000
Tctl:
  temp2_input: 36.625

it8686-isa-0a40
Adapter: ISA adapter
temp3:
  temp3_input: 36.000
  temp3_max: 127.000
  temp3_min: 127.000
  temp3_alarm: 0.000
  temp3_type: 5.000
  temp3_offset: 0.000
  temp3_beep: 1.000
````

cooper151288, 2018-11-25:
Between here and the other sensor driver pages and their contributors, you should be able to put something together. Spare fans and extension cables (so you can run it out of the case) can work wonders for finding out what goes where. Some boards have even gone as far as adding test points so you have an easier, safer place to measure voltages with a voltmeter/oscilloscope. Thats mostly just high-end asus boards though.

I'll probably do one for my board too- not many people have this particular board, and it may be useful to document its quirks (theres a temp sensor i cant see somewhere that seems to measure the internal air temp near the PSU and affects fan max speed)