lm-sensors/lm-sensors #138: Unknown chip with ID 0x8586

it8586, 2018-10-20:
Hello,
I'm trying to get readout from my fan controller. I'm using Lenovo Ideapad 320-15AST so tried thinkpad_acpi module as well as pwmconfig + fancontrol. Command `sensors-detect` gives me

```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x8586
```
and after that `sensors` returns
```
thinkpad-isa-0000
Adapter: ISA adapter
fan1:           0 RPM

amdgpu-pci-0008
Adapter: PCI adapter
vddgfx:       +1.55 V  
vddnb:        +0.81 V  
fan1:             N/A
temp1:        +36.0°C  (crit =  +1.0°C, hyst = -273.1°C)
power1:           N/A  
```
There was already request for this '0x8586' chip https://www.spinics.net/lists/lm-sensors/msg43696.html but 3 years have passed since, so perhaps something has changed? On https://datasheetspdf.com/datasheet/IT8586E.html I don't see this chip, but perhaps we can use data from similar chips?
I'm also wondering if analyzing DSDT (https://github.com/hirschmann/nbfc/wiki/Analyze-your-notebook%27s-DSDT) or probing registers with nbfc tool can help https://github.com/hirschmann/nbfc/wiki/Probe-the-EC%27s-registers ? I'm willing to experiment and provide information if needed.

groeck, 2018-10-30:
This is most likely an embedded controller with no well defined functionality. As usual with ITE chips, no information about the chip is available. Sure, you could try to analyze the DSDT, but it is typically a bit more complex and difficult than the text on the referenced web pages suggests.

it8586, 2019-06-26:
Still looking for solution, and just found datasheet for this chip: http://www.datasheetcafe.com/it8586e-datasheet-pdf/. Does this help?

groeck, 2019-06-27:
That link doesn't point to a datasheet, it points to a schematic
showing how the chip is used on some board.

The IT8586 is an embedded controller. It would have to be accessed through ACPI,
if the means to do so were known. Unfortunately, that is not usually the case.
You would need support from the board vendor to get the necessary information,
then find someone to write the necessary driver.

Guenter

marzagheddon, 2024-04-20:
Sorry to necropost, but I have the same exact problem for the same machine. Has somebody found a fix