lm-sensors/lm-sensors #139: iwlwifi-virtual-0 is gone in newer kernel(s) such as 4.19-xx when compared to 4.18.xx

TuxineSw, 2018-10-28:
senors-kernel-4.18.16
@tux ~ $ sensors
iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +47.0°C  

it8622-isa-0a30
Adapter: ISA adapter
in0:          +0.92 V  (min =  +1.80 V, max =  +1.45 V)  ALARM
in1:          +1.28 V  (min =  +1.99 V, max =  +2.56 V)  ALARM
in2:          +2.23 V  (min =  +2.83 V, max =  +0.58 V)  ALARM
in3:          +2.23 V  (min =  +2.72 V, max =  +0.18 V)  ALARM
in4:          +2.02 V  (min =  +2.75 V, max =  +3.01 V)  ALARM
in5:          +1.81 V  (min =  +1.55 V, max =  +2.53 V)
in6:          +1.70 V  (min =  +1.21 V, max =  +1.33 V)  ALARM
3VSB:         +3.50 V  (min =  +5.64 V, max =  +6.12 V)  ALARM
Vbat:         +3.48 V  
+3.3V:        +3.60 V  
fan2:         784 RPM  (min =   35 RPM)
fan3:        1388 RPM  (min =   10 RPM)
temp1:        +37.0°C  (low  = +118.0°C, high = -65.0°C)  ALARM  sensor = thermistor
temp2:        +39.0°C  (low  =  -9.0°C, high = -27.0°C)  sensor = thermistor
temp3:        +37.0°C  (low  = -59.0°C, high = +118.0°C)  sensor = disabled
intrusion0:  ALARM

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:       +1.01 V  
fan1:        1305 RPM
temp1:        +45.0°C  (crit = +94.0°C, hyst = -273.1°C)
power1:       21.22 W  (cap =  48.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +39.2°C  (high = +70.0°C)
Tctl:         +39.2°C  

-----------------------------------------------------------------------------------------------------------------------
senors-kernel-4.19.xx
@tux ~ $ sensors
it8622-isa-0a30
Adapter: ISA adapter
in0:          +0.91 V  (min =  +1.80 V, max =  +1.45 V)  ALARM
in1:          +1.27 V  (min =  +1.99 V, max =  +2.56 V)  ALARM
in2:          +2.22 V  (min =  +2.83 V, max =  +0.58 V)  ALARM
in3:          +2.22 V  (min =  +2.72 V, max =  +0.18 V)  ALARM
in4:          +2.00 V  (min =  +2.75 V, max =  +3.01 V)  ALARM
in5:          +1.80 V  (min =  +1.55 V, max =  +2.53 V)
in6:          +1.70 V  (min =  +1.21 V, max =  +1.33 V)  ALARM
3VSB:         +3.50 V  (min =  +5.64 V, max =  +6.12 V)  ALARM
Vbat:         +3.48 V  
+3.3V:        +3.60 V  
fan2:         789 RPM  (min =   35 RPM)
fan3:        1394 RPM  (min =   10 RPM)
temp1:        +37.0°C  (low  = +118.0°C, high = -65.0°C)  ALARM  sensor = thermistor
temp2:        +38.0°C  (low  =  -9.0°C, high = -27.0°C)  sensor = thermistor
temp3:        +37.0°C  (low  = -59.0°C, high = +118.0°C)  sensor = disabled
intrusion0:  ALARM

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:       +1.01 V  
fan1:        1195 RPM
temp1:        +44.0°C  (crit = +94.0°C, hyst = -273.1°C)
power1:       16.20 W  (cap =  48.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +38.5°C  (high = +70.0°C)
Tctl:         +38.5°C  
-------------------------------------------------------------------------------------------------------------------------------

In KDE System-Monitor in "Thermal Zone" the "0-iwlwifi"  will appear.

groeck, 2018-10-30:
This is due to a change in the thermal subsystem of the Linux kernel. See Linux kernel upstream commit f6b6b52ef7a5 ("thermal_hwmon: Pass the originating device down to hwmon_device_register_with_info") for details. Any solution in lm-sensors would have to be discussed and agreed upon with the thermal subsystem maintainers to avoid future surprises.

olysonek, 2018-10-31:
Please try running the following branch, there's a work-in-progress fix for the problem: https://github.com/lm-sensors/lm-sensors/tree/wip_fix_sysfs_access

I haven't discussed it with the thermal subsystem maintainers yet, but I'll do that, thanks.

olysonek, 2018-11-06:
The following branch contains the fix I'd like to use (the last two commits being most significant). I'd appreciate any testing or review.
https://github.com/lm-sensors/lm-sensors/tree/WIP_fix_bus_identification

I'll discuss it with the thermal subsystem maintainers now.

groeck, 2018-11-22:
for k10temp: see upstream commit 53dfa0088edd ("hwmon: (k10temp) Support all Family 15h Model 6xh and Model 7xh processors").

TuxineSw, 2018-11-22:
> The following branch contains the fix I'd like to use (the last two commits being most significant). I'd appreciate any testing or review.
> https://github.com/lm-sensors/lm-sensors/tree/WIP_fix_bus_identification
> 
> I'll discuss it with the thermal subsystem maintainers now.

I can confirm that this version works on my machine.
tux /home/tuxine/Downloads/lm_sensors-fix/lm-sensors-WIP_fix_bus_identification # sensors
iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +38.0°C

it8622-isa-0a30
Adapter: ISA adapter
in0:          +1.07 V  (min =  +1.80 V, max =  +1.45 V)  ALARM
in1:          +1.27 V  (min =  +1.99 V, max =  +2.56 V)  ALARM
in2:          +2.22 V  (min =  +2.83 V, max =  +0.58 V)  ALARM
in3:          +2.22 V  (min =  +2.72 V, max =  +0.18 V)  ALARM
in4:          +2.02 V  (min =  +2.75 V, max =  +3.01 V)  ALARM
in5:          +1.81 V  (min =  +1.55 V, max =  +2.53 V)
in6:          +1.72 V  (min =  +1.21 V, max =  +1.33 V)  ALARM
3VSB:         +3.50 V  (min =  +5.64 V, max =  +6.12 V)  ALARM
Vbat:         +3.48 V
+3.3V:        +3.60 V
fan2:         774 RPM  (min =   35 RPM)
fan3:        1355 RPM  (min =   10 RPM)
temp1:        +36.0°C  (low  = +118.0°C, high = -65.0°C)  ALARM  sensor = thermistor
temp2:        +32.0°C  (low  =  -9.0°C, high = -27.0°C)  sensor = thermistor
temp3:        +36.0°C  (low  = -59.0°C, high = +118.0°C)  sensor = disabled
intrusion0:  ALARM

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:       +0.98 V
fan1:        1180 RPM
temp1:        +45.0°C  (crit = +94.0°C, hyst = -273.1°C)
power1:       18.24 W  (cap =  48.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +32.2°C  (high = +70.0°C)
Tctl:         +32.2°C

tux /home/tuxine/Downloads/lm_sensors-fix/lm-sensors-WIP_fix_bus_identification #    

Thank you very much for your good work!

commit 7e976f13 (2018-11-23): libsensors: Treat devices without a known ancestor bus as virtual

Currently only hwmon devices that don't have a parent device are treated
as virtual. Let's extend the concept to hwmon devices, that don't have
a *recognized ancestor* device (hwmon devices that according to the kernel
don't reside on a bus that we recognize). This change is meant to address
cases where a hwmon device has a thermal class device for a parent, but
the thermal class device doesn't have a parent device.

These kind of hwmon devices started appearing in the 4.19 kernel
due to commit f6b6b52ef7a54160c0. It was not re [... cut]
closed 2018-11-23 by olysonek